zephyrproject-rtos/zephyr #43046: Wifi sample not working with disco_l475_iot1

manuja2000, 2022-02-22:
**Describe the bug**
samples/net/wifi not working with disco_l475_iot1, showing MPU FAULT, Instruction Access Violation error.

**To Reproduce**
Steps to reproduce the behavior:

Terminal 1
1. cd zephyrproject/zephyr
2. west build -b disco_l475_iot1 samples/net/wifi
3. west flash

Terminal 2
1. minicom
(Serial Device: /dev/ttyACM0)

**Expected behavior**
Board be capable to have console

**Logs and console output**
[image]

**Environment:**
 - OS: Linux(Oracle VM VirtualBox)
 - Toolchain :Zephyr SDK

**Reference**
https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/wifi

erwango, 2022-02-22:
Thanks for reporting this issue

> Could it be the reason of the wifi sample not working?

No, I'm seeing the same issue w/o this change

ABOSTM, 2022-02-23:
After analysis, fault is due to bad SPI Chip Select initialization structure within eswifi driver:
Issue appear with PR #39341, which make use of macro `SPI_DT_SPEC_INST_GET() `which has been introduced by #30536.
`SPI_DT_SPEC_INST_GET()` end up to call `SPI_CS_CONTROL_PTR_DT()` which is defined like this

> #define SPI_CS_CONTROL_PTR_DT(node_id, delay_)			  \
> 	(&(struct spi_cs_control) {				  \
> 		.gpio = SPI_CS_GPIOS_DT_SPEC_GET(node_id), \
> 		.delay = (delay_),				  \
> 	})

From what I understand, this macro will return the address of a struct. But, if this macro is used from within a function (which is currently the case for eswifi driver), then this structure is defined locally in the stack. And this causes troubles as it may be overwritten as soon as we leave the function. 
I tried to remove the macro from the function and instead use it to initialize a global variable like it is done in can_mcp2515 driver:

> static struct eswifi_spi_data eswifi_spi0 = {
> 	.bus = SPI_DT_SPEC_INST_GET(0,
> 							     SPI_OP_MODE_MASTER | SPI_TRANSFER_MSB |
> 							     SPI_WORD_SET(16) | SPI_HOLD_ON_CS |
> 							     SPI_LOCK_ON, 1000U)};

And this fix the issue.

I could not found such restriction mentioned in the macro documentation.

@mbolivar as you are at the origin of this macro,
is it a known restriction ?
Would you consider bug in the macro itself, or in the way the macro is used ?

@bbilas , @lmajewski , It seems ethernet/dsa_ksz8xx drivers also falls in this trap.

lmajewski, 2022-02-23:
@ABOSTM - in the 
`dsa_ksz8xxx_configure_bus()` function (dsa_ksz8xxx.c) the `SPI_DT_SPEC_INST_GET()` macro fills in the `struct spi_dt_spec spi` structure (not pointer) in the globally defined `struct ksz8xxx_data`.

ABOSTM, 2022-02-24:
@lmajewski I still think it is an issue, because within `struct spi_dt_spec spi`  there is 
    spi --> config --> cs
which is a pointer and which is filled by `SPI_CS_CONTROL_PTR_DT()` with local struct

lmajewski, 2022-02-24:
@ABOSTM 
The SHA1: 8779972955e992183be6ea60f9a855ef2f388b2f made this change.

However, the `ip_k66f` board is not using GPIOs for CS for the KSZ8794 communication (the CS is controlled by the SPI HW block). I might just be lucky here.

@paveldedourek-eaton Maybe you are using it?

paveldedourek-eaton, 2022-02-25:
@lmajewski You are right, I used sw control of Chip Select.

DT:
```
&spi1 {
	status="okay";
	cs-gpios = <&gpiod 14 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
 
  pinctrl-0 = < &spi1_sck_pb3 >,
              < &spi1_miso_pb4 >,
              < &spi1_mosi_pb5 >;
....
};
```

On stm32: spi_stm32_cs_control() is used to control such GPIO.

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/spi/spi_ll_stm32.c

https://github.com/zephyrproject-rtos/zephyr/blob/1b998046340f56d9a9733c3845a8a5a733146297/include/devicetree/spi.h

BTW I took this example https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/sensor/bme280 as reference for my trials with SPI/I2C.

bbilas, 2022-02-27:
> After analysis, fault is due to bad SPI Chip Select initialization structure within eswifi driver: Issue appear with PR #39341, which make use of macro `SPI_DT_SPEC_INST_GET() `which has been [...]

Sorry for the late answer but I was in the hospital. It seems that you are right and we shouldn't initialize that structure in that way. I've prepared a fix for that but due to lack of hardware I'm not able to verify that, please see:  https://github.com/zephyrproject-rtos/zephyr/pull/43250

ABOSTM, 2022-02-28:
@manuja2000 I confirm that #43250 fixes this issue.
Nevertheless, there is a new recent regression #43205 that prevent to run this samples/net/wifi
but a fix is available #43226

closed 2022-02-28 by carlescufi