zephyrproject-rtos/zephyr #43052: DTS: fixed-partition: properties for describe read/write access level

nvlsianpu, 2022-02-22:
**Is your feature request related to a problem? Please describe.**
Currently a fixed-partition node might declare the `read-only` property. This is enough if we come to a immutable partitions, like a bootloader or fixed application.

It is easy to find a case where appropriate is that a partition should be simultaneously read-only or writable, or inaccessible  depends on which a software actor attempts to access it. For instance if there is the bootloader and application in the device, then running application resides in its own app-partition. The app-partition is `read-write` for the bootloader and is at last `read-only` for the application itself (very likely even inaccessible).

**Describe the solution you'd like**
Need to define proper access level designation. Worth to consider at last access permission suitable for:
secure-element, immutable-bootloader, mutable-bootloader, secure-application, application, thrd-party application

It seams to describe read and write access separately.

**Additional context**
this ticket was created as my conclusion to #30587
I want to emphasize that this ticket is for make consideration on DTS, not for a C API thread or a protection mechanism implementation thread.

nvlsianpu, 2022-02-22:
cc @SebastianBoe @de-nordic @Laczen

Laczen, 2022-02-22:
@nvlsianpu, I had included the read-only property in my proposal to improve working with flash partitions last summer. At that time it was found unreasonable to include this property and to disallow any writing to a flash partition that had the read-only property. At the time I had made all the requested changes to the PR. Including it now would seem unreasonable as well.

Extending it to disallowing read access means that the flash driver would need to change and take into account "partitions" as well. If this gets into the flash driver it would be better to define memory regions with their properties and to remove partitioning completely.

The app partition should be writable for the app, e.g. for image confirmation. And should also be readable if it needs to report it's version.