zephyrproject-rtos/zephyr #43054: fs/nvs move direct flash read/write into configurable routines

Laczen, 2022-02-22:
**Is your enhancement proposal related to a problem? Please describe.**
At the moment nvs is using direct calls to the flash driver for reading and writing data. This works fine but creates a problem to support working with a (read-only) nvs system from a non-secure image. Non-secure images can provide routines for accessing
flash trough the TF-M api, but they can't use the flash driver directly.

**Describe the solution you'd like**
A solution for this problem could be to provide the flash routines as configurable items in a cfg structure. These routines could then be overriden by the appropiate routines when required. This would be similar to the configuration method that is used in littlefs.

**Describe alternatives you've considered**
None

Laczen, 2022-02-22:
@sigvartmh, @SebastianBoe, @nvlsianpu, @de-nordic

SebastianBoe, 2022-02-22:
Why can't non-secure images use the flash driver directly?

SebastianBoe, 2022-02-22:
Perhaps this should be rephrased:

"This works fine but creates a problem to support working with a (read-only) nvs system from a non-secure image."

to

"This works fine but creates a problem to support working with a (read-only) nvs system located in **secure** flash from a non-secure image. Non-secure images can **use** routines for accessing **secure**
flash trough the TF-M api, but they can't use the flash driver directly."

If not then I'm not sure what is going on.

Laczen, 2022-02-22:
> Why can't non-secure images use the flash driver directly?

That could also be done, but this would require all TF-M api to be included into the flash drivers (all of them) and would require the dts to be in charge of the definition of the secure areas. Secure regions (and not partitions) need to be defined in the dts and the TF-M api would need to get the definitions from the dts. This would probably be a good solution also but is a lot more work.

As an extra bonus for the change I propose nvs could also be used on eeprom, ram, ...

You are correct about the rephrasing, I will change the according to your proposal.

closed 2023-04-27 by Laczen