zephyrproject-rtos/zephyr #43087: XCC build failures for all cavs tests/platforms

cgturner1, 2022-02-22:
When trying to build any testcase/sample using the XCC toolchain (xt-xcc) for any of the intel_adsp boards (although i suspect this will happen for any target) there is a build failure related to `#pragma GCC diagnostic not allowed inside functions`

Please also mention any information which could help others to understand
the problem you're facing:
 - targeted any intel_adsp target and affects all testcases/samples
 - Occurs with XCC toolchain
 
**Expected behavior**
Testcase successfully build when using twister

**Impact**
Currently unable to test any intel_adsp platform with XCC

**Logs and console output**
Sample partial output for one testcase below (errors are the same for all testcases).. (same pragma related error is repeated throughout the logs):
```
INFO    -  93/388 intel_adsp_cavs25         tests/benchmarks/sched/benchmark.kernel.scheduler  FAILED Build failure (build)
-- Application: /zephyrproject/zephyr/tests/benchmarks/sched
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /zephyrproject/zephyr/.cache
-- Zephyr version: 3.0.99 (/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: intel_adsp_cavs25
-- Found host-tools: zephyr 0.13.2 (/opt/toolchains/zephyr-sdk-0.13.2)
-- Found dtc: /opt/toolchains/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: xcc (/opt/toolchains/xcc/install/tools)
-- Found BOARD.dts: //zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs25/intel_adsp_cavs25.dts
-- Generated zephyr.dts: //zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: //zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/dts.cmake
Parsing //zephyrproject/zephyr/Kconfig
Loaded configuration '/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs25/intel_adsp_cavs25_defconfig'
Merged configuration '/zephyrproject/zephyr/tests/benchmarks/sched/prj.conf'
Configuration saved to '/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/.config'
Kconfig header saved to '/zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 4.2.0
-- The CXX compiler identification is GNU 4.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/xcc/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc
-- Configuring done
-- Generating done
-- Build files have been written to: /zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler
[1/134] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/zephyrproject/zephyr), build: 8c7d8ecc9aae
[2/134] Preparing syscall dependency handling

[3/134] Generating ../../../../include/generated/core-isa-dM.h
[4/134] Generating ../../../../include/generated/zsr.h
[5/134] Generating misc/generated/syscalls_subdirs.trigger
[6/134] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[7/134] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[8/134] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[9/134] Generating include/generated/driver-validation.h
[10/134] Building C object zephyr/CMakeFiles/offsets.dir/arch/xtensa/core/offsets/offsets.c.obj
[11/134] Generating include/generated/offsets.h
[12/134] Generating linker_zephyr_pre0.cmd
[13/134] Generating linker_zephyr_pre1.cmd
[14/134] Building ASM object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/xtensa-asm2-util.S.obj
[15/134] Building ASM object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/window_vectors.S.obj
[16/134] Building ASM object zephyr/arch/arch/xtensa/core/CMakeFiles/arch__xtensa__core.dir/crt1.S.obj
[17/134] Building ASM object zephyr/arch/arch/xtensa/core/startup/CMakeFiles/arch__xtensa__core__startup.dir/memerror-vector.S.obj
[18/134] Building ASM object zephyr/arch/arch/xtensa/core/startup/CMakeFiles/arch__xtensa__core__startup.dir/memctl_default.S.obj
[19/134] Building ASM object zephyr/arch/arch/xtensa/core/startup/CMakeFiles/arch__xtensa__core__startup.dir/reset-vector.S.obj
[20/134] Linking ASM static library zephyr/arch/arch/xtensa/core/startup/libarch__xtensa__core__startup.a
[21/134] Building C object zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj
FAILED: zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj 
/opt/toolchains/xcc/install/tools/RG-2017.8-linux/XtensaTools/bin/xt-xcc -DKERNEL -D_FORTIFY_SOURCE=2 -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I../../../../../../kernel/include -I../../../../../../arch/xtensa/include -I../../../../../../include -Izephyr/include/generated -I../../../../../../soc/xtensa/intel_adsp/cavs_v25 -I../../../../../../soc/xtensa/intel_adsp/common/include -I../../../../../../subsys/testsuite/include -isystem ../../../../../../lib/libc/minimal/include -Os -imacros /zephyrproject/zephyr/twister-out/intel_adsp_cavs25/tests/benchmarks/sched/benchmark.kernel.scheduler/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -imacros/zephyrproject/zephyr/include/toolchain/xcc_missing_defs.h -fms-extensions -imacros /zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -fgnu89-inline -Werror=implicit-int -fno-pic -ffunction-sections -fdata-sections -mlongcalls -std=c99 -MD -MT zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj -MF zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj.d -o zephyr/kernel/CMakeFiles/kernel.dir/fatal.c.obj -c /zephyrproject/zephyr/kernel/fatal.c
/zephyrproject/zephyr/kernel/fatal.c: In function ‘k_sys_fatal_error_handler’:
/zephyrproject/zephyr/kernel/fatal.c:43: error: #pragma GCC diagnostic not allowed inside functions
[... 176 lines cut ...]
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:233: error: #pragma GCC diagnostic not allowed inside functions
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:238: error: #pragma GCC diagnostic not allowed inside functions
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:238: error: #pragma GCC diagnostic not allowed inside functions
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v1’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v2’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v3’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v4’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v5’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v6’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: expected ‘;’ before ‘_v7’
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:238: error: #pragma GCC diagnostic not allowed inside functions
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: ‘_v4’ undeclared (first use in this function)
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: ‘_v5’ undeclared (first use in this function)
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: ‘_v6’ undeclared (first use in this function)
/zephyrproject/zephyr/arch/xtensa/core/xtensa-asm2.c:234: error: ‘_v7’ undeclared (first use in this function)

```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain XCC
 - Commit SHA: f401be157f9aa7f959f18dbf371bab7db0134a5b

**Additional context**
This was initially introduced in https://github.com/zephyrproject-rtos/zephyr/pull/42165 and was reverted (which resolved the issues) now this seems to have been reintroduced within the past day and the potential fixes in https://github.com/zephyrproject-rtos/zephyr/pull/42686 do not seem to fix the errors.

nordic-krch, 2022-02-23:
Can you try to set `CONFIG_LOG2_ALWAYS_RUNTIME=y` and then build a sample?

XCC is based on GCC that does not support some of the needed features and a way to go should be to use that option with this toolchain.

cgturner1, 2022-02-23:
Hey @nordic-krch,

So i did set `CONFIG_LOG2_ALWAYS_RUNTIME=y` and ran the full twister test suite for one target platform that was failing all the test cases and this fixed a significant amount of build failures bringing up the passing rate to 73% but there are still ~41 tests/samples that are reporting the same `#pragma GCC diagnostic not allowed inside functions` error. 

Is there another place recently introduced the pragma error that is not being caught by `CONFIG_LOG2_ALWAYS_RUNTIME=y`?

list of still failing samples and tests below:

- samples/hello_world/sample.basic.helloworld        
- samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher.same_prio 
- samples/philosophers/sample.kernel.philosopher.lifos 
- samples/philosophers/sample.kernel.philosopher.semaphores 
- samples/arch/smp/pktqueue/sample.smp.pktqueue      
- samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher.semaphores 
- samples/philosophers/sample.kernel.philosopher.same_prio 
- samples/philosophers/sample.kernel.philosopher.static 
- samples/application_development/out_of_tree_driver/sample.drivers.out_of_tree 
- samples/subsys/portability/cmsis_rtos_v2/philosophers/sample.portability.cmsis_rtos_v2.philosopher 
- samples/sensor/thermometer/sample.sensor.thermometer 
- samples/synchronization/sample.kernel.synchronization 
- samples/subsys/portability/cmsis_rtos_v1/timer_synchronization/sample.portability.cmsis_rtos_v1.timer_synchronization 
- samples/philosophers/sample.kernel.philosopher.preempt_only 
- samples/philosophers/sample.kernel.philosopher.stacks 
- samples/subsys/portability/cmsis_rtos_v1/philosophers/sample.portability.cmsis_rtos_v1.philosopher 
- samples/subsys/portability/cmsis_rtos_v1/philosophers/sample.portability.cmsis_rtos_v1.philosopher.same_prio 
- samples/subsys/tracing/tracing.user                
- samples/philosophers/sample.kernel.philosopher     
- samples/subsys/portability/cmsis_rtos_v1/philosophers/sample.portability.cmsis_rtos_v1.philosopher.semaphores 
- samples/philosophers/sample.kernel.philosopher.fifos 
- samples/philosophers/sample.kernel.philosopher.coop_only 
- samples/modules/nanopb/sample.modules.nanopb       
- samples/subsys/portability/cmsis_rtos_v2/timer_synchronization/sample.portability.cmsis_rtos_v2.timer_synchronization 
- tests/misc/test_build/buildsystem.kconfig.utf8_in_values 
- tests/posix/getopt/shell.getopt                    
- tests/subsys/shell/shell/shell.min_wildcards       
- tests/subsys/shell/shell/shell.min_tab_auto        
- tests/subsys/shell/shell/shell.min_history         
- tests/subsys/shell/shell/shell.min_metakeys        
- tests/lib/sprintf/libraries.libc.sprintf           
- tests/subsys/shell/shell/shell.min_cmds_resize     
- tests/subsys/shell/shell/shell.min_help_all        
- tests/subsys/shell/shell/shell.min_cmds_select     
- tests/subsys/shell/shell/shell.min_colors          
- tests/subsys/shell/shell/shell.min_tab             
- tests/subsys/shell/shell/shell.min_cmds_all        
- tests/subsys/shell/shell/shell.core                
- tests/subsys/shell/shell/shell.min_help            
- tests/subsys/shell/shell/shell.min                 
- tests/subsys/shell/shell/shell.min_cmds            

Full error logs for tests/subsys/shell/shell/shell.core as an example (happy to provide the full logs as well if it helps):

```
INFO    - 339/377 intel_adsp_cavs15         tests/subsys/shell/shell/shell.core                FAILED Build failure (build)
INFO    - /zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/subsys/shell/shell/shell.core/build.log
ERROR   - Loading Zephyr default modules (Zephyr base).
-- Application: /zephyrproject/zephyr/tests/subsys/shell/shell
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /zephyrproject/zephyr/.cache
-- Zephyr version: 3.0.99 (/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: intel_adsp_cavs15
-- Found host-tools: zephyr 0.13.2 (/toolchains/zephyr-sdk-0.13.2)
-- Found dtc: /toolchains/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: xcc (/toolchains/xcc/install/tools)
-- Found BOARD.dts: /zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/intel_adsp_cavs15.dts
-- Generated zephyr.dts: /zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/subsys/shell/shell/shell.core/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/subsys/shell/shell/shell.core/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/subsys/shell/shell/shell.core/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /zephyrproject/zephyr/twister-out/intel_adsp_cavs15/tests/subsys/shell/shell/shell.core/zephyr/dts.cmake
[... 402 lines cut ...]
[96/150] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/stdlib/strtol.c.obj
[97/150] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/string/strspn.c.obj
[98/150] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_common.c.obj
[... cut ...]
[114/150] Building C object zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj
[115/150] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/string/string.c.obj
[116/150] Building C object zephyr/lib/libc/minimal/CMakeFiles/lib__libc__minimal.dir/source/time/gmtime.c.obj
[117/150] Building C object CMakeFiles/app.dir/src/main.c.obj
[118/150] Building C object zephyr/soc/xtensa/intel_adsp/common/CMakeFiles/intel_adsp_common.dir/trace_out.c.obj
ninja: build stopped: subcommand failed.
```

marc-hb, 2022-02-24:
Can you share a sample (twister? west?) command to reproduce? For some reason we never hit this while building SOF, here's a recent PR where everything builds fine: https://sof-ci.01.org/sofpr/PR5410/build12120/build/

keqiaozhang, 2022-02-24:
@marc-hb, because we didn't use the latest zephyr commit to build SOF. We have a zephyr "pre-test" Jenkins job to select the stable zephyr commit for our SOF build.

nordic-krch, 2022-02-24:
It seems that `CONFIG_LOG2_ALWAYS_RUNTIME` was not set. Cmake warning is generated only when it [is not set](https://github.com/zephyrproject-rtos/zephyr/blob/5dddf9f0f8d89470e1cd63d61feb2ce40b7e7070/subsys/logging/CMakeLists.txt#L24):
```
CMake Warning at ../../../../subsys/logging/CMakeLists.txt:32 (message):
  Compiler version requires CONFIG_LOG2_ALWAYS_RUNTIME to be set

dcpleung, 2022-02-24:
The issue at hand is that `LOG_*()` are still expanded when `CONFIG_LOG=n`. As `CONFIG_LOG2_ALWAYS_RUNTIME` depends on `CONFIG_LOG`, the macro expansion goes the path with `_Pragma()` and auto type, which results in build errors.

closed 2022-02-28 by carlescufi