zephyrproject-rtos/zephyr #43094: CMake stack overflow after enabling or changing logging mode.

smrtos, 2022-02-23:
**Describe the bug**
Recursive CMake calls cause stack overflow in Cmake after enabling or changing logging mode.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/builds
2. cmake -B build -S ../samples/hello_world/ -GNinja -DBOARD=qemu_x86_64
3. cd build
4. ninja run (hello, world is seen as expected)
[image]

5. ninja menuconfig
6. **Enable** logging as below and exit menuconfig
[image]

7. ninja
8. See error like below. Same error also shown when **changing** logging mode.
    `ninja menuconfig` cannot run after this error. Entire build folder has to be removed and start all over again.

```
CMake Error at /home/ming/zephyrproject2/zephyr/cmake/modules/version.cmake:35 (include):
  Maximum recursion depth of 1000 exceeded
Call Stack (most recent call first):
  /home/ming/zephyrproject2/zephyr/share/zephyr-package/cmake/ZephyrConfigVersion.cmake:59 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/zephyr_package_search.cmake:78 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfigVersion.cmake:78 (check_zephyr_package)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/ming/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  /home/ming/zephyrproject2/zephyr/cmake/app/boilerplate.cmake:20 (find_package)
...
```

Investigation:
Seems there's some recursive calls in the Cmake script causing stack overflow.

smrtos, 2022-02-23:
Actually, just open menuconfig and change nothing can also reproduce this issue.

Steps:
1. cd zephyr/builds
2. cmake -B build -S ../samples/hello_world/ -GNinja -DBOARD=qemu_x86_64
3. cd build
4. ninja
5. ninja menuconfig
6. (change nothing, just press ESC and quit WITH SAVING)
7. Same error as above shows up.

smrtos, 2022-02-23:
The issue is introduced by this commit: https://github.com/zephyrproject-rtos/zephyr/commit/61453e4a586fb0d865550f9f92708f3c05412bf9

smrtos, 2022-02-24:
Even simpler reproduce steps:

1. cmake -B build -S ../samples/basic/blinky  -GNinja -DBOARD=reel_board
2. cd build
3. ninja
4. touch zephyr/.config
5. ninja

Same error as above.

lairdjm, 2022-02-25:
Can we mark this as high priority given that any sort of configuration before building completely breaks it and is impossible?

tejlmand, 2022-02-25:
@smrtos Investigated this issue and is able to reproduce now.

First thing I notice is that you apparently have two (or more) Zephyr workspaces, which is perfectly valid and supported.

- /home/ming/zephyrproject/zephyr
- /home/ming/zephyrproject2/zephyr

I noticed the looping happens between those two versions and I have managed to reproduce it locally under the following conditions:
- One Zephyr revision is before those changes, for example v2.7.0
- Other Zephyr revision is new, for example `zephyr/main`
- **only** the **old** Zephyr is registered in the CMake package registry (`west zephyr-export`).

If both workspaces are exported, then things will start to work again.

So you should go into:
```
$ cd /home/ming/zephyrproject2/zephyr
$ west zephyr-export
```

That should make things work again.

I'm looking into how this may be fixed in Zephyr, but in the meantime you have a local solution.

tejlmand, 2022-02-25:
> Can we mark this as high priority given that any sort of configuration before building completely breaks it and is impossible?

Marked it `low` cause the getting started guide clearly states that `west zephyr-export` should be executed, and apparently that guide has not been followed on one of the workspaces.

https://docs.zephyrproject.org/latest/getting_started/index.html#get-zephyr-and-install-python-dependencies
> 3. Export a [Zephyr CMake package](https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#cmake-pkg). This allows CMake to automatically load boilerplate code required for building Zephyr applications.
>
> `west zephyr-export`

that said, I will still look into a fix asap.

smrtos, 2022-02-26:
> > Can we mark this as high priority given that any sort of configuration before building completely breaks it and is impossible?
> 
> Marked it `low` cause the getting started guide clearly states that `west zephyr-export` should be executed, and apparently that guide has not been followed on one of the workspaces.
> 
> [docs.zephyrproject.org/latest/getting_started/index.html#get-zephyr-and-install-python-dependencies](https://docs.zephyrproject.org/latest/getting_started/index.html#get-zephyr-and-install-python-dependencies)
> 
> > 3. Export a [Zephyr CMake package](https://docs.zephyrproject.org/latest/guides/zephyr_cmake_package.html#cmake-pkg). This allows CMake to automatically load boilerplate code required for building Zephyr applications.
> > 
> > `west zephyr-export`
> 
> that said, I will still look into a fix asap.

Thanks! Yes, the `zephyrproject2` workspace is indeed not exported. I just tried your solution. It can work now.

smrtos, 2022-02-26:
> Marked it `low` cause the getting started guide clearly states that `west zephyr-export` should be executed, and apparently that guide has not been followed on one of the workspaces. [...]

Thanks! Yes, the zephyrproject2 workspace is indeed not exported. 
Your solution can fix in my scenario.

tejlmand, 2022-03-01:
> I already did that yesterday, and it did not resole the issue.

Thanks, digging a bit deeper this is also depending on the order of which CMake decides to load each package for testing, as well as the way CMake stores `Zephyr_DIR` to the cache.

This has been fixed in #43311

PR #43311 (2022-03-07): Zephyr CMake package handling / Zephyr_DIR cache variable
Fixes: #43094

This PR improves the handling of Zephyr CMake package by ensuring the `Zephyr_DIR` CMake cache variables is set correctly when loading the Zephyr CMake package.

Furthermore it improves handling when loading Zephyr CMake package in systems where both latest Zephyr and Zephyr versions <= 3.0.0 are installed side by side.

Both when they are registered in the CMake package registry, as well as when only a single installation is registered.
commit f96ee77c (2022-03-07): cmake: function to update Zephyr_DIR when loading old Zephyr packages

Fixes: #43094

This commit introduces a function which updates Zephyr_DIR to point to
the directory of the Zephyr package being loaded.

For Zephyr 3.0 and earlier, the Zephyr_DIR might in some cases be
`Zephyr_DIR-NOTFOUND` or pointing to the Zephyr package including the
boilerplate code instead of the Zephyr package of the included
boilerplate code.

This code ensures that when a package is loaded then Zephyr_DIR will
point correctly.
This ensures that when Zephyr releases <=3.0 is loaded, then Zephyr_DIR
will point corre [... cut]
closed 2022-03-07 by carlescufi