zephyrproject-rtos/zephyr #43095: Inconsistent logging config result resulted from menuconfig.

smrtos, 2022-02-23:
**Describe the bug**
Inconsistent logging config result resulted from menuconfig.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr
2. cmake -B build -S samples/hello_world/ -GNinja -DBOARD=qemu_x86_64
2. cd build
3. ninja menuconfig
4. Check logging and enter the logging sub menu. It looks like this now:
[image]
5.  Check "Synchronous v2 (DEPRECATED)".  The "Mode" field will also change.  This is as expected.
     Shown as below:
[image]

6. Enter Mode submenu. Check "Deferred logging". As below:
[image]

7. Press ESC to return to parent menu. Now it shows inconsistent result as both deferred and synchronous:
[image]

8. Check the final effetive zephyr/.config file. The inconsistency is persisted.
[image]

**Expected behavior**
Only one logging mode should be displayed in menuconfig and  persisted in .config file.

carlescufi, 2022-03-01:
@tejlmand is this an instance of the [stuck symbol problem](https://docs.zephyrproject.org/latest/guides/build/kconfig/tips.html#stuck-symbols-in-menuconfig-and-guiconfig)?

tejlmand, 2022-03-03:
@smrtos this happens because of the deprecation of those symbols introduced with this PR #41755 .

@nordic-krch will you take a look at this ?
Initial look at the Kconfig indicates to me that maybe `choice LOG_MODE` should `depends on !<DEPRECATED_SYMBOLS>` so that if user selects any of the deprecated symbols, then it's not possible to also select a choice item.

But I think you are the best to judge the intentions with the deprecation introduced with https://github.com/zephyrproject-rtos/zephyr/pull/41755

nordic-krch, 2022-03-03:
Fixed a bit dependencies. @smrtos note that options are deprecated and not recommended to be used. They were added only to support case when someone already had them set in `prj.conf`.

closed 2022-03-03 by nashif