zephyrproject-rtos/zephyr #43099: ARCH roots issue

erwango, 2022-02-23:
**Describe the bug**
I'm not able to build since https://github.com/zephyrproject-rtos/zephyr/pull/41301.
This is likely due to my local installation, but I can't get what should be updated.

For instance, when running `west build -p auto -b nucleo_f401re samples/hello_world/`, I'm getting the following error:
```
CMake Error at /local/mcu/zephyrproject/zephyr/cmake/arch.cmake:41 (message):
  Could not find ARCH=nucleo_f401re for BOARD=nucleo_f401re, please check
  your installation.  ARCH roots searched:
```
Issue shows up with with https://github.com/zephyrproject-rtos/zephyr/commit/254e35d61d59feca6e9b91fcc9ce7d3416917c22

I've tried to re-run `source zephyr-env.sh` with no success

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04.6 LTS
 - Toolchain (e.g Zephyr SDK, ...): zephyr-sdk-0.13.2
 - CMake: 3.22.1
 - Python 3.8

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
Here is the full log to the error:
```
$ west -vvv build -p auto -b nucleo_f401re samples/hello_world/
args namespace: Namespace(command='build', help=None, verbose=3, zephyr_base=None)
ZEPHYR_BASE=/local/mcu/zephyrproject/zephyr (origin: env)
args: Namespace(board='nucleo_f401re', build_dir=None, build_opt=[], cmake=False, cmake_only=False, command='build', dry_run=False, force=False, help=None, pristine='auto', source_dir=None, target=None, test_item=None, verbose=3, zephyr_base=None) remainder: ['samples/hello_world/']
source_dir: samples/hello_world/ cmake_opts: None
setting up build directory
setting up source directory
config dir-fmt: build
build is a zephyr build directory
build dir: build
pristine: auto auto_pristine: True
/local/mcu/zephyrproject/zephyr/build is a zephyr build directory
setting up source directory
sanity checking the build
/local/mcu/zephyrproject/zephyr/samples/hello_world is NOT a valid zephyr build directory
APPLICATION_SOURCE_DIR: /local/mcu/zephyrproject/zephyr/samples/hello_world
pristine: True
CACHED_BOARD: nucleo_f401re
-- west build: generating a build system
cmake version 3.22.1 is OK; minimum version is 3.13.1
Running CMake: /local/home/frq07517/.local/bin/cmake -DWEST_PYTHON=/usr/bin/python3.8 -B/local/mcu/zephyrproject/zephyr/build -S/local/mcu/zephyrproject/zephyr/samples/hello_world -GNinja
Including boilerplate (Zephyr base (cached)): /local/mcu/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /local/mcu/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 3.0.99 (/local/mcu/zephyrproject/zephyr)
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nucleo_f401re
CMake Error at /local/mcu/zephyrproject/zephyr/cmake/arch.cmake:41 (message):
  Could not find ARCH=nucleo_f401re for BOARD=nucleo_f401re, please check
  your installation.  ARCH roots searched:

  /local/mcu/zephyrproject/zephyr
Call Stack (most recent call first):
  /local/mcu/zephyrproject/zephyr/cmake/app/boilerplate.cmake:387 (include)
  /local/mcu/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /local/mcu/zephyrproject/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

```

FRASTM, 2022-02-23:
I do not see that error building on dd3f251f1006ae84283acf60ddc2c7f2cefad1e6
and when my cmake is `cmake version 3.21.1 is OK; minimum version is 3.13.1`

erwango, 2022-02-23:
@tejlmand Could you provide more details on error message:
`Could not find ARCH=nucleo_f401re `

Why would ARCH be the board ID ?

henrikbrixandersen, 2022-02-23:
I can confirm this issue. I can reproduce it with CMake v3.22.2. No issues when using CMake 3.21.4.

tejlmand, 2022-02-24:
This is caused by a bug in the CMake 3.22 PyPI package, ref: https://gitlab.kitware.com/cmake/cmake/-/issues/23187

and this code is triggering said bug: https://github.com/zephyrproject-rtos/zephyr/blob/5ee306944b37fac3143f30e759478c6e8a9d36a4/cmake/modules/arch.cmake#L32-L33

Looking into how we can restructure code, so we don't trigger the CMake bug.

tejlmand, 2022-02-24:
see also: https://github.com/scikit-build/cmake-python-distributions/issues/221

yperess, 2022-09-29:
I'm running into this issue from the Pigweed CI now only the check wasn't catching it. It's on CMake 3.24.2. Can we just get rid of the cmake version check? Or just ban `cmake_path` in favor of https://cmake.org/cmake/help/latest/command/get_filename_component.html?

yperess, 2022-09-29:
I set priority medium because this is breaking downstream CIs (feel free to adjust)

mbolivar-nordic, 2022-09-29:
I'm using 3.24.2 as well and I don't see this. Using the cmake that comes with Arch Linux.

yperess, 2022-09-30:
It seems that the toolchain used to build cmake for our CI is the same as the one used for PyPI which is how we're coming across this. The issue still exists in 3.24.2 depending on how cmake is built. I propose that we remove the version check and just see if `cmake_path` returns the same path, thoughts @mbolivar-nordic?

mbolivar-nordic, 2022-10-07:
> thoughts @mbolivar-nordic?

No objections, but this is really a question for @tejlmand

PR #43147 (2022-02-25): cmake: test CMake version and issue error if using 3.22.1 / 3.22.2 from PyPI
Fixes: #43099

The CMake 3.22.1 / 3.22.2 PyPI suffers a bug in the
 `cmake_path(... PARENT_PATH)` implementation.

Therefore, when CMake version 3.22.1 / 3.22.2 is detected, test if the CMake
version is suffering from the bug, and in case the bug is present, fail
with an error regarding the issue.

The reason for failing, and not implementing work arounds is that Zephyr
already uses `cmake_path()` at two locations, and we cannot prevent
contributors from adding code which uses this function.

Secondly, Zephyr modules  [... cut]
commit 8b199b0f (2022-02-25): cmake: test CMake and issue error if using 3.22.1 / 3.22.2 from PyPI

Fixes: #43099

The CMake 3.22.1 / 3.22.2 PyPI version suffers a bug in the
`cmake_path(... PARENT_PATH)` implementation.

Therefore, when CMake version 3.22.1 / 3.22.2 is detected, test if the
CMake version is suffering from the bug, and in case the bug is present,
fail with an error regarding the issue.

The reason for failing, and not implementing work arounds is that Zephyr
already uses `cmake_path()` at two locations, and we cannot prevent
contributors from adding code which uses this function.

Secondly, Zephyr modules  [... cut]
closed 2023-02-26 by github-actions[bot]