zephyrproject-rtos/zephyr #43117: Not possible to create more than one shield.

rkudaj, 2022-02-23:
**Describe the bug**
When multiple shields are defined, only the shield last in the -DSHIELD
list gets defined in `.config`. This is due to too many backslashes
used defining it for an env setting.

Please also mention any information which could help others to understand
We have our in-company designed board.

**To Reproduce**
Steps to reproduce the behavior:
1.west build -b xxx_nrf5340_cpuapp -- -DSHIELD="cf_shield1 cf_shield2"

**Expected behavior**
Build should be finalized with success

**Impact**
showstopper

**Logs and console output**
In function `spi_transceive':
build/zephyr/include/generated/syscalls/spi.h:42: undefined reference to `__device_dts_ord_102'
/home/user/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(spi_driver_interface.c.obj): in function `uwb_spi_init':
src/spi_driver_interface.c:51: undefined reference to `__device_dts_ord_102'
/home/user/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a/src/spi_driver_interface.c:25: undefined reference to `__device_dts_ord_102'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - commit 3f826560aaf81a444018293bd6acce3c339fe150 (HEAD, tag: zephyr-v2.7.0, tag: v2.7.0, manifest-rev)

**Additional context**
It seems that fix for this issue has already been done, here:
https://github.com/zephyrproject-rtos/zephyr/pull/40829/commits/ec3b870a92e98d0cd0c513178851ebd7fec75d67

but it was not backported to zephyr version 2.7.0.

erwango, 2022-02-23:
Thanks for reporting this, I've requested backport of https://github.com/zephyrproject-rtos/zephyr/pull/40829

erwango, 2022-02-23:
@rkudaj would you mind testing https://github.com/zephyrproject-rtos/zephyr/pull/43122 and see if fixing your issue ?

rkudaj, 2022-02-24:
Hi, 
I tested , all looks good:)
regards

closed 2022-05-11 by github-actions[bot]