zephyrproject-rtos/zephyr #43130: STM32WL ADC idles / doesn't work

nobodywasishere, 2022-02-23:
**Describe the bug**
The STM32WL on the LoRa E5 Dev Board fails to read from the ADC with the example code in PlatformIO.

Please also mention any information which could help others to understand
the problem you're facing:
 - Zephyr v2.7.1
 - Ran in debug mode to step through to see where it gets stuck but end up looping through lines 93-99 of `idle.c` (this may be normal - I'm very new to Zephyr)

**To Reproduce**
Source code [here](https://github.com/nobodywasishere/ee193/tree/main/03-adc/zephyr-adc)
Steps to reproduce the behavior:
1. `pio run -t upload`, build and upload to the device
2. `picocom --baud 115200 /dev/ttyUSB0` connect to the device over serial
3. Press reset, will simply say 'Looping...' and nothing else
4. Running in debug mode, it gets into a loop

**Expected behavior**
Prints out the values of the ADC to the serial monitor.

**Impact**
Severe annoyance (will have to figure out how to use ADC manually)

**Logs and console output**

[PlatformIO clean/build/upload log](https://github.com/zephyrproject-rtos/zephyr/files/8127984/upload.log)

```
$ picocom --baud 115200 /dev/ttyUSB0 

picocom v3.1

...

Type [C-a] [C-h] to see available commands
Terminal ready
*** Booting Zephyr OS build zephyr-v20701  ***
Looping...
```

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: Zephyr PlatformIO
 - Version: v2.7.1

**Additional context**

Similar to #37352 but I have CONFIG_PM disabled.

erwango, 2022-02-24:
Thanks for reporting.
I've been able to reproduce the issue on `samples/drivers/adc` (using V2.7.1) adding a nucleo_wl55jc.overlay with the following dts fragment:
```
/ {
	zephyr,user {
		/* adjust channel number according to pinmux in board.dts */
		io-channels = <&adc1 3>;
	};
};

&adc1 {
	pinctrl-0 = <&adc_in3_pb4>;
	status = "okay";
};
```

I've also managed to test it successfuly on [this branch](https://github.com/zephyrproject-rtos/zephyr/pull/34751) that introduced the ADC support for WL, so this is likely a regression. WWe'll have a look.

abxy, 2022-02-24:
Sounds like the issue I fixed with this PR: https://github.com/zephyrproject-rtos/zephyr/pull/41548

erwango, 2022-02-24:
I've done a git bisect that pointed me to 58b8c4006e1bc0bb2ee8056b807ecb11e1be71bb at the initial culprit

> Sounds like the issue I fixed with this PR: #41548

Thanks @abxy. I had a try and issue at that point was a bit different: Driver was not blocked as could be seen on V2.7.1 but reporting 0:
```
ADC reading: 0                                                                  
ADC reading: 0                                                                  
ADC reading: 0                                                                  
ADC reading: 0    
```
I had a try to backport #41548 but there were too many changes to do it quickly.
Need to see how to fix V2.7 branch w/o picking up all changes on V3.0

erwango, 2022-02-25:
See https://github.com/zephyrproject-rtos/zephyr/pull/43217

closed 2022-03-03 by erwango

erwango, 2022-03-03:
> I am now running into this issue using Zephyr version: 3.0.99 and build zephyr-v3.0.0-594-g9510f961793a. The ADC is just reporting 0's.

Weird, was working fine last week.

erwango, 2022-03-04:
@nobodywasishere I've just tested and it's working fine on my side on `zephyr-v3.0.0-700-g8b4a3e2328`:
```
*** Booting Zephyr OS build zephyr-v3.0.0-700-g8b4a3e23287f  ***                
ADC reading: 1390                                                               
ADC reading: 1443                                                               
ADC reading: 1449                                                               
ADC reading: 1451                                                               
ADC reading: 1452                                                               
ADC reading: 1452    
```

erwango, 2022-03-09:
> Just so I understand, what does the number <&adc1 XX> signify?

it represents the channel of an adc