zephyrproject-rtos/zephyr #43132: Thingy:52 i2c_nrfx_twim: Error 0x0BAE0001 occurred for message

WilliamGFish, 2022-02-24:
**Describe the bug**
Thingy:52 does not boot, sx1509b failing to initialise

Using thingy:52 and receiving this error on boot.

[00:00:00.440,368] \033[1;31m<err> i2c_nrfx_twim: Error 0x0BAE0001 occurred for message 0\033[0m
[00:00:00.440,368] \033[1;31m<err> sx1509b: GPIO_P0: reset m0 failed: -5
\033[0m
[00:00:00.440,399] \033[1;31m<err> sx1509b: GPIO_P0 init failed: -5\033[0m
[00:00:00.440,460] \033[1;31m<err> regulator_fixed: GPIO port: GPIO_P0 not ready\033[0m
[00:00:00.440,460] \033[0m<inf> regulator_fixed: ccs-pwr-ctrl onoff: -19\033[0m
[00:00:00.440,521] \033[1;31m<err> regulator_fixed: GPIO port: GPIO_P0 not ready\033[0m
[00:00:00.440,521] \033[0m<inf> regulator_fixed: mic-pwr-ctrl onoff: -19\033[0m
[00:00:00.440,551] \033[1;31m<err> regulator_fixed: GPIO port: GPIO_P0 not ready\033[0m
[00:00:00.440,582] \033[0m<inf> regulator_fixed: mpu-pwr-ctrl onoff: -19\033[0m
[00:00:00.440,612] \033[0m<inf> regulator_fixed: spk-pwr-ctrl sync: 0\033[0m
[00:00:00.440,643] \033[0m<inf> regulator_fixed: vdd-pwr-ctrl sync: 0\033[0m

**Impact**
Thingy:52 is unusable.

**Environment (please complete the following information):**
 - OS: Windows)
 - Toolchain : gcc
 - Version used: 3.0.0 r3

anangl, 2022-03-02:
I cannot reproduce such behavior. I checked this with the hello_world sample on commit 7555cf9dd7ce99073c7772ce163f3ea810fb0757 and I got the following output log:
```
[00:00:00.360,382] <inf> sx1509b: GPIO_P0 init ok
[00:00:00.361,297] <inf> regulator_fixed: ccs-pwr-ctrl onoff: 0
[00:00:00.362,152] <inf> regulator_fixed: mic-pwr-ctrl onoff: 0
[00:00:00.363,037] <inf> regulator_fixed: mpu-pwr-ctrl onoff: 0
[00:00:00.363,067] <inf> regulator_fixed: spk-pwr-ctrl sync: 0
[00:00:00.363,098] <inf> regulator_fixed: vdd-pwr-ctrl sync: 0
```
For what application are you observing this? Can you check in `zephyr.dts` that `i2c0` pins are assigned properly, i.e.:
```
sda-pin = < 0x7 >;
scl-pin = < 0x8 >;
```
and these pins are not driven by anything else (accidentally assigned also to some other peripheral, for example)? Otherwise, it looks like there is some hardware issue in your Thingy:52.

WilliamGFish, 2022-03-02:
How odd, but I've just merged all the latest changes from the main branch and it's all good.

Thanks for the quick response keep up the great efforts.

Billy..