zephyrproject-rtos/zephyr #43135: stm32: uart: Support for wakeup from stop

ycsin, 2022-02-24:
The STM32's UART driver currently lacks the ability to wake up from stop mode. For low power application that requires to listen for UART RX (modem, sensors), the applications needs to delicately constraint the PM mode while the MCU is communicating with these external peripherals. This makes the LPM not as easy to use, and for applications where the MCU should be in STOP mode unless the RX receives something, this means that LPM cant be enabled at all.

**Describe the solution you'd like**
Add support for wake up from stop mode for the STM32's UART driver.

**Describe alternatives you've considered**
I've tried to implement this feature but stumbled upon the step where I need to switch the clock source of WKUP capable UART instances to HSI16 using `LL_RCC_SetUSARTClockSource` before the MCU goes to sleep, this function accepts defined macros which can be different across MCUs

**Reference(s)**
- https://github.com/STMicroelectronics/STM32CubeG0/blob/master/Projects/NUCLEO-G0B1RE/Examples_LL/LPUART/LPUART_WakeUpFromStop_Init/Src/main.c

erwango, 2022-02-24:
I'm currently working on https://github.com/zephyrproject-rtos/zephyr/pull/42097 which should allow to configure alternate clocks like HSI16. At least in a first step, this would not be dynamic, but would allow to configure the right clock that enables to enter STOP mode.
There are still many steps from #42097 to what you exactly need, but that's the way to go and work is started.

erwango, 2022-08-12:
Fixed by #48640

closed 2022-08-12 by erwango