zephyrproject-rtos/zephyr #43142: When Ethernet and PPP exist at the same time, data cannot be sent normally

maweicoltsmart, 2022-02-24:
The PCB I designed myself has an Ethernet interface and a GSM PPP interface. My application creates a UDP connection, this UDP connection can correctly receive and send data through Ethernet before PPP gets IP, but after PPP gets IP, this UDP connection can only receive data, but cannot send data , I'm guessing that the data may be sent from PPP. Is there any way to solve this situation of multiple network card communication?

rlubos, 2022-02-24:
I'm assuming you're using socket interface? You could use `SO_BINDTODEVICE` socket option to bind your socket to a specific network interface.

maweicoltsmart, 2022-02-25:
@rlubos 
Thanks, the problem has been solved with the suggestion you gave. Thanks again!

closed 2022-02-25 by maweicoltsmart