zephyrproject-rtos/zephyr #43159: hal: stm32: ltdc pins should be very-high-speed

emillindq, 2022-02-24:
**Describe the bug**
Recently LTDC pins for various STM32 series were added in its hal's dtsi files. However, these pins should be configured to be configured with very-high-speed slew-rate, else (at least for high-resolution displays) it will not work. It's probably also a good idea to configure them with internal pull-ups/downs to control the signals before the LTDC peripheral is configured and takes control over the pins. So this should be added for every signal
```
bias-pull-up;
slew-rate = "very-high-speed";
```

emillindq, 2022-02-24:
> @emillindq Any idea of the impacts of using very-high-speed slew-rate for non high-resolution displays ?

I am unable to verify this, but I don't see how sharper edges on lower speed might cause problems other than higher electromagnetic emissions. But IMHO that should rather be solved with hardware. If nothing else, a user can copy the dts pin definitions to a local dtsi file and change back the slew rate. This is what we have done, however to enable the higher speed, and also because the ltdc pins weren't there when we got started with ltdc.

gmarull, 2022-02-25:
I think that boards should be the ultimate responsible for this sort of settings. The pinctrl guide discourages adding this sort of defaults because they usually don't fit all usecases (see the note at the end of https://docs.zephyrproject.org/latest/guides/pinctrl/index.html#pin-configuration). I personally think ST should remove the defaults from autogenerated nodes, or at least remove some of them (e.g. the I2C pull-up, which is wrong).

emillindq, 2022-02-25:
> I think that boards should be the ultimate responsible for this sort of settings. The pinctrl guide discourages adding this sort of defaults because they usually don't fit all usecases (see the note [...]

You are speaking truth, I've run into problems with the default i2c pull-ups myself. Since it's possible to add properties to a dts node using &, I'm closing this issue. Thanks!

closed 2022-02-25 by emillindq