zephyrproject-rtos/zephyr #43163: Applications not pulling LVGL cannot be configured or compiled

gmarull, 2022-02-24:
**Describe the bug**

Out of tree applications not pulling `lvgl` module cannot be compiled.

**To Reproduce**

It can be reproduced e.g. with https://github.com/teslabs/spinner. The project uses the following manifest:

```
manifest:
  self:
    path: spinner

  remotes:
    - name: zephyr
      url-base: https://github.com/zephyrproject-rtos

  projects:
    - name: zephyr
      remote: zephyr
      repo-path: zephyr
      revision: main
      import:
        name-allowlist:
          - cmsis
          - hal_stm32
```

**Expected behavior**

Everything works without LVGL being pulled.

**Impact**

Blocker

**Logs and console output**

```
warning: LV_Z_POINTER_KSCAN (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:46, boards/shields/buydisplay_2_8_tft_touch_arduino/Kconfig.defconfig:46, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:43) defined without a type

warning: LV_Z_POINTER_KSCAN_DEV_NAME (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:49, boards/shields/buydisplay_2_8_tft_touch_arduino/Kconfig.defconfig:49, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:46) defined without a type                                                                                                                                                                                            warning: LV_Z_POINTER_KSCAN_SWAP_XY (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:52, boards/shields/buydisplay_2_8_tft_touch_ard
uino/Kconfig.defconfig:52, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:49) defined without a type
                                                                             
warning: LV_Z_POINTER_KSCAN_INVERT_X (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:55, boards/shields/buydisplay_2_8_tft_touch_ar
duino/Kconfig.defconfig:55, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:52) defined without a type
                                      
warning: LV_Z_POINTER_KSCAN_INVERT_Y (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:58, boards/shields/buydisplay_2_8_tft_touch_ar
duino/Kconfig.defconfig:58, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:55) defined without a type
                                                                                                                                                          
warning: LV_COLOR_16_SWAP (defined at boards/shields/buydisplay_2_8_tft_touch_arduino/Kconfig.defconfig:40, boards/shields/st7789v_generic/Kconfig.defconf
ig:29) defined without a type    

warning: <choice LV_COLOR_DEPTH_BITS> (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:39, boards/shields/buydisplay_2_8_tft_touch_a
rduino/Kconfig.defconfig:36, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:36, boards/shields/ls0xx_generic/Kconfig.defconfig:28, boar
ds/shields/ssd1306/Kconfig.defconfig:41, boards/shields/st7789v_generic/Kconfig.defconfig:25, boards/shields/waveshare_epaper/Kconfig.defconfig:37) defined with type unknown                                                                                                                                                                             
warning: <choice LV_COLOR_DEPTH_BITS> (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.defconfig:39, boards/shields/buydisplay_2_8_tft_touch_a
rduino/Kconfig.defconfig:36, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:36, boards/shields/ls0xx_generic/Kconfig.defconfig:28, boar
ds/shields/ssd1306/Kconfig.defconfig:41, boards/shields/st7789v_generic/Kconfig.defconfig:25, boards/shields/waveshare_epaper/Kconfig.defconfig:37) define
d without a prompt

warning: the default selection LV_COLOR_DEPTH_32 (undefined) of <choice LV_COLOR_DEPTH_BITS> (defined at boards/shields/adafruit_2_8_tft_touch_v2/Kconfig.
defconfig:39, boards/shields/buydisplay_2_8_tft_touch_arduino/Kconfig.defconfig:36, boards/shields/buydisplay_3_5_tft_touch_arduino/Kconfig.defconfig:36, 
boards/shields/ls0xx_generic/Kconfig.defconfig:28, boards/shields/ssd1306/Kconfig.defconfig:41, boards/shields/st7789v_generic/Kconfig.defconfig:25, board
s/shields/waveshare_epaper/Kconfig.defconfig:37) is not contained in the choice

...
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: b74397db13

**Additional context**
 N/A

gmarull, 2022-02-24:
cc: @brgl

brgl, 2022-02-24:
@gmarull on it, can you assign the issue to me?

closed 2022-02-25 by mbolivar-nordic