zephyrproject-rtos/zephyr #43190: Bluetooth: audio: HCI command timeout on LE Setup Isochronous Data Path

MariuszSkamra, 2022-02-25:
**Describe the bug**
Zephyr is the Unicast Server.
The host fails to receive HCI Command Complete on LE Setup Isochronous Data Path command. The last debug message that is processed is ISO Data. The HCI Command Complete even if sent by controller is not processed by host stack.

BT_ASSERT_MSG(err == 0, "k_sem_take failed with err %d", err);

**Impact**
Cannot establish stream

**Logs and console output**
```
[00:04:06.296,000] <dbg> bt_ascs: ascs_ep_get_status_enable: dir 0x01 cig 0x01 c                                                                                                                                                                                        is 0x01
[00:04:06.296,000] <dbg> bt_hci_core: process_events: count 3
[00:04:06.296,000] <dbg> bt_hci_core: process_events: ev->state 0
[00:04:06.296,000] <dbg> bt_hci_core: process_events: ev->state 0
[00:04:06.296,000] <dbg> bt_hci_core: process_events: ev->state 4
[00:04:06.296,000] <dbg> bt_hci_core: bt_send: buf 0x817ea50 len 20 type 2
[00:04:06.296,000] <dbg> bt_hci_core: hci_tx_thread: Calling k_poll with 3 event                                                                                                                                                                                        s
[00:04:06.360,000] <dbg> bt_hci_core: bt_recv: buf 0x817ce34 len 7 type 1
[00:04:06.360,000] <dbg> bt_hci_core: bt_recv: evt 0x13, len 5, evt_flags 0x01
[00:04:06.360,000] <dbg> bt_hci_core: hci_num_completed_packets: num_handles 1
[00:04:06.360,000] <dbg> bt_hci_core: hci_num_completed_packets: handle 64 count                                                                                                                                                                                         2
[00:04:06.456,000] <dbg> bt_hci_core: bt_recv: buf 0x817f664 len 12 type 5
[00:04:06.456,000] <dbg> bt_hci_core: bt_recv: buf 0x817d1e0 len 31 type 1
[00:04:06.456,000] <dbg> bt_hci_core: bt_recv: evt 0x3e, len 29, evt_flags 0x02
[00:04:06.456,000] <dbg> bt_hci_core: hci_rx_thread: buf 0x817f664 type 5 len 12
[00:04:06.456,000] <dbg> bt_ascs: ascs_iso_recv: stream 0x8187968 ep 0x81878d0 l                                                                                                                                                                                        en 0
Incoming audio on stream 0x8186460 len 0

[00:04:06.456,000] <dbg> bt_hci_core: hci_rx_thread: calling fifo_get_wait
[00:04:06.456,000] <dbg> bt_hci_core: hci_rx_thread: buf 0x817d1e0 type 1 len 31
[00:04:06.456,000] <dbg> bt_hci_core: hci_event: event 0x3e
[00:04:06.456,000] <dbg> bt_hci_core: hci_le_meta_event: subevent 0x19
[00:04:06.456,000] <dbg> bt_hci_core: bt_hci_cmd_create: opcode 0x206e param_len                                                                                                                                                                                         13
[00:04:06.456,000] <dbg> bt_hci_core: bt_hci_cmd_create: buf 0x817e7c0
[00:04:06.456,000] <dbg> bt_hci_core: bt_hci_cmd_send_sync: buf 0x817e7c0 opcode                                                                                                                                                                                         0x206e len 16
[00:04:06.456,000] <dbg> bt_hci_core: process_events: count 3
[00:04:06.456,000] <dbg> bt_hci_core: process_events: ev->state 4
[00:04:06.456,000] <dbg> bt_hci_core: send_cmd: calling net_buf_get
[00:04:06.456,000] <dbg> bt_hci_core: send_cmd: calling sem_take_wait
[00:04:06.456,000] <dbg> bt_hci_core: send_cmd: Sending command 0x206e (buf 0x81                                                                                                                                                                                        7e7c0) to driver
[00:04:06.456,000] <dbg> bt_hci_core: bt_send: buf 0x817e7c0 len 16 type 0
[00:04:06.456,000] <dbg> bt_hci_core: process_events: ev->state 1
[00:04:06.456,000] <dbg> bt_hci_core: process_events: ev->state 0
[00:04:06.456,000] <dbg> bt_hci_core: hci_tx_thread: Calling k_poll with 4 event                                                                                                                                                                                        s
[00:04:06.472,000] <dbg> bt_hci_core: bt_recv: buf 0x817f664 len 12 type 5
```

```
< ACL Data TX: Handle 64 flags 0x00 dlen 16                                                                                                      #14702 [hci1] 94131.446182
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016
          Data: 010301010403020200
        ASCS: Audio Stream Sink Endpoint (Handle: 0x0016)
        ASE ID: 1
        Enabling
          CIG: 1
          CIS: 1
          Metadata: 0x03020200
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                           #14703 [hci1] 94131.498789
        Num handles: 1
        Handle: 64
        Count: 2
> ISO Data RX: Handle 96 plen 12                                                                                                                 #14704 [hci1] 94131.594853
        Packet boundary flag: 0x02
        Timestamp flag: 1
        ISO data length: 8
          Timestamp: 1560449589
          Packet seq. nb.: 0
          SDU length: 0
          Packet status flags: 0x00
> HCI Event: LE Meta Event (0x3e) plen 29                                                                                                        #14705 [hci1] 94131.594864
      LE Connected Isochronous Stream Established (0x19)
        Status: Success (0x00)
        Connection Handle: 96
        CIG Synchronization Delay: 2310 us (0x000906)
        CIS Synchronization Delay: 2310 us (0x000906)
        Master to Slave Latency: 2310 us (0x000906)
        Slave to Master Latency: 22310 us (0x005726)
        Master to Slave PHY: 0x02
          LE 2M
        Slave to Master PHY: 0x02
          LE 2M
        Number of Subevents: 3
        Master to Slave Burst Number: 1
        Slave to Master Burst Number: 1
        Master to Slave Flush Timeout: 1
        Slave to Master Flush Timeout: 3
        Master to Slave MTU: 40
        Slave to Master MTU: 10
        ISO Interval: 8
< HCI Command: LE Setup Isochronous Data Path (0x08|0x006e) plen 13                                                                              #14706 [hci1] 94131.604310
        Handle: 96
        Data Path Direction: Input (Host to Controller) (0x00)
        Data Path: HCI (0x00)
        Coding Format: u-law log (0x00)
        Company Codec ID: Ericsson Technology Licensing (0)
        Vendor Codec ID: 0
        Controller Delay: 0 us (0x000000)
        Codec Configuration Length: 0
        Codec Configuration: 
> ISO Data RX: Handle 96 plen 12                                                                                                                 #14707 [hci1] 94131.610777
        Packet boundary flag: 0x02
        Timestamp flag: 1
        ISO data length: 8
          Timestamp: 794016175
> ISO Data RX: Handle 96 plen 12                                                                                                                 #14707 [hci1] 94131.610777
        Packet boundary flag: 0x02
        Timestamp flag: 1
        ISO data length: 8
          Timestamp: 794016175
          Packet seq. nb.: 1
          SDU length: 0
          Packet status flags: 0x00
> ISO Data RX: Handle 96 plen 12                                                                                                                 #14708 [hci1] 94131.610784
        Packet boundary flag: 0x02
        Timestamp flag: 1
        ISO data length: 8
          Timestamp: 3729340653
          Packet seq. nb.: 2
          SDU length: 0
          Packet status flags: 0x00
> HCI Event: Command Complete (0x0e) plen 6                                                                                                      #14709 [hci1] 94131.610785
      LE Setup Isochronous Data Path (0x08|0x006e) ncmd 1
        Status: Success (0x00)
        Connection handle: 96
```

Thalley, 2022-03-02:
Not sure what's causing it. The command complete event comes in a timely manner, so there shouldn't be any timeout. 

The controller used is clearly not working as intended, as
1) It sends HCI ISO data packets before the command complete
2) The first 2 ISO SDUs seems to come at the same time with the same timestamp (this could be a bug in the monitor though). 
3) The timestamp between the first SDU (794016175) and the second SDU (3729340653) does not make sense. 
4) We receive HCI ISO (#14704) before the CIS has been established (#14705)

Can you try to enable `CONFIG_BT_DEBUG_ISO` as well? 

The topmost debug log seems to be cut off in the length (i.e. we never see the command complete event in it). Can you please provide the full log?

Thalley, 2022-03-08:
@MariuszSkamra Can you provide more information?

MariuszSkamra, 2022-03-08:
@Thalley Looks like the data path setup was done improperly. Check the referenced patches. However, this not solves the issue still as Rx & Tx QoS parameters are not NULL even if the CIS was not configured in given direction. This will lead to situation that Host will try to configure the Data Path in direction that is not enabled for a given CIS.
The API description says:
```
/** @brief ISO Channel QoS structure. */
struct bt_iso_chan_qos {
	/** @brief Channel Receiving QoS.
	 *
	 *  Setting NULL disables data path BT_HCI_DATAPATH_DIR_CTLR_TO_HOST.
	 *
	 *  Can only be set for a connected isochronous channel, or a broadcast
	 *  isochronous receiver.
	 */
	struct bt_iso_chan_io_qos	*rx;
	/** @brief Channel Transmission QoS
	 *
	 *  Setting NULL disables data path BT_HCI_DATAPATH_DIR_HOST_TO_CTRL.
	 *
	 *  Can only be set for a connected isochronous channel, or a broadcast
	 *  isochronous transmitter.
	 */
	struct bt_iso_chan_io_qos	*tx;
};
```

But those are non-NULL always, because are initialized in broadcast_sink_ep_init, broadcast_source_ep_init and unicast_client_ep_init

Thalley, 2022-03-08:
> @Thalley Looks like the data path setup was done improperly. Check the referenced patches. However, this not solves the issue still as Rx & Tx QoS parameters are not NULL even if the CIS was not configured in given direction. This will lead to situation that Host will try to configure the Data Path in direction that is not enabled for a given CIS. The API description says:
> 
> ```
> /** @brief ISO Channel QoS structure. */
> struct bt_iso_chan_qos {
> 	/** @brief Channel Receiving QoS.
> 	 *
> 	 *  Setting NULL disables data path BT_HCI_DATAPATH_DIR_CTLR_TO_HOST.
> 	 *
> 	 *  Can only be set for a connected isochronous channel, or a broadcast
> 	 *  isochronous receiver.
> 	 */
> 	struct bt_iso_chan_io_qos	*rx;
> 	/** @brief Channel Transmission QoS
> 	 *
> 	 *  Setting NULL disables data path BT_HCI_DATAPATH_DIR_HOST_TO_CTRL.
> 	 *
> 	 *  Can only be set for a connected isochronous channel, or a broadcast
> 	 *  isochronous transmitter.
> 	 */
> 	struct bt_iso_chan_io_qos	*tx;
> };
> ```
> 
> But those are non-NULL always, because are initialized in broadcast_sink_ep_init, broadcast_source_ep_init and unicast_client_ep_init
> 
> So it requires further fixes.
> 
> @Thalley BTW. Would it be rationale to assign the QoS pointer during ASCS QoS configuration in case of Unicast Client/Server?

I agree, there are some issue there. 
I think they have all been solved by https://github.com/zephyrproject-rtos/zephyr/pull/42964 including not setting the QoS pointers until the BAP QoS operation.

PR #43758 (2022-03-15): Bluetooth: iso: Fix data path configuration
This fixes data path configuration that was done for both directions
even if CIS was unidirectional. To ensure the data path shall be
configured, the QoS SDU is checked.

Fixes: #43190
commit 0481b697 (2022-03-15): Bluetooth: iso: Fix data path configuration

This fixes data path configuration that was done for both directions
even if CIS was unidirectional. To ensure the data path shall be
configured, the QoS SDU is checked.

Fixes: #43190
closed 2022-03-10 by nashif