zephyrproject-rtos/zephyr #43192: lvgl: upgrade LVGL to 8.1 build error

ck-telecom, 2022-02-25:
**Describe the bug**

`[60/413] Building C object modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl__zephyr.dir/D_/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c.obj
FAILED: modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl__zephyr.dir/D_/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c.obj
C:\gnu_arm_embedded\bin\arm-none-eabi-gcc.exe -DKERNEL -DLV_CONF_INCLUDE_SIMPLE=1 -DLV_CONF_PATH=zephyr/lv_conf.h -DNRF52832_XXAA -D_FORTIFY_SOURCE=2 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -ID:/personal/pinetime/zephyr/lib/gui/lvgl -ID:/personal/pinetime/zephyr/include -Izephyr/include/generated -ID:/personal/pinetime/zephyr/soc/arm/nordic_nrf/nrf52 -ID:/personal/pinetime/zephyr/lib/libc/newlib/include -ID:/personal/pinetime/zephyr/soc/arm/nordic_nrf/common/. -ID:/personal/pinetime/zephyr/subsys/bluetooth -ID:/personal/pinetime/zephyr/subsys/settings/include -ID:/personal/pinetime/modules/hal/cmsis/CMSIS/Core/Include -ID:/personal/pinetime/modules/hal/nordic/nrfx -ID:/personal/pinetime/modules/hal/nordic/nrfx/drivers/include -ID:/personal/pinetime/modules/hal/nordic/nrfx/mdk -ID:/personal/pinetime/zephyr/modules/hal_nordic/nrfx/. -ID:/personal/pinetime/modules/lib/gui/lvgl/zephyr/.. -ID:/personal/pinetime/modules/debug/segger/SEGGER -ID:/personal/pinetime/modules/debug/segger/Config -ID:/personal/pinetime/zephyr/modules/segger/. -ID:/personal/pinetime/modules/crypto/tinycrypt/lib/include -ID:/personal/pinetime/pinetime/app/src -ID:/personal/pinetime/pinetime/app/src/bluetooth/service -ID:/personal/pinetime/pinetime/app/src/apps -ID:/personal/pinetime/pinetime/app/src/event/. -Os -imacros D:/personal/pinetime/pinetime/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros D:/personal/pinetime/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=D:/personal/pinetime/pinetime/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=D:/personal/pinetime/zephyr=ZEPHYR_BASE -fmacro-prefix-map=D:/personal/pinetime=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=c99 -MD -MT modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl__zephyr.dir/D_/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c.obj -MF modules\lvgl\CMakeFiles\..__modules__lib__gui__lvgl__zephyr.dir\D_\personal\pinetime\modules\lib\gui\lvgl\src\extra\widgets\win\lv_win.c.obj.d -o modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl__zephyr.dir/D_/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c.obj -c D:/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c
In file included from d:\personal\pinetime\modules\lib\gui\lvgl\lvgl.h:54,
                 from d:\personal\pinetime\modules\lib\gui\lvgl\src\lvgl.h:17,
                 from D:/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.h:16,
                 from D:/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c:9:
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:22:2: error: #error "lv_slider: lv_bar is required. Enable it in lv_conf.h (LV_USE_BAR 1)"
   22 | #error "lv_slider: lv_bar is required. Enable it in lv_conf.h (LV_USE_BAR 1)"
      |  ^~~~~
In file included from d:\personal\pinetime\modules\lib\gui\lvgl\lvgl.h:54,
                 from d:\personal\pinetime\modules\lib\gui\lvgl\src\lvgl.h:17,
                 from D:/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.h:16,
                 from D:/personal/pinetime/modules/lib/gui/lvgl/src/extra/widgets/win/lv_win.c:9:
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:37:29: error: 'LV_BAR_MODE_NORMAL' undeclared here (not in a function); did you mean 'LV_BLEND_MODE_NORMAL'?
   37 |     LV_SLIDER_MODE_NORMAL = LV_BAR_MODE_NORMAL,
      |                             ^~~~~~~~~~~~~~~~~~
      |                             LV_BLEND_MODE_NORMAL
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:38:34: error: 'LV_BAR_MODE_SYMMETRICAL' undeclared here (not in a function)
   38 |     LV_SLIDER_MODE_SYMMETRICAL = LV_BAR_MODE_SYMMETRICAL,
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:39:28: error: 'LV_BAR_MODE_RANGE' undeclared here (not in a function)
   39 |     LV_SLIDER_MODE_RANGE = LV_BAR_MODE_RANGE
      |                            ^~~~~~~~~~~~~~~~~
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:44:5: error: unknown type name 'lv_bar_t'
   44 |     lv_bar_t bar;       /*Add the ancestor's type first*/
      |     ^~~~~~~~
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h: In function 'lv_slider_set_value':
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:86:5: warning: implicit declaration of function 'lv_bar_set_value'; did you mean 'lv_slider_set_value'? [-Wimplicit-function-declaration]
   86 |     lv_bar_set_value(obj, value, anim);
      |     ^~~~~~~~~~~~~~~~
      |     lv_slider_set_value
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h: In function 'lv_slider_set_left_value':
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:97:5: warning: implicit declaration of function 'lv_bar_set_start_value' [-Wimplicit-function-declaration]
   97 |     lv_bar_set_start_value(obj, value, anim);
      |     ^~~~~~~~~~~~~~~~~~~~~~
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h: In function 'lv_slider_set_range':
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:108:5: warning: implicit declaration of function 'lv_bar_set_range'; did you mean 'lv_slider_set_range'? [-Wimplicit-function-declaration]
  108 |     lv_bar_set_range(obj, min, max);
      |     ^~~~~~~~~~~~~~~~
      |     lv_slider_set_range
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h: In function 'lv_slider_set_mode':
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:118:5: warning: implicit declaration of function 'lv_bar_set_mode'; did you mean 'lv_slider_set_mode'? [-Wimplicit-function-declaration]
  118 |     lv_bar_set_mode(obj, (lv_bar_mode_t)mode);
      |     ^~~~~~~~~~~~~~~
      |     lv_slider_set_mode
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:118:27: error: 'lv_bar_mode_t' undeclared (first use in this function); did you mean 'lv_fs_mode_t'?
  118 |     lv_bar_set_mode(obj, (lv_bar_mode_t)mode);
      |                           ^~~~~~~~~~~~~
      |                           lv_fs_mode_t
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:118:27: note: each undeclared identifier is reported only once for each function it appears in
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:118:41: error: expected ')' before 'mode'
  118 |     lv_bar_set_mode(obj, (lv_bar_mode_t)mode);
      |                                         ^~~~
      |                                         )
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h: In function 'lv_slider_get_value':
d:\personal\pinetime\modules\lib\gui\lvgl\src/widgets/lv_slider.h:132:12: warning: implicit declaration of function 'lv_bar_get_value'; did you mean 'lv_slider_get_value'? [-Wimplicit-function-declara
`
Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   pinetime_devkit0
 - What have you tried to diagnose or workaround this issue?
   find out the Kconfig maybe not correct

**To Reproduce**
Steps to reproduce the behavior:
1. enable CONFIG_LVGL=y
               CONFIG_LV_CONF_MINIMAL=y
build and see error

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**

**Additional context**

gmarull, 2022-02-25:
This issue does not include proper steps to reproduce, e.g. which sample to use?

ck-telecom, 2022-02-26:
I tried samples/subsys/display/lvgl with board pinetime_devkit0 is ok, but the project config file is full enabled, for my project with flash size limits so:
- I try to disable some widgets without CONFIG_LV_CONF_MINIMAL=y, but widgets building will cause some build error
- I try to enable CONFIG_LV_CONF_MINIMAL, some widgets build error happens

gmarull, 2022-02-26:
Can you provide a sample where this can be reproduced?

ck-telecom, 2022-02-27:
diff --git a/samples/subsys/display/lvgl/prj.conf b/samples/subsys/display/lvgl/prj.conf
index 8ac5276a02..f937facb52 100644
--- a/samples/subsys/display/lvgl/prj.conf
+++ b/samples/subsys/display/lvgl/prj.conf
@@ -7,9 +7,4 @@ CONFIG_DISPLAY_LOG_LEVEL_ERR=y
 CONFIG_LOG=y

 CONFIG_LVGL=y
-CONFIG_LV_MEM_CUSTOM=y
-CONFIG_LV_USE_LOG=y
-CONFIG_LV_USE_LABEL=y
-CONFIG_LV_USE_BTN=y
-CONFIG_LV_USE_IMG=y
-CONFIG_LV_FONT_MONTSERRAT_14=y
+CONFIG_LV_CONF_MINIMAL=y

make changes above with samples/subsys/display/lvgl then build with the following cmd:
`west build -b pinetime_devkit0 samples/subsys/display/lvgl/`

and check the error, thanks

gmarull, 2022-03-01:
Seems to be a bug (or an assumption) in how LVGL uses Kconfig (`lv_conf_internal.h`):

```c
#ifndef LV_USE_SLIDER
#  ifdef CONFIG_LV_USE_SLIDER
#    define LV_USE_SLIDER CONFIG_LV_USE_SLIDER
#  else
#    define LV_USE_SLIDER     1   /*Requires: lv_bar*/
#  endif
#endif
```

If `CONFIG_LV_USE_SLIDER=n`, there's nothing in `autoconf.h`. Meaning, we'll end up with `LV_USE_SLIDER` defined as 1 (enabled). This also happens to many other widgets (maybe that explains the increase in ROM usage).

LVGL should do instead (assuming it has an equivalent to IS_ENABLED):

```c
#ifndef LV_USE_SLIDER
#   define LV_USE_SLIDER IS_ENABLED(CONFIG_LV_USE_SLIDER)
#endif
```

It seems like a big regression on the Zephyr side.

cc: @kisvegabor @brgl @jfischer-no

kisvegabor, 2022-03-01:
I similar issue was fixed a few days ago. Can you update to the latest LVGL from master? There are no breaking change, so the current master should be a drop in replacement of v8.2.

brgl, 2022-03-01:
@kisvegabor Nope, turned out there are some conflicts between the zephyr fork and ucrrent master, I just tried pulling the tag. I'll spend some time on it later today, I'm busy with other stuff ATM. Let me know if you want to take a look in the meantime.

kisvegabor, 2022-03-01:
You also just replace [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) with this latest version and run it.

brgl, 2022-03-01:
@gmarull Can you assign this to me? I will simply update LVGL directly to v8.2.0 tomorrow.

brgl, 2022-03-02:
@ck-telecom Can you give the following PR a spin: https://github.com/zephyrproject-rtos/zephyr/pull/43365 ?

gmarull, 2022-03-02:
I see the same problem with 8.2. The issue, described in https://github.com/zephyrproject-rtos/zephyr/issues/43192#issuecomment-1055250384, hasn't fixed in 8.2.0 nor in master.

The way Kconfig is used is wrong for boolean options in LVGL:

```c
#ifndef LV_USE_SLIDER
#  ifdef CONFIG_LV_USE_SLIDER
#    define LV_USE_SLIDER CONFIG_LV_USE_SLIDER
#  else
#    define LV_USE_SLIDER     1   /*Requires: lv_bar*/
#  endif
#endif
```

will always result in `#define LV_USE_SLIDER 1` regardless of `CONFIG_LV_USE_SLIDER` because boolean options are not defined when they are set to `n`.

brgl, 2022-03-03:
Right, this is only addressed in master. And even then it looks buggy to me:

```
  50 #ifdef CONFIG_LV_COLOR_DEPTH
  51     #define _LV_KCONFIG_PRESENT
  52 #endif
```

```
1444 #ifndef LV_USE_SLIDER
1445     #ifdef _LV_KCONFIG_PRESENT
1446         #ifdef CONFIG_LV_USE_SLIDER
1447             #define LV_USE_SLIDER CONFIG_LV_USE_SLIDER
1448         #else
1449             #define LV_USE_SLIDER 0
1450         #endif
1451     #else
1452         #define LV_USE_SLIDER     1   /*Requires: lv_bar*/
1453     #endif
1454 #endif
```

gmarull, 2022-03-04:
Seems to fix the problem. I think Zephyr should always use `CONFIG_LV_CONF_MINIMAL=y`, and manually enable other parts as needed. I'm not sure why LVGL doesn't do that, ROM usage almost doubles for nothing in the `samples/subsys/display/lvgl` sample...

kisvegabor, 2022-03-04:
> I'm not sure why LVGL doesn't do that

Our way of thinking was to provide a minimal but in most of the cases usable default config to make it easier to get started with LVGL. Once the user has a running UI, it's easier optimize it by disabling some configs.

PR #43365 (2022-03-07): lvgl: bump supported version to v8.2.0
This updates one Kconfig option name and bumps the manifest to use the v8.2.0 tag from the zephyr fork. 

Fixes #43192
closed 2022-03-07 by MaureenHelm