zephyrproject-rtos/zephyr #43205: stm32 uart console broken since 099850e916ad86e99b3af6821b8c9eb73ba91abf

henrikbrixandersen, 2022-02-25:
**Describe the bug**
The UART console on my `stm32f3_disco` board is no longer functional (doesn't accept keystrokes, sometimes receives gibberish) since commit 099850e916ad86e99b3af6821b8c9eb73ba91abf (as revealed by `git bisect`).

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b stm32f3_disco -s samples/hello_world/ -- -DCONFIG_SHELL=y
2. west flash
3. Connect to the serial console and start typing

**Expected behavior**
Console should accept keystrokes.

**Impact**
Serial console not usable.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA: 3470ace98d4dae4918cf74491c07d74ad69d3559

henrikbrixandersen, 2022-02-25:
Commit 099850e916ad86e99b3af6821b8c9eb73ba91abf  was introduced in #42845.

carlescufi, 2022-02-25:
@npitre no need for hardware, you can use `qemu_cortex_m0` to reproduce.
@nordic-krch FYI

npitre, 2022-02-25:
FYI I reproduced the issue. I'm digging.
Seems to affect modem drivers too.

npitre, 2022-02-26:
The fix is in PR #43226.

PR #43226 (2022-02-28): ring_buffer: make finish methods final
This fixes #43205.
closed 2022-02-28 by carlescufi