zephyrproject-rtos/zephyr #43206: Create resource_table section for stm32mp1 when CONFIG_CMAKE_LINKER_GENERATOR=y

tejlmand, 2022-02-25:
**Is your enhancement proposal related to a problem? Please describe.**
The stm32mp1 SoC uses a custom `linker.ld` that creates an extra section `resource_table`.

https://github.com/zephyrproject-rtos/zephyr/blob/3470ace98d4dae4918cf74491c07d74ad69d3559/soc/arm/st_stm32/stm32mp1/linker.ld#L16-L22

However, when using `CONFIG_CMAKE_LINKER_GENERATOR=y` that section is not created.

**Describe the solution you'd like**
The `resource_table` should be created when using  `CONFIG_CMAKE_LINKER_GENERATOR=y`.

erwango, 2022-11-21:
@arnopo

arnopo, 2022-11-21:
The issue is valid, but I don't know what is behind this config, so if we can use it and if there is a way to declare the` .resource_table` section when the config is enable.

Who could help on this point?