zephyrproject-rtos/zephyr #43235: STM32 platform does not handle large i2c_write() correctly

qin-zou, 2022-02-26:
**Describe the bug**
i2c_write() is generating extra starts at multiple of 255 bytes boundary for STM32 platform (v2).

Please also mention any information which could help others to understand
the problem you're facing:
For example, I have an array of data that's is 512 bytes, and I am writing that data to slave addr 0x5f.
uint8_t d[512];
i2c_write(dev, &data[0], 512, 0x5f); 

I am expecting to see: Start 5f/w d[0] .... d[511] Stop
Instead, I am seeing: start 5f/w d[0] .... d[254] Start 5f/w d[255] ... d[509] Start 5f/w d[510] d[511] Stop

I think the issue is in this section of code ```drivers/i2c/i2c_ll_stm32.c``` function ```i2c_stm32_transfer```
```
        while (num_msgs > 0) {
                uint8_t *next_msg_flags = NULL;

                if (num_msgs > 1) {
                        next = current + 1;
                        next_msg_flags = &(next->flags);
                }
                do {
                        uint32_t temp_len = current->len;
                        uint8_t tmp_msg_flags = current->flags & ~I2C_MSG_RESTART;
                        uint8_t tmp_next_msg_flags = next_msg_flags ?
                                                        *next_msg_flags : 0;

                        if (current->len > 255) {
                                current->len = 255U;
                                current->flags &= ~I2C_MSG_STOP;
                                if (next_msg_flags) {
                                        *next_msg_flags = current->flags &
                                                          ~I2C_MSG_RESTART;
                                }
                        }
                        if ((current->flags & I2C_MSG_RW_MASK) ==
                                                                I2C_MSG_WRITE) {
                                ret = stm32_i2c_msg_write(dev, current,
                                                          next_msg_flags,
                                                          slave);
                        } else {
                                ret = stm32_i2c_msg_read(dev, current,
                                                         next_msg_flags, slave);
                        }

                        if (ret < 0) {
                                goto exit;
                        }
                        if (next_msg_flags) {
                                *next_msg_flags = tmp_next_msg_flags;
                        }
                        current->buf += current->len;
                        current->flags = tmp_msg_flags;
                        current->len = temp_len - current->len;
                } while (current->len > 0);
                current++;
                num_msgs--;
        }
```
With i2c_write(), num_msgs = 1, and next_msg_flags will be NULL.
```
                                ret = stm32_i2c_msg_write(dev, current,
                                                          next_msg_flags,
                                                          slave);
````
Since next_msg_flags is NULL, ```LL_I2C_EnableReloadMode``` will not trigger in ```drivers/i2c/i2c_ll_stm32_v2.c``` function ```msg_init```, which is called by ```stm32_i2c_msg_write```.
And each time ```stm32_i2c_msg_write``` is called, a start condition is generated.

ft, 2022-03-03:
Indeed. I have been chasing this down as well, because writes to a 256-byte page EEPROM caused larger transactions, breaking the writes completely. I came to the same conclusion Qin Zou did.

Here's an attempt at fixing this: https://github.com/ft/zephyr/commit/1fbb8aa40dfea4f541e1ed231c65bbde7d0c1ce0

I've tested it a little bit in the current project I am working on, and it seems fine.

WDYT?

ft, 2022-03-08:
The push I did just now only adjusts a little bit of wording in the commit message.

erwango, 2022-03-09:
@ft If you feel confident about your fix, don't hesitate to propose a PR with it.

PR #43577 (2022-03-15): stm32,i2c: Fix large I2C transactions
The stm32's I2C peripheral has a maximum transmission size. Larger transaction,
that I2C itself allows, can be achieved be using the peripheral's reload-mode.

In order to do that, st's low-lever drivers need to be informed accordingly.
The previous iteration of the code mishandled the next_msg_flags parameter,
causing the issue to manifest itself.

This refactors the inner loop of i2c_stm32_transfer() into its own function.
This passes the message parameter by value in order to be able to mutate its
state while keeping the original datum from the user int [... cut]
commit 1f0b1403 (2022-03-15): stm32,i2c: Fix large I2C transactions

The stm32's I2C peripheral has a maximum transmission size. Larger trans-
action, that I2C itself allows, can be achieved be using the peripheral's
reload-mode.

In order to do that, st's low-lever drivers need to be informed according-
ly. The previous iteration of the code mishandled the next_msg_flags para-
meter, causing the issue to manifest itself.

This refactors the inner loop of i2c_stm32_transfer() into its own func-
tion. This passes the message parameter by value in order to be able to
mutate its state while keeping the original datum from the [... cut]
closed 2022-03-15 by carlescufi