zephyrproject-rtos/zephyr #43245: GitHub settings: Update topics

beriberikix, 2022-02-26:
This repo already has a few [topics](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/classifying-your-repository-with-topics) and at the time of writing they are:
- `iot`
- `real-time`
- `microcontroller`
- `embedded`
- `mcu`
- `rtos`

Notably missing is [`zephyr`](https://github.com/topics/zephyr) & [`zephyr-rtos`](https://github.com/topics/zephyr-rtos). I propose we at least add those at this time as they will help with searching, especially for related repos. 

There's no penalty for adding many relevant topics and we may consider adding additional ones like supported architectures. For example, see PlatformIO's topic list:
[image]

beriberikix, 2022-03-03:
I submitted `zephyr-rtos`, `zephyros` & `zephyr` to GitHub as an official topic and it is now merged . You can see right now that PlatformIO is the first result. It would be beneficial to add these topics to include the official project in search results :)

[image]

henrikbrixandersen, 2022-03-29:
@stephanosio Does this fall under your umbrella?

stephanosio, 2022-03-29:
> @stephanosio Does this fall under your umbrella?

I would like to get some feedback from others before proceeding with it.

cc @bprestonlf @kestewart @nashif @carlescufi @MaureenHelm @galak @gmarull or anyone else highly involved with the project

carlescufi, 2022-06-07:
Done, I updated the topics with a few select ones. Should do for now.

closed 2022-06-07 by carlescufi