zephyrproject-rtos/zephyr #43246: Bluetooth: Host: Ext Adv on native_posix

LingaoM, 2022-02-27:
Use native_posix as Bluetooth Mesh Demo
Use Another board as Controller，attach to host.

Note：EXT_ADV used.

https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/bluetooth/hci/userchan.c#L67
Not Mark discardable flag, cause deadlock.

May be introduce by https://github.com/zephyrproject-rtos/zephyr/pull/41337

@hermabe CC

Thalley, 2022-02-28:
@LingaoM Can you please provide more information, as requested by the bug report template?

How exactly can this be reproduced? Any logs you can share? What commit SHA are you on?

LingaoM, 2022-03-01:
@Thalley  https://github.com/zephyrproject-rtos/zephyr/commit/8b54f08a690e2817be111ed43cd33ec05dfe25b3

LingaoM, 2022-03-01:
`rx.hdr[sizeof(*hdr)] == BT_HCI_EVT_LE_EXT_ADVERTISING_REPORT)` has been removed by https://github.com/zephyrproject-rtos/zephyr/pull/41337 cause deadlock https://github.com/zephyrproject-rtos/zephyr/issues/30955

hermabe, 2022-03-01:
Extended advertising reports cannot be discarded as the host needs to do reassembly of long advertising data. Legacy advertising reports can be discarded because each report is complete.

LingaoM, 2022-03-01:
From the current design, whether discarded or not, there will be problems, so it needs to be reconstructed

Thalley, 2022-03-01:
> From the current design, whether discarded or not, there will be problems, so it needs to be reconstructed

Can you elaborate more on how/why it is causing a deadlock?

LingaoM, 2022-03-01:
The Bluetooth receiving thread may not be able to process broadcast
packets because the system API(bt_hci_cmd_send_sync) is in block state.
If HCI driver is still waiting buffer for adv report, an assertion will
be triggered.

Thalley, 2022-03-01:
> The Bluetooth receiving thread may not be able to process broadcast packets because the system API(bt_hci_cmd_send_sync) is in block state. If HCI driver is still waiting buffer for adv report, an assertion will be triggered.

Ah, so the `bt_hci_cmd_send_sync` can not finish because the event RX buffers are full of adv reports? 

But as @hermabe said, the extended advertising reports cannot be discarded that easily anymore, and doesn't the same issue exist for any other event unsolicited event (including PA reports as well)? 

We current have a system with 2 priorities for events that somewhat handles this. Perhaps we need a 3rd priority, such that 
Highest priority: Disconnect events etc.
Medium priority: Events from commands
Lowest priority: Events from scanning/PA syncs

cvinayak, 2022-04-27:
@LingaoM  and @Thalley have you had any progress here?

Thalley, 2022-04-27:
> @LingaoM and @Thalley have you had any progress here?

I have not looked more into this

PavelVPV, 2022-08-16:
@LingaoM is this still an issue? Could you re-check?

closed 2022-08-16 by LingaoM