zephyrproject-rtos/zephyr #43248: Bluetooth: Mesh: Unable used with ext adv on native_posix

LingaoM, 2022-02-27:
I use the `main` branch based on `ext_adv` In the mesh PTS test under native_posix, it was found that the test failed at all.

[00:00:07.830,000] [0m<dbg> bt_hci_core: hci_le_meta_event: subevent 0x0d[0m
[00:00:07.830,000] [0m<dbg> bt_scan: bt_hci_le_adv_ext_report: Adv number of reports 1[0m
[00:00:07.830,000] [0m<dbg> bt_scan: le_adv_recv: 22:44:7A:94:5E:EF (random) event 0, len 29, rssi -76 dBm[0m
[00:00:07.830,000] [0m<dbg> bt_keys: bt_keys_find_irk: 22:44:7A:94:5E:EF (random)[0m
[00:00:07.830,000] [0m<dbg> bt_hci_core: hci_rx_thread: calling fifo_get_wait[0m
[00:00:07.840,000] [0m<dbg> bt_mesh_pb_adv: prov_retransmit: [0m

There is a certain probability that the test fails. I think the system log has stopped.

LingaoM, 2022-02-27:
**Steps to do:**

Add below config to tests/bluetooth/tester/overlay-native.conf

``` sh
CONFIG_BT_BUF_CMD_TX_SIZE=255
CONFIG_BT_BUF_EVT_RX_SIZE=255
CONFIG_BT_BUF_EVT_DISCARDABLE_SIZE=255
CONFIG_BT_BUF_ACL_TX_SIZE=251
CONFIG_BT_BUF_ACL_RX_SIZE=251
CONFIG_BT_EXT_ADV=y
CONFIG_BT_MESH_ADV_EXT=y
CONFIG_BT_LOG_LEVEL_DBG=y
CONFIG_BT_DEBUG_LOG=y
CONFIG_LOG_BACKEND_NATIVE_POSIX=y
```

`west build -b native_posix zephyr/tests/bluetooth/tester/ -DOVERLAY_CONFIG=overlay-native.conf
`

`python3 ./autoptsclient-zephyr.py "C:\Users\DELL\Desktop\zblue3.0\zblue-3.0\zblue_3.0.pqw6" /home/lingao/zephyrproject/build/zephyr/zephyr.exe -i 10.224.203.10 -l 10.224.203.72 --hci 1 -c MESH/NODE/PROV/BV-07-C
`

LingaoM, 2022-02-28:
One of the fixes:https://github.com/zephyrproject-rtos/zephyr/pull/43253

LingaoM, 2022-02-28:
Link https://github.com/zephyrproject-rtos/zephyr/issues/43246

carlescufi, 2022-03-01:
@LingaoM is this a duplicate of #43246?

LingaoM, 2022-03-05:
> @LingaoM is this a duplicate of #43246?

No, this problem is currently caused by at least two reasons.

carlescufi, 2022-04-06:
@PavelVPV could you try to reproduce this locally?

cvinayak, 2022-04-27:
@PavelVPV did you get to investigate the reasons why HCI Commands did not go through causing the k_sem_take timeout assertion?

PavelVPV, 2022-04-27:
@cvinayak No, I didn't

cvinayak, 2022-04-28:
@PavelVPV When you get a chance to work on this, please enable the HCI Debug logs on the host side to get the HCI opcode that is waiting.

carlescufi, 2022-05-19:
Closing since this is not reproducible. Could be the same issue as #43246 if reproduced.

closed 2022-05-19 by carlescufi