zephyrproject-rtos/zephyr #43249: MBEDTLS_ECP_C not build when MBEDTLS_USE_PSA_CRYPTO

nandojve, 2022-02-27:
**Describe the bug**
Currently version of TFM/PSA not build when enable `MBEDTLS_ECP_C` and `MBEDTLS_USE_PSA_CRYPTO`. I've been testing using `main` repository and `mps2_an521_ns` qemu.

**To Reproduce**
This is a proprietary implementation which get problem when combine MBEDTLS configurations at user-tls.conf.
Steps to reproduce the behavior:

With below configuration Zephyr App build and works properly, the algorithm works as expected and no issues are detected.
```
#define MBEDTLS_ENTROPY_C

#define MBEDTLS_ECP_C
#define MBEDTLS_ECP_DP_SECP256R1_ENABLED
#define MBEDTLS_ECP_DP_SECP384R1_ENABLED
#define MBEDTLS_ECDSA_C

#define MBEDTLS_X509_CSR_WRITE_C
#define MBEDTLS_X509_CREATE_C
#define MBEDTLS_PEM_WRITE_C
#define MBEDTLS_PEM_PARSE_C
#define MBEDTLS_BASE64_C
#define MBEDTLS_SHA512_C
#define MBEDTLS_SHA384_C
#define MBEDTLS_SHA256_C
#define MBEDTLS_OID_C
#define MBEDTLS_ASN1_WRITE_C
#define MBEDTLS_PK_C
#define MBEDTLS_PK_PARSE_C
#define MBEDTLS_PK_WRITE_C
#define MBEDTLS_MD_C
#define MBEDTLS_BIGNUM_C
```

When PSA_CRYPTO is enabled, as below, there are buildings errors.

```diff
+ #define MBEDTLS_USE_PSA_CRYPTO

#define MBEDTLS_ENTROPY_C

#define MBEDTLS_ECP_C
#define MBEDTLS_ECP_DP_SECP256R1_ENABLED
#define MBEDTLS_ECP_DP_SECP384R1_ENABLED
#define MBEDTLS_ECDSA_C

#define MBEDTLS_X509_CSR_WRITE_C
#define MBEDTLS_X509_CREATE_C
#define MBEDTLS_PEM_WRITE_C
#define MBEDTLS_PEM_PARSE_C
#define MBEDTLS_BASE64_C
#define MBEDTLS_SHA512_C
#define MBEDTLS_SHA384_C
#define MBEDTLS_SHA256_C
#define MBEDTLS_OID_C
#define MBEDTLS_ASN1_WRITE_C
#define MBEDTLS_PK_C
#define MBEDTLS_PK_PARSE_C
#define MBEDTLS_PK_WRITE_C
#define MBEDTLS_MD_C
#define MBEDTLS_BIGNUM_C
```

```
[252/263] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf
...
/home/gfbudke/zephyr-sdk-0.13.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.3.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(pk_wrap.c.obj): in function `psa_set_key_type':
/home/gfbudke/workspace/elgin/modules/crypto/mbedtls/include/psa/crypto_struct.h:450: undefined reference to `psa_set_key_domain_parameters'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/gfbudke/.local/bin/cmake --build /home/gfbudke/workspace/elgin/daf-firmware/build --target run
```

**Expected behavior**
The `MBEDTLS_ENTROPY_C` should work with `PSA_CRYPTO`.

**Impact**
Impossible to use feature.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.1
 - Current Main
 - CC: @erwango @theotherjimmy

microbuilder, 2022-02-28:
The next release of TF-M enables MbedTLS 3.1.0, which provides better PSA API integration. There are still a number of issues with the 3.0.0 release used in the current TF-M version we reference in Zephyr.

nandojve, 2022-02-28:
Thank you @microbuilder ,

I wondering if is there any release schedule in mind?

microbuilder, 2022-02-28:
TF-M has three releases a year, and we can only update after a TF-M releases, so there should be an upmerge available in the next two months.

joerchan, 2022-03-01:
They switched to 2 releases a year, next release is in May.
<html>
<body>

Version | Feature Freeze | Release
-- | -- | --
v1.6.0 | 2022-4-1 | 2022-4-15

</body>
</html>

carlescufi, 2022-03-01:
@joerchan do you know if 3.1.0 will fix this isue?

microbuilder, 2022-03-01:
@joerchan We could do an upmerge to TF-M, which would bring us up to MbedTLS 3.1 on the secure side, but I'm not sure where we are with all the cherry-picking post 1.5.0?

microbuilder, 2022-03-02:
@joerchan At some point in the near future, we really need to prioritize using TF-M as the PSA backend. I've heard from Arm that MbedTLS 3.1 should now make it possible to do TLS with the TF-M backend via the PSA APIs, but the MbedTLS management is more in @d3zd3z and @ceolin 's purview. Have you tried connecting the two in the Nordic SDK yet? I don't expect this will easily 'just work'.

joerchan, 2022-03-03:
Yes, we are looking into that specific problem. From what I know MBedTLS 3.1 still needs additional changes in order to have all crypto functionality under PSA API and in the Crypto partition in TF-M. We have something that works with parts of the crypto still in the non-secure domain.

PS: Somewhat confusing that the project and repository is called MbedTLS when the library for the tls stack is called mbedtls, so note the capitalization in the following explanation.

The MBedTLS project exposes the three libraries:
 - mbedtls (TLS library)
 - mbedcrypto (Crypto library)
 - mbedx509 (Certificates library
 
When building with TF-M the mbedtls and mbedx509 libraries will be part of the non-secure image, while mbedcrypto will be included in the secure image TF-M, and is exported through the TF-M crypto partition.
In this configuration the PSA crypto headers has to now be taken from the TF-M project, and not the MBedTLS project.

The explanation for the linker error comes from not using the correct set of PSA crypto headers.
The TLS library has now linked against the PSA crypto headers export by the MBedTLS project, and not TF-M project.

The incorrect header file included from mbedtls library `modules/crypto/mbedtls/include/psa/crypto_struct.h`:
```
static inline void psa_set_key_type( psa_key_attributes_t *attributes,
                                    psa_key_type_t type )
{
    if( attributes->MBEDTLS_PRIVATE(domain_parameters) == NULL )
    {
        /* Common case: quick path */
        attributes->MBEDTLS_PRIVATE(core).MBEDTLS_PRIVATE(type) = type;
    }
    else
    {
        /* Call the bigger function to free the old domain paramteres.
         * Ignore any errors which may arise due to type requiring
         * non-default domain parameters, since this function can't
         * report errors. */
        (void) psa_set_key_domain_parameters( attributes, type, NULL, 0 );
    }
}
```

The correct header to include `modules/tee/tf-m/trusted-firmware-m/interface/include/psa/crypto_struct.h`:
```
static inline void psa_set_key_type(psa_key_attributes_t *attributes,
                                    psa_key_type_t type)
{
    attributes->type = type;
}
```

A possibly workaround would be to make sure that the TF-M include path is first among the includes.
Example:
```
zephyr_library_include_directories(
    $<TARGET_PROPERTY:tfm,TFM_BINARY_DIR>/install/interface/include
  )
 ```
 
Not sure where exactly place this workaround, need more information on how to reproduce the error.

d3zd3z, 2022-03-03:
> PS: Somewhat confusing that the project and repository is called MbedTLS when the library for the tls stack is called mbedtls, so note the capitalization in the following explanation.

As I understand, the official name of the project is "mbed TLS" (note in particular the lowercase 'm' and the space). Sometimes, when writing, they will capitalize the 'm' if it occurs in a place where a word would be capitalized in English (such as the beginning of a sentence). The docs somewhat randomly capitalize the 'm' in other contexts (the probably is a result of an uncapitalized name being unusual in English). (mbed has the same issues)

The run together version 'mbedtls' is commonly used for symbols (within the library), and seems to be how the name is commonly rendered into variable names.

joerchan, 2022-06-22:
@nandojve As far as I can see the problem is still there. Your reproduce point is a bit vague, would you be able to clarify, perhaps by providing a change to a zephyr sample that would lead to the error?

d3zd3z, 2022-06-24:
I have been unable to build with `MBEDTLS_USE_PSA_CRYPTO` even after the 3.1.0 release. I get a small number of conflicting symbols. There is a bit deeper of a question as to where the operations actually get implemented, and should all of them be handled by the PSA calls into the trusted side.

nandojve, 2022-10-24:
I updated internal code to latest Zephyr version last week. I hope I can re-test soon and give a proper feedback.

nandojve, 2022-10-31:
I can attest that this problem still present.

When enabling MBEDTLS_USE_PSA_CRYPTO, as described on the initial issue, I got the exact same problem at building:

```
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/mbedtls/libmodules__mbedtls.a(pk_wrap.c.obj): in function `psa_set_key_type':
$HOME/workspace/elgin/modules/crypto/mbedtls/include/psa/crypto_struct.h:464: undefined reference to `psa_set_key_domain_parameters'
```

So, after enabling the MBEDTLS_PSA_CRYPTO_CLIENT to provide the missing `psa_set_key_domain_parameters` function I got this:

```
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: modules/trusted-firmware-m/libtfm_api.a(tfm_crypto_ipc_api.c.obj): in function `psa_reset_key_attributes':
$HOME/workspace/elgin/daf-firmware/build/tfm/install/interface/src/tfm_crypto_ipc_api.c:132: multiple definition of `psa_reset_key_attributes'; modules/mbedtls/libmodules__mbedtls.a(psa_crypto_client.c.obj):$HOME/workspace/elgin/modules/crypto/mbedtls/library/psa_crypto_client.c:34: first defined here
```

On the first try to identify better the problem, I just applied below patch:

```diff
diff --git a/library/psa_crypto_client.c b/library/psa_crypto_client.c
index 629feb7df..7a6f0e429 100644
--- a/library/psa_crypto_client.c
+++ b/library/psa_crypto_client.c
@@ -30,11 +30,13 @@
 #define mbedtls_free   free
 #endif
 
+#if 0
 void psa_reset_key_attributes( psa_key_attributes_t *attributes )
 {
     mbedtls_free( attributes->domain_parameters );
     memset( attributes, 0, sizeof( *attributes ) );
 }
+#endif
 
 psa_status_t psa_set_key_domain_parameters( psa_key_attributes_t *attributes,
                                             psa_key_type_t type,
```

This generate more problems, as below:
```
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld: secure_fw/partitions/crypto/libtfm_psa_rot_partition_crypto.a(crypto_asymmetric.o): in function `tfm_crypto_asymmetric_encrypt':
$HOME/workspace/elgin/modules/tee/tf-m/trusted-firmware-m/secure_fw/partitions/crypto/crypto_asymmetric.c:204: undefined reference to `mbedcrypto__psa_reset_key_attributes'
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld: secure_fw/partitions/crypto/libtfm_psa_rot_partition_crypto.a(crypto_key_management.o): in function `tfm_crypto_reset_key_attributes':
$HOME/workspace/elgin/modules/tee/tf-m/trusted-firmware-m/secure_fw/partitions/crypto/crypto_key_management.c:244: undefined reference to `mbedcrypto__psa_reset_key_attributes'
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld: secure_fw/partitions/crypto/mbedcrypto/library/libmbedcrypto.a(psa_crypto.o): in function `psa_key_derivation_start_hmac':
$HOME/workspace/elgin/modules/crypto/mbedtls/library/psa_crypto.c:4382: undefined reference to `mbedcrypto__psa_reset_key_attributes'
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld: secure_fw/partitions/crypto/mbedcrypto/library/libmbedcrypto.a(psa_crypto.o): in function `mbedcrypto__psa_get_key_attributes':
$HOME/workspace/elgin/modules/crypto/mbedtls/library/psa_crypto.c:1183: undefined reference to `mbedcrypto__psa_reset_key_attributes'
$HOME/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld: $HOME/workspace/elgin/modules/crypto/mbedtls/library/psa_crypto.c:1235: undefined reference to `mbedcrypto__psa_reset_key_attributes'
```

So, I rollback and try same approach here: $HOME/workspace/elgin/daf-firmware/build/tfm/install/interface/src/tfm_crypto_ipc_api.c:132

```diff
+#if 0
void psa_reset_key_attributes(psa_key_attributes_t *attributes)
{
    struct tfm_crypto_pack_iovec iov = {
        .srv_id = TFM_CRYPTO_RESET_KEY_ATTRIBUTES_SID,
    };
    psa_invec in_vec[] = {
        {.base = &iov, .len = sizeof(struct tfm_crypto_pack_iovec)},
    };
    psa_outvec out_vec[] = {
        {.base = attributes, .len = sizeof(psa_key_attributes_t)},
    };

    (void)API_DISPATCH(tfm_crypto_reset_key_attributes,
                       TFM_CRYPTO_RESET_KEY_ATTRIBUTES);
    return;
}
+#endif
```

Once I dropped this `auto-generated method` I could link the code. The application ran but I'm currently not accessing PSA methods to confirm if the end result is correct.

I hope this could give few more insights to solve the issue.

mimok, 2022-11-16:
@nandojve: I'm facing the exact same issue. I believe you approach is the right one. Indeed to import a key using `psa_import_key()` you have to set the key attributes from NSPE (i.e. normal word). All these data are stored in NSPE memory before to be sent to the SPE during key import. So, in my opinion, functions supporting attributes handling have to be placed in NSPE memory.

`psa_reset_key_attributes()` function placed in tfm_crypto_ipc_api.c and tfm_crypto_func_api.c are intended to be used to forward attribute reset to SPE which does not make sense if key attributes are located in NSPE memory. I believe these functions are remnants of an old code (well, actually, this is an assumption).

In my case, by doing what you propose, I successfully established a TLS connexion using an ecdsa key with all crypto backed up by TF-M (To be honest, TLS connexion crashes after handshake but I believe this is not related to the current issue, I'm investigating the problem)

microbuilder, 2022-11-16:
@mimok I'm seeing the same thing with the TLS handshake and ECDSA. Handshake seems to complete looming at wireshark, but TLS goes off into the weeds shortly after. I didn't have this issue with Zephyr 3.1 and ECDSA. No solution yet, but wanted to confirm I see the same with a similar setup.

mimok, 2022-11-16:
@microbuilder Doubling the stack size (from 4096 to 8096) solves the problem (well 8096 bytes might be to much but for now I don't care). So this is only a stack overflow problem (at least on my side).

microbuilder, 2022-11-16:
> @microbuilder Doubling the stack size (from 4096 to 8096) solves the problem (well 8096 bytes might be to much but for now I don't care). So this is only a stack overflow problem (at least on my side).

This really feels like a stack overflow or memory issue, but which stack or memory pool are you increasing? `CONFIG_MAIN_STACK_SIZE`?

mimok, 2022-11-16:
I increased `CONFIG_MAIN_STACK_SIZE` from 4096 to 8096 after reading a post saying that a `usage_fault` in `z_swap()` could be solved by increasing the stack size.

joerchan, 2022-12-01:
@mimok You sure the issue is not coming from inline functions in the PSA headers? If you compile in the wrong variant of the headers (i.e MBedtLS PSA header instead of TF-M PSA header) then the inline functions will be doing the wrong things.
The issue is that the different PSA headers are not compatible, and possibly not always including the correct set of headers based on the configuration,.

mimok, 2022-12-01:
@joerchan No header problem here i think. 

When trying to use opaque keys with TLS and PSA crypto, `psa_set_key_domain_parameters()` appears not being defined. This function is defined in `psa_crypto_client.c` when macro `MBEDTLS_PSA_CRYPTO_CLIENT` is defined. The same macro also enable the definition of `psa_reset_key_attributes()` which therefore collides during linkage with the definition of the same function in TF-M PSA interface (from either `tfm_crypto_ipc_api.c` or `tfm_crypto_func_api.c`). Considering that key attributes are stored in NSPE memory before key import, handling them in SPE does not really make sense. 

In addition, `psa_set_key_domain_parameters()` and `psa_reset_key_attributes()` are defined together in  `psa_crypto_client.c` so I suppose that the right `psa_reset_key_attributes()` function that has to be used when using `psa_set_key_domain_parameters()` is actually the function coming from the same file.

However, TF-M integration is a bit cumbersome so I'm not 100% confident in my explanations.

mimok, 2022-12-01:
last thing, not sure what would be the meaning of reseting key attributes owned by SPE from NSPE and TF-M has its own `psa_reset_key_attributes()` function defined in `tfm_crypto_secure_api.c`.

nandojve, 2023-01-08:
Folks,

This issue has been open for almost one year. Now Zephyr is moving to LTS and there is no possibility to use crypto feature with above configuration. Is there any consideration to make this fixed for LTS?

CC: @erwango

microbuilder, 2023-01-08:
> Folks
> 
> 
> 
> This issue has been open for almost one year. Now Zephyr is moving to LTS and there is no possibility to use crypto feature with above configuration. Is there any consideration to make this fixed for LTS?
> 
> 
> 
> CC: @erwango

The main issue here is TF-M, not Zephyr, and TF-M exposing a different set of headers, which is still the case as of MbedTLS 3.3, where there were some tests in another PR last week. We'll update to TF-M 1.7.0 and MbedTLS 3.2.1 next week, but the PSA APIs between the two still aren't fully compatible on first glance.

If you aren't using TF-M, however, you should be free to use the PSA APIs.

microbuilder, 2023-01-09:
@nandojve Related issue/comment here: https://github.com/zephyrproject-rtos/zephyr/pull/53543#issuecomment-1373565013 ... I'll look at this this week, though, trying to update to 1.7.0 and see if the TF-M 1.7.0 and MbedTLS 3.2.1 combination makes using TF-M as the PSA backend from Zephyr possible at last, or not.

nandojve, 2023-01-09:
Hi @microbuilder ,

Thank you so much for your fast assistance. I'll cross fingers here : )

mimok, 2023-03-11:
Is there any news on this topic ? I have seen TF-M crypto sample has been removed in 3.3.0, is it related to this issue or something similar ?

mimok, 2023-03-11:
So if I understand well this is not just a zephyr os problem ? Anybody using both TF-M and mbedTLS would face this problem, right ?

microbuilder, 2023-03-11:
> So if I understand well this is not just a zephyr os problem ? Anybody using both TF-M and mbedTLS would face this problem, right ?

There are multiple issues, some of which are related to MbedTLS support for using PSA APIs and redirecting those to TF-M, as well as differences in headers between MbedTLS and TF-M. The main issue with psa_crypto is changes that are required to the Zephyr CMAKE integration, though. The MbedTLS team is looking at improving the ability to integrate MbedTLS and TF-M, and we're trying to sort out the CMAKE issues ourselves.

microbuilder, 2023-03-11:
> So if I understand well this is not just a zephyr os problem ? Anybody using both TF-M and mbedTLS would face this problem, right ?

Some related PRs and change requests, for example:

- https://github.com/Mbed-TLS/mbedtls/pull/7151
- https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/19806

mimok, 2023-03-13:
Thank you for your feedback. I'm willing to help if this can make any difference.

microbuilder, 2023-06-05:
@nandojve The changes introduced in https://github.com/zephyrproject-rtos/zephyr/pull/58023 should go a long way to enabling TF-M as a back end for TLS. This now causes the x.509 and TLS parts of MbedTLS to be linked against NS, and for Crypto to be linked against S, with the PSA Crypto calls from TLS on NS-side calling into TF-M on the S side.

Some of the issues around specific config flags in MbedTLS and TF-M are being sorted out, and there is still work to do on PSA API implementations, but #58023 should be a large step towards enabling the NS image to make use of the S image as a crypto back-end, and hide some of the config flag details required to make everything work together.

microbuilder, 2023-06-05:
cc @mimok ... see the above comment as well.

trustngotech, 2023-06-07:
@microbuilder Thank you for the feedback! This is a major improvement in my opinion as i'm planning to use this feature in a commercial demo.

microbuilder, 2023-06-07:
> @microbuilder Thank you for the feedback! This is a major improvement in my opinion as i'm planning to use this feature in a commercial demo.

It needs more testing, and I'm sure there are problems to sort out, but please raise issues (even better, PRs) in the upstream TF-M or here in Zephyr if you find any issues. Agree it's a long overdue step forward, though.

commit 3398c987 (2024-01-17): modules: mbedtls: Use TF-M PSA API headers

Use TF-M PSA API headers when compiling with TF-M enabled.

Fixes: #43249
closed 2024-01-17 by carlescufi