zephyrproject-rtos/zephyr #43271: tests: acrn_ehl_crb:  tests/arch/x86/info failed

enjiamai, 2022-02-28:
**Describe the bug**
The  tests/arch/x86/info test failed in acrn_ehl_crb start from commit e510dba95ec2b0a55658698d9d0369f5403e7874 <tests: x86/info: only tests PC compatible platforms> because this commit enable this testing for acrn_ehl_crb also.
Before this commit this test was skipped, it is just enabled. So need to check why this test failed on this board.  

**To Reproduce**
Steps to reproduce the behavior:
1. export LANG=en_US.UTF-8
2. twister -p acrn_ehl_crb --device-testing --device-serial-pty="/opt/remotehw/acrn-test-pty.exp,ehlsku11-2" --west-flash="/opt/remotehw/remotehw-x86-acrn.sh,ehlsku11-2" -T tests/arch/x86/info/ -vv
or 
2. west build -p always -b acrn_ehl_crb tests/arch/x86/info/
3. west flash /opt/remotehw/remotehw-x86-acrn.sh ehlsku11-2
4. In another terminal , execute "remotehw-ehlsku11-2-get-console"
5. See the output log.

**Expected behavior**
The tests/arch/x86/info testsuite can pass the test.

**Impact**
No serious impact.

**Logs and console output**
```
DEBUG   - DEVICE: ACRN:\>vm_console 0
DEBUG   - DEVICE:
DEBUG   - DEVICE: ----- Entering VM 0 Shell -----
DEBUG   - DEVICE: *** Booting Zephyr OS build v2.7.99-4497-gc0dd594d4d3d  ***
DEBUG   - Timed out while monitoring serial output on acrn_ehl_crb
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.2
 - Commit SHA or Version : c0dd594d4d3d10e74d9dc4bb5d3d29caa6e4f78f

smrtos, 2022-03-08:
The test runs with 2 CPUs. CPU 0 starts CPU 1 by sending IPI via CPU 1's local APIC ID. The local APIC ID is obtained through the ACPI (CONFIG_ACPI=y). But in below line 140, CPU 1's local APIC ID is 4 at run time. When changing it to 2, the test can pass.
Will discuss with ACRN about it.

[image]

smrtos, 2022-03-18:
As discussed with ACRN team, I need to provide the detailed info about the ACRN environment we are using. Checking with test environment maintainer.

smrtos, 2022-03-24:
In code [here](https://github.com/zephyrproject-rtos/zephyr/blob/890f6199367953e52f2ffcf10420c98d9ce3a320/arch/x86/core/intel64/locore.S#L149):

[image]

The `x86_cpu_loapics[]` holds the local apic IDs of the usable CPU expected by Zephyr. 
Such expectation comes from the ACPI table when `CONFIG_ACPI=y` in this scenario. 

line 145-147: current AP's local APIC id is stored into eax.

line 149: ebp holds the address of `x86_cpuboot[]`.
line 150: ebx is used as loop index through the `x86_cpuboot[]` and cleared to 0.
line 151-152: if ebx reaches the max configured cpu number CONFIG_MP_NUM_CPUS, it means current AP's local apic ID doesn't match any of the expectation in `x86_cpu_loapics[]`.  **Current AP will jump to dead loop at `unknown_loapic_id`.**

line 153-154: if find a matching local apic id, jump to go64 and proceed with initialization.
line 155: increase the loop index ebx
line 156: move ebp to the next entry in `x86_cpuboot[]`

If a matching local apic id found, the `z_x86_cpu_init()` will be [called](https://github.com/zephyrproject-rtos/zephyr/blob/890f6199367953e52f2ffcf10420c98d9ce3a320/arch/x86/core/intel64/locore.S#L264) and the [cpuboot->ready will be increased.](https://github.com/zephyrproject-rtos/zephyr/blob/890f6199367953e52f2ffcf10420c98d9ce3a320/arch/x86/core/intel64/cpu.c#L198). 

[image]

Thus the BSP will not hang at below location.

[image]

Zephyr always start the comparison for local apic id in the order of logical cpu number 0, 1, etc. (above line 150 and 155).
The root cause of this issue is the local apic ID obtained from ACPI for logical cpu 1 doesn't match the one obtained from the running AP's APIC ID register (above line 145).

smrtos, 2022-03-24:
Confirmed with ACRN, the board info XML used to build the ACRN binary and ACPI data doesn't match the board on which the test runs, which leads to the local APIC id arrangement mismatch. 

The fix is to re-capture the board info with `boardinspection.py` on the test board and rebuild the ACRN binary and ACPI data. No zephyr change needed.

enjiamai, 2022-03-24:
The is the environment set up issue, not a  bug. So close it.

closed 2022-03-24 by enjiamai