zephyrproject-rtos/zephyr #43276: tests: up_squared:  testsuite tests/kernel/sched/deadline/ failed

enjiamai, 2022-02-28:
**Describe the bug**
The testsuite tests/kernel/sched/deadline/ failed on up_squared board. After bisect it start to failed at commit: b0458201ccd272f4d7e5b7fddbca952f77f67a3c <arch/riscv: Do not use irq_lock() on arch_irq_offload>

This commit remove the spinlock on arch_irq_offload(), this should only effect the risc-v,  but  actually the tests/kernel/sched/deadline/  failed due to this commit. 

```
diff --git a/arch/riscv/core/irq_offload.c b/arch/riscv/core/irq_offload.c
index 22b22431c4..a6936f944b 100644
--- a/arch/riscv/core/irq_offload.c
+++ b/arch/riscv/core/irq_offload.c
@@ -33,13 +33,8 @@ void z_irq_do_offload(void)

 void arch_irq_offload(irq_offload_routine_t routine, const void *parameter)
 {
-       unsigned int key;
-
-       key = irq_lock();
        _offload_routine = routine;
        offload_param = parameter;

        __asm__ volatile ("ecall");
-
-       irq_unlock(key);
 }
```

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -p up_squared --device-testing --device-serial /dev/ttyUSB1 -T tests/kernel/sched/deadline/ --west-flash="$HOME/up_squared_efi.sh" -j 1 -vv
or
2. west build -b up_squared tests/kernel/sched/deadline/
3. west flash $HOME/up_squared_efi.sh
4. See error log output

**Expected behavior**
This testsuite should pass the test.

**Impact**
This might have some impact on scheduler, not sure now.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.0.0-483-g145bcd7ab96e  ***
Running test suite suite_deadline
===================================================================
START - test_deadline
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/deadline/src/main.c:103: test_deadline: (d0 <= d1 is false)
threads ran in wrong order
FAIL - test_deadline in 0.115 seconds
===================================================================
START - test_yield
E: Page fault at address (nil) (error code 0x2)
E: Linear address not present in page tables
E: Access violation: supervisor thread not allowed to write
E: PTE: not present
E: RAX: 0x000000000010e850 RBX: 0x000000000010fa70 RCX: 0x0000000000988d9f RDX: 0x0000000000000000
E: RSI: 0x000000000010d0b0 RDI: 0x00000000005984c0 RBP: 0x00000000005b3350 RSP: 0x00000000005b3330
E:  R8: 0x0000000000000001  R9: 0x000000000010a7e4 R10: 0x0000000000000000 R11: 0x0000000000000000
E: R12: 0x0000000000000046 R13: 0x00000000000003e8 R14: 0x0000000000000001 R15: 0x000000000010a863
E: RSP: 0x00000000005b3330 RFLAGS: 0x0000000000010016 CS: 0x0018 CR3: 0x00000000007c1000
E: RIP: 0x0000000000108414
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x10fa40 (test_yield)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.13.2
 - Commit SHA or Version:  145bcd7ab96e

andyross, 2022-03-01:
That bisect can't be right.  Much more likely this is due to the changes in #42806.  Can you try reverting those?  Just set IRQ_OFFSET_NESTED=n in the prj.conf and you should get the earlier behavior.

andyross, 2022-03-01:
Actually, looking closer, that test doesn't use irq_offload() at all.  So... something else is wrong.

closed 2022-03-08 by enjiamai