zephyrproject-rtos/zephyr #43277: usb/dfu: upgrade request is not called while used from mcubot, update dosen't happen

nvlsianpu, 2022-02-28:
**Describe the bug**
When zephyr dfu class is used (along with MCUboot's dual-bank mode), it downloads the update image to the image-1 slot.
Once download is finished the class driver should call boot_request_upgrade(), but it doesn't which causes the problem.

Zephyr's usb dfu class implementation works fine as an application.

Need to find wheter issue is caused by mcuboot or zephyr
 - ...

**To Reproduce**
build mcuboot with either CONFIG_BOOT_USB_DFU_WAIT=y or  CONFIG_BOOT_USB_DFU_GPIO=y. Test along with dfu-util, similar as for USB DFU class zephyr-rtos sample (use --alt 1). It can be observed that update doesn't happen.

**Expected behavior**

**Impact**

**Logs and console output**

**Environment (please complete the following information):**
 - zephyr 3.0.0

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

carlescufi, 2022-03-01:
cc @jfischer-no

nvlsianpu, 2022-03-02:
This possible bug was recognized faulty. Root reason for not working usb dfu class along with mcuboot was missing of the image in slot-0.

closed 2022-03-02 by nvlsianpu