zephyrproject-rtos/zephyr #43285: nRF5x System Off demo fails to put the nRF52840DK into system off

gte1, 2022-02-28:
Using zephyr-v3.0.0 release

Please also mention any information which could help others to understand
the problem you're facing:
 - Board nrf52840dk_nrf52840

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nrf52840dk_nrf52840 samples\boards\nrf\system_off
2. west flash

**Expected behavior**
System off entered

**Impact**
Not possible to enter system off

**Logs and console output**

*** Booting Zephyr OS build zephyr-v3.0.0 ***

nrf52840dk_nrf52840 system off demo
Retained data not supported
Busy-wait 2 s
Busy-wait 2 s with UART off
Sleep 2 s
Sleep 2 s with UART off
Entering system off; press BUTTON1 to restart
ERROR: System off failed

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: GNU ARM Embedded 9-2019-q4-major
 -  zephyr-v3.0.0

**Additional context**
Also reported to Nordic for nRFConnect SDK 1.9 in https://devzone.nordicsemi.com/f/nordic-q-a/85097/nrfconnect-sdk-1-9-system-off-sample-fails-to-put-the-nrf52840dk-into-system-off

gmarull, 2022-03-01:
Can you check if https://github.com/zephyrproject-rtos/zephyr/pull/43187 fixes the issue?

gte1, 2022-03-01:
Yes #43187 seems to fix the issue

*** Booting Zephyr OS build zephyr-v3.0.0  ***

nrf52840dk_nrf52840 system off demo
Retained data not supported
Busy-wait 2 s
Busy-wait 2 s with UART off
Sleep 2 s
Sleep 2 s with UART off
Entering system off; press BUTTON1 to restart

PR #43187 (2022-03-01): samples: nrf: system_off: Fix force state usage
The sample cannot spin in an infinity loop because the idle thread will
not run and consequently the forced state will not be used.

Fixes #43285