zephyrproject-rtos/zephyr #43286: Zephyr 3.1 Release Checklist

dkalowsk, 2022-02-28:
Looking for a sign off on the v3.1.0 release notes- [ ] Major enhancements:
   - See https://github.com/zephyrproject-rtos/zephyr/projects/9

- [ ] Pre RC1 Steps
   - [ ] Create v3.1.1 milestone so that features that don't make it to the release can be assigned a milestone
   - [ ] Verify that all external components and external dependencies are up to date, for example 
      - mbedTLS
      - tinycrypt
      - FatFS
   - [ ] Check known vulnerabilities, and fix exploitable vulnerabilities or verify them as un-exploitable (Security working group).

See https://docs.zephyrproject.org/latest/development_process/release_process.html#tagging for tagging details

--------
- [ ] Release Notes WIP
  - [ ] Make sure to tell everybody to use the **past tense** (i.e. "X has been added" or "X is now Y")
  - [ ] Create draft `doc/releases/release-notes-3.1.rst` with feature summary from https://www.zephyrproject.org/developers/#releases-overview #22923
  - [ ] Update `doc/releases/release-notes-3.1.rst` detailed sections (relative to 3.0.0 release) as important issues are addressed (subsystem owners). See #25869
--------
- [ ] Finalize Release Notes
  - [ ] Add list of GitHub issues (spell check issue list and fix in GitHub as needed) using `scripts/release/list_issues.py`
  - [ ] Review `doc/releases/release-notes-3.1.rst` overview summary and details sections (see Sign-off below)
    - [ ] Request summary details from code owners
- [ ] Update `doc/reference/api/overview.rst`
- [ ] Update `doc/conf.py` (add version to pick list)
- [ ] Update `doc/LICENSING.rst` for new components that do not have a license
--------
- [ ]  Final Release Notes Sign-off
  - Subsystems
    - [ ] #42974
    - [ ] #42975
    - [ ] #42976
    - [ ] #42977
    - [ ] #42978
    - [ ] #42979
    - [ ] #42980
    - [ ] #42981
    - [ ] #42982
  - Architectures
    - [ ] #42983
    - [ ] #42984
    - [ ] #42985
    - [ ] #42986
    - [ ] Xtensa, NIOS-II - @nashif, @dcpleung 
  - Board an Soc support - many
  - [ ] #42987
--------
- [ ] Update version in VERSION
- [ ] Tag both v3.1.0 and zephyr-v3.1.0
- [ ] Tag v3.1.0 in [example-application](https://github.com/zephyrproject-rtos/example-application)
- [ ] Create release and add notes on https://github.com/zephyrproject-rtos/zephyr/releases. Make sure you use the `zephyr-v3.1.0` tag so that the `.zip` that GitHub offers has a reasonable name. Add the release notes to the GitHub release.
--------
- [ ] After Release
  - [ ] Update patchlevel in Makefile (PATCHLEVEL = 99)
  - [ ] Update https://www.zephyrproject.org/developer-resources/#current-release (email Brett)
  - [ ] Create `v3.1-branch` branch for 3.1.x releases (from v3.1.0 tag, don't include the switch to patchlevel 99)
  - [ ] Create `backport v3.1-branch` label
  - [ ] Merge all post-release PRs in https://github.com/zephyrproject-rtos/zephyr/projects/27
  - [ ] Move all PRs/Issues in the v3.1.0 milestone to the next milestone
  - [ ] Close v3.1.0 milestone 
  - [ ] Email devel@lists.zephyrproject.org & announce@lists.zephyrproject.org lists about release and announce the opening of the merge window
  - [ ] Create checklist issue for next release
  - [ ] Add a comment to this issue linking to the next release checklist issue
  - [ ] Marketing Blog for announcing release (https://www.zephyrproject.org/blog)
  - [ ] Update https://www.zephyrproject.org with latest stats from GitHub (email Brett)
  - [ ] Store the number of currently open PRs to keep track of post-release PR overhead over time

dkalowsk, 2022-02-28:
Link back to the previous release issue: https://github.com/zephyrproject-rtos/zephyr/issues/39587

carlescufi, 2022-06-06:
There were ~560 Pull Requests open when the release was cut. After merging all the ones that were ready after release, we got down to ~490.