zephyrproject-rtos/zephyr #43306: sam_e70b_xplained: the platform will be not normal after running test case tests/subsys/usb/desc_sections/

Zhaoningx, 2022-03-01:
Describe the bug
Although the test case `tests/subsys/usb/desc_sections/` is successful to test, after running this, the platform won't work and never get any log from the port and will show me `time out` error. I revisited these two days' commits which got merged, I find the commit [`1e747eca0d1a`](https://github.com/zephyrproject-rtos/zephyr/commit/1e747eca0d1a063c64438d18ca7e556a88ccc648) is some impacts on the platform.

To Reproduce
first step,
twister --device-testing --device-serial /dev/ttyACM0 -p sam_e70b_xplained  -T tests/subsys/usb/desc_sections/  -v
then,
twister --device-testing --device-serial /dev/ttyACM0 -p sam_e70b_xplained  -T tests/kernel/queue/  -v

see error:
no output.

Environment:
zephyr SDK: 13.2

enjiamai, 2022-03-01:
After this PR https://github.com/zephyrproject-rtos/zephyr/pull/43101 merged, the flashing will failed after executing testsuite tests/subsys/usb/desc_sections/,  this cause all the following testing failed in our daily testing automation. Could you please take a look on this? @pdgendt , thank you!

pdgendt, 2022-03-01:
Hi @enjiamai, I’m currently on vacation without a development setup. I will try to fix it next week.

enjiamai, 2022-03-01:
Ok, thanks for letting us know this! @pdgendt . We might skip this test until your fix on this.

Zhaoningx, 2022-03-02:
> After this PR #43101 merged, the flashing will failed after executing testsuite tests/subsys/usb/desc_sections/, this cause all the following testing failed in our daily testing automation. Could you please take a look on this? @pdgendt , thank you!

Hi @pdgendt @enjiamai, **update**: not only `testsuite tests/subsys/usb/desc_sections/`, all testsuite tests/subsys/usb will cause the rest of test cases failed.

pdgendt, 2022-03-07:
Hey @enjiamai, as I don't have the hardware to confirm, I'll probably need you to verify on an actual device.

Could it be that `NRST` is pulled down on your board? In [the datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/SAME70-xpld.pdf) there is for example header J506 which could have that effect, as far as I can see.
By default the `URSTEN` is set to 1, so an external user reset would halt the chip, once the peripheral is enabled (as is done in the driver).

enjiamai, 2022-03-07:
Got it, @pdgendt , I can give it a try. Will feedback you soon later, thanks~

Zhaoningx, 2022-03-08:
Hi @pdgendt, I tested the pin `RST` in J506 when the board is normal, it showed me `3.2V`, and when running test cases `tests/derives/hwinfo/api`, the pin `RST` jumped temporarily to `2.5V`, and then back to `3.2V`, and the board can't be flashed, it seems the pin `RST`  never be pulled down when running some test cases, I'm not familiar with it, so any comments here.

Zhaoningx, 2022-03-08:
I looked into the test cases `tests/drivers/hwinfo/api`, I found that if I enable the configure `CONFIG_HWINFO =y` can reproduce this issue under the condition I skipped all test cases of `tests/drivers/hwinfo/api`.

pdgendt, 2022-03-08:
@Zhaoningx  The default state of the `URSTEN` after reset is 1, so maybe this should be matched with the `dts` binding set for `user-nrst`.

Could you try by adding the following to `boards/arm/sam_e70_xplained/sam_e70b_xplained.dts`
```dts
&rstc {
	user-nrst;
};
```

If this would help, we can add it to `dts/arm/atmel/same70.dtsi`.

Zhaoningx, 2022-03-09:
Hi @pdgendt, it works well after adding it. Thanks for fixing this.

enjiamai, 2022-03-09:
> Hi @pdgendt, it works well after adding it. Thanks for fixing this.

Hi @pdgendt , we verified it and no negative impacts on all other tests. Could you please submit this fix? Thanks!

yerabolu, 2022-03-15:
@pdgendt : Could you please submit the fix? Thank You!!

PR #43844 (2022-03-17): dts: arm: atmel: Enable user-nrst by default
The datasheet for both sam4s/same70 specifies that the NRST pin
is an input after reset, used as a user reset.

Add the user-nrst property to match the default.

Fixes #43306
commit be2bd814 (2022-03-17): dts: arm: atmel: Enable user-nrst by default

The datasheet for both sam4s/same70 specifies that the NRST pin
is an input after reset, used as a user reset.

Add the user-nrst property to match the default.

Fixes #43306
closed 2022-03-17 by carlescufi