zephyrproject-rtos/zephyr #43308: driver: serial: stm32: uart will lost data when use dma mode[async mode]

Jeepgoing, 2022-03-01:
**Describe the bug**
User data will be lost when dma buffer is full

e.g.
DMA buffer length is 256 bytes, my package length is 10 bytes, uart driver will lost 4 bytes
when received 26 packages.

Platform is STM32G473QE

erwango, 2022-03-01:
@Jeepgoing Thanks for submitting this issue. To ease the investigation, would you have some code to share that can be used to reproduce the issue (a branch on you personal zephyr fork for instance) ?

Jeepgoing, 2022-03-01:
```
    if ((obj->data.offset >= (UART_RX_BUF_LEN - 80)) ||
            (obj->data.length >= (UART_RX_BUF_LEN/2))) {
        r**et = uart_rx_disable(obj->data.dev);**
        if (ret) {
            myprintk(LOG_LEVEL_ERROR, "uart_rx_disable %d error is %d", chan, ret);
        }

        ret = k_sem_take(&obj->data.rx_disabled, K_MSEC(100));
        if (ret) {
            myprintk(LOG_LEVEL_ERROR, "uart_rx_disable %d timeout", chan);
        }

        if (obj->data.rx_buf_num == 1) {
            obj->data.next_buf = obj->data.rx_buf[obj->data.rx_buf_num];
            obj->data.rx_buf_num = 0;
        } else {
            obj->data.next_buf = obj->data.rx_buf[obj->data.rx_buf_num];
            obj->data.rx_buf_num = 1;
        }
        ret = **uart_rx_enable(obj->data.dev, obj->data.next_buf, UART_RX_BUF_LEN,
                UART_RX_TIMEOUT);**
        if (ret) {
            myprintk(LOG_LEVEL_ERROR, "uart_rx_enable %d error is 0x%x", chan, ret);
        }
    }

```

My workaround method is restart uart when dma offset larger than buffer length - 80, and this method is in my application,
but i am worried about this.
Is this method can added under there [uart_stm32.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_stm32.c)
https://github.com/zephyrproject-rtos/zephyr/blob/0e8f556b1765d077d80ed12fc3b8317ec76c2eb9/drivers/serial/uart_stm32.c#L852-L854

if (data->dma_rx.offset > (data->dma_rx.buffer_length - CONFIG_UART_ASYNC_ALIGN_LEN)) {
	if (data->rx_next_buffer != NULL) {
		uart_stm32_dma_replace_buffer(data->uart_dev);
	}
}

Jeepgoing, 2022-03-01:
> ```
>     if ((obj->data.offset >= (UART_RX_BUF_LEN - 80)) ||
>             (obj->data.length >= (UART_RX_BUF_LEN/2))) {
>         r**et = uart_rx_disable(obj->data.dev);**
>         if (ret) { [...]

Jeepgoing, 2022-03-01:
And this means https://github.com/zephyrproject-rtos/zephyr/blob/0e8f556b1765d077d80ed12fc3b8317ec76c2eb9/drivers/serial/uart_stm32.c#L1171 will never occur

erwango, 2022-06-29:
@Jeepgoing Would you have an easy to use test application to check this ?

Jeepgoing, 2022-07-04:
> @Jeepgoing Would you have an easy to use test application to check this ?

1. zephyr uart driver disable circular buffer here
https://github.com/zephyrproject-rtos/zephyr/blob/db3f8c16e9c95801961a433c07f481bfcb385466/drivers/serial/uart_stm32.c#L1465
2.  this is buffer full isr
https://github.com/zephyrproject-rtos/zephyr/blob/db3f8c16e9c95801961a433c07f481bfcb385466/drivers/serial/uart_stm32.c#L1185
3. you can change this
https://github.com/zephyrproject-rtos/zephyr/blob/db3f8c16e9c95801961a433c07f481bfcb385466/tests/drivers/uart/uart_async_api/src/test_uart_async.c#L82

from 10 to 9, and the second memcmp will be failed.

dcpleung, 2022-08-15:
Mark it as having `future` milestone unless @erwango and @FRASTM think otherwise.

FRASTM, 2022-08-18:
Sorry, this is still not clear whether you send 260 bytes (dma is transferring 260), and/or if you intent to receive only 10.
Maybe a simple sample application (based on the uart_async_api.c model ) could help understanding, ex: 
```
void test_full_buffer_read_setup(void)
void test_full_buffer_read(void)
```

erwango, 2022-08-24:
> intent to receive only 10. Maybe providing a simple sample application (based on the uart_async_api.c model ) could help understanding, something like :
> 
> ```
> void test_full_buffer_read_setup(void)
> void test_full_buffer_read(void)
> ```

@Jeepgoing Ping

ta2-info, 2022-09-11:
I succeeded in reproducing it for the purpose of resolving this problem.
but...
I'm assuming this is a user program issue, not a driver issue.

This happened when calling time-consuming operations such as printk while processing callbacks.

erwango, 2022-09-12:
> I'm assuming this is a user program issue, not a driver issue.

ok, I will close it then.

erwango, 2022-09-12:
> This happened when calling time-consuming operations such as printk while processing callbacks.

@Jeepgoing Is that possible that this is the same reason in your case as well ?

Jeepgoing, 2022-09-12:
> > This happened when calling time-consuming operations such as printk while processing callbacks.
> 
> @Jeepgoing Is that possible that this is the same reason in your case as well ?

No, the driver only adapt to fixed length, not applicable to variable length.

fouge, 2022-10-21:
Hey
I notice the same behavior as @Jeepgoing 

From my comment on the PR adding the async feature: 
I am trying to use UART at 1Mbps but it seems like the buffer replacement when DMA buffer is full is taking too much time so I am loosing bytes. One operation I want to perform in the callback is to copy the buffer into another one to reconstitute a message. (500+ bytes). Message length isn't fixed.
I think we should use the circular mode for DMA instead of the normal one and call `UART_RX_RDY` on half and full buffer so that we never miss any byte as we don't have to manage RX DMA buffers anymore once RX is enabled.

As a quick workaround, I tried reloading the DMA config with a new buffer each time the UART goes into idle between messages as there I have more time, and I don't miss any byte anymore.

erwango, 2022-10-21:
> I'll work on a PR with the circular mode if you think it's the way to go @erwango @FRASTM

Circular mode would be a nice addition indeed! Thanks for proposing.

My expectations would be:
1- take advantage of the dma api:
https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/drivers/dma.h#L175-L176

2- Configurable by device tree (I think `channel-config` is the right prop to do so):
https://github.com/zephyrproject-rtos/zephyr/blob/main/dts/bindings/dma/st,stm32-dma-v1.yaml#L17-L25

closed 2023-01-04 by github-actions[bot]

Jeepgoing, 2023-01-04:
> Hey I notice the same behavior as @Jeepgoing
> 
> From [my comment on the PR](https://github.com/zephyrproject-rtos/zephyr/pull/30917#issuecomment-1286508166) adding the async feature: I am trying to use UART at 1Mbps but it seems like the buffer replacement when DMA buffer is full is taking too much time so I am loosing bytes. One operation I want to perform in the callback is to copy the buffer into another one to reconstitute a message. (500+ bytes). Message length isn't fixed. I think we should use the circular mode for DMA instead of the normal one and call `UART_RX_RDY` on half and full buffer so that we never miss any byte as we don't have to manage RX DMA buffers anymore once RX is enabled.
> 
> As a quick workaround, I tried reloading the DMA config with a new buffer each time the UART goes into idle between messages as there I have more time, and I don't miss any byte anymore.
> 
> I'll work on a PR with the circular mode if you think it's the way to go @erwango @FRASTM

Thank for you PR