zephyrproject-rtos/zephyr #43319: Hardware reset cause api sets reset pin bit everytime the api is called

ShubhadaHarkut-eaton, 2022-03-01:
Hardware reset cause api sets reset pin bit everytime the api is called, Irrespective of the reset reason.
Incase of Software reset, There are 2 bits set that is (software reset and reset pin ) the cause id returned is "3"
Similarly in case of Reset due to Watchdog there are 2 bits set (watchdog and reset pin) the cause id value returned is "17".

Target used : stm32f767 
Zephyr version : 3.0.0-rc

**To Reproduce**
case 1 : software reset 
Enable the hardware info and software reboot in proj.conf of the project.
CONFIG_HWINFO=y
CONFIG_REBOOT=y

Code : 
main.cpp 
#include <zephyr.h>
#include "drivers/hwinfo.h"
#include <sys/reboot.h>
#include <sys/printk.h>
#include <stdbool.h>

void main( void )
{
	uint32_t reset_cause_id = 0;
	int8_t ret_val = hwinfo_get_reset_cause(&reset_cause_id);
	reset_cause_id =16;
	switch (reset_cause_id) {
	case RESET_SOFTWARE:
			printk("SOFTWARE RESET \n");
			break;
		case RESET_PIN:
			printk("RESET PIN \n");
			break;
		case RESET_BROWNOUT:
			printk("RESET_BROWNOUT \n");
			break;
		case RESET_POR:
			printk("RESET_POR\n");
			break;
		case RESET_WATCHDOG:
			printk("WATCHDOG \n");
			break;
		case RESET_DEBUG:
			printk("RESET_DEBUG \n");
			break;
		case RESET_SECURITY:
			printk("RESET_SECURITY \n");
			break;
		case RESET_LOW_POWER_WAKE:
			printk("RESET_LOW POWER \n");
			break;
		case RESET_CPU_LOCKUP:
			printk("RESET_CPU LOCKUP \n");
			break;
		case RESET_PARITY:
			printk("RESET_PARITY \n");
			break;
		case RESET_PLL:
			printk("RESET_PLL \n");
			break;
		case RESET_CLOCK:
			printk("RESET_CLOCK \n");
			break;
		default:
			printk("UNKOWN \n");
			break;
		}
	hwinfo_clear_reset_cause();
       sys_reboot(SYS_REBOOT_WARM);

     while (1) {
      
      //control dosent come here 
       printk("in the loop \n");
	}
Validated it in the debug mode (step by step debugging)

**Expected behavior**
The expected Cause id : 2 but the observed result is 3 (reset pin and software reset)
Multiple reset cause id observed even for the single reset cause. 
default the reset pin cause bit is set along with the actual reset cause bit 

**Impact**
Multiple reset cause id observed even for the single reset cause. 
default the reset pin cause bit is set along with the actual reset cause bit 

**Environment *
Windows OS
IDE Used : Platform IO and Eclipse

erwango, 2022-03-02:
Thanks for raising this issue.

@ShubhadaHarkut-eaton, could you check if https://github.com/zephyrproject-rtos/zephyr/pull/37524 would fix this issue?

alexanderwachter, 2022-03-02:
Additionally i would mention that the reset cause is not an id, but flags. You cannot use a switch(){}
It is perfectly fine if more bits are set at the same time.

erwango, 2022-03-10:
@ShubhadaHarkut-eaton Can you comment on the provided indications?
Otherwise I'm going to close this issue as current consensus is that current behavior should be kept.

ShubhadaHarkut-eaton, 2022-03-11:
@erwango My doubt is resolved now . Good to close this ticket

closed 2022-03-11 by ShubhadaHarkut-eaton