zephyrproject-rtos/zephyr #43330: usb_dc_nrfx.c starts usbd_work_queue with no name

kyrreaa, 2022-03-01:
It is not very clean to start a work_q with no name when all others name them properly, or at least put in the name of their k_work_q struct. Working with system threads it is harder when there is no name.

Suggest adding proper config to the start of usbd_work_queue giving it name like the other work queues.

henrikbrixandersen, 2022-11-17:
> Suggest adding proper config to the start of usbd_work_queue giving it name like the other work queues.

That's a very good idea. Care to submit a PR for this?

markCwatson, 2022-11-30:
@henrikbrixandersen I want to start contributing to the Zephyr Project. I am looking for "Good first issues". I would like to tackle this one please. Can you assign it to me?

PR #52753 (2022-12-05): drivers: usb: usb_dc_nrfx: add usbd worker thread name. 
Note: there remains other drivers using worker threads that do not explicitly name them. May create a different issue to capture that work since this issue pertains to the usb_dc_nrfx specifically.

Fixes: #43330
commit 5381e0b7 (2022-12-05): drivers: usb: usb_dc_nrfx: add usbd worker thread name.

I added a thread name to the usbd worker thread.

Fixes: #43330