zephyrproject-rtos/zephyr #43335: Automatic Automated Backports?

cfriedt, 2022-03-01:
**Is your enhancement proposal related to a problem? Please describe.**
Even though there is already a fair bit of automation for backports, it still relies on the manual step of applying backport tags such as `backport v2.7-branch`.

Although PR's are created automatically after adding the backport tag, maintainers / release managers find it difficult to keep up with the numerous bugfixes that go into main.

Additionally, it's quite possible that bugfixes which **should be backported** have not been.

As of today, backports are all manually triggered. However, I would say that the large majority of backports are valid and end up being merged into release branches.

With that, it makes sense to automatically consider **every** `bug` as a backport candidate and then manually filter-out any which do not apply.  This will rely on a strong differentiation between bugs and feature enhancements on the part of the release / bug triage team.

**Describe the solution you'd like**
Suggest running a script on a weekly basis to look at `bug` PRs that have *already* been merged into `main`.

Within the script, run the following approximate steps:
* For each supported release branch, check the PR merge date against the date the branch was made
* if merge date > branch date, add tag to PR for backport to branch

**Describe alternatives you've considered**

**Additional context**
cc @nashif, @carlescufi, @mbolivar, @dkalowsk, @MaureenHelm, @henrikbrixandersen, @dleach02

cfriedt, 2022-03-02:
> This also raises a question on the need to have an issue filled for each PR that should be merged on a "maintenance branch". Merging a bug fix PR out of the stabilization period doesn't require an associated issue. But since most of bug fixes are back-ported, we frequently need to raise issues just to satisfy the merge criteria on the back-port target branch, which is adding its own overhead. Can we do something about that as well ? Specially if we start saying that each `bug` PR is a potential back-port candidate.

@erwango  - That's a really good point. It should be possible to automatically create a new issue if there is no linked issue in the original PR to `main`. However, those backport PRs would **necessarily** need to be labeled `DNM` and **not** moved out of triage until the issue is filled in with sufficient detail.

henrikbrixandersen, 2022-03-04:
How would you automatically determine if a bug being fixed in `main` is also present in the release branches? The bug fix may apply cleanly even if the bug is not present on that branch.

cfriedt, 2022-03-08:
https://github.com/zephyrproject-rtos/zephyr/wiki/Zephyr-Change-Control-Board-%28CCB%29
https://wiki.ubuntu.com/StableReleaseUpdates
https://lwn.net/Articles/764647/

cfriedt, 2022-03-08:
> How would you automatically determine if a bug being fixed in `main` is also present in the release branches? The bug fix may apply cleanly even if the bug is not present on that branch.

Also an excellent point. Especially difficult with bugs that are nondeterministic or otherwise not easy to reproduce.

At the other end of the spectrum, we have negative side-effects, like what @nashif mentioned. We do not want to start adding bugfixes for bugfixes (regressions) - we need a buffer (a.k.a. soak period) preferably with lots of vendor testing as well.

cfriedt, 2022-03-08:
We also discussed whether to apply backports to non-LTS release branches. Some are still on v2.6.0, for example. Having 10 different development branches does not make a lot of sense.

erwango, 2022-03-09:
> t the other end of the spectrum, we have negative side-effects, like what @nashif mentioned. Every backport has some amount of risk. We do not want to start adding bugfixes for bugfixes (regressions) - we need a buffer (a.k.a. soak period) preferably with lots of vendor testing as well.

Don't we also have backport on the V3.0 for instance?

nashif, 2022-03-09:
also see discussion here https://github.com/zephyrproject-rtos/zephyr/discussions/43561 and provide your feedback there.

closed 2022-08-11 by cfriedt