zephyrproject-rtos/zephyr #43344: intel_adsp_cavs: samples/subsys/logging/syst is failing with a timeout

yerabolu, 2022-03-02:
**Describe the bug**
samples/subsys/logging/syst is failing with a timeout as the switching to sys-t raw data doesn't happen and logger is waiting. 

Please also mention any information which could help others to understand
the problem you're facing:
- happens on any of the cavs hardware once they are included in the platform allow list. 

**To Reproduce**
Steps to reproduce the behavior:
1.  add the intel_adsp_cavs15 or cavs18 or cavs25 to sample.yaml
```
sample:
  name: syst sample
tests:
  sample.logger.syst.v1.deferred:
    extra_args: OVERLAY_CONFIG=overlay_deferred.conf
    platform_allow: mps2_an385 qemu_x86 sam_e70_xplained intel_adsp_cavs15
    tags: logging
    harness: console
    harness_config:
      type: one_line
      regex:
        - "SYS-T RAW DATA: "

```
2. twister -W -p intel_adsp_cavs15 --device-testing --device-serial-pty=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh --west-flash=$HOME/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstwist.sh -T samples/subsys/logging/syst

3. see below error. 

**Expected behavior**
Test should execute successfully similar to qemu_x86.

**Logs and console output**
```
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 0 of length 16384 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Selected output stream 7 (GCAP = 0x1006701)
DEBUG   - DEVICE: INFO:cavs-fw:Mapped PCI bar 4 of length 1048576 bytes.
DEBUG   - DEVICE: INFO:cavs-fw:Detected cAVS 1.5 hardware
DEBUG   - DEVICE: INFO:cavs-fw:Trimming 16 bytes of extended manifest
DEBUG   - DEVICE: INFO:cavs-fw:Resetting HDA device
DEBUG   - DEVICE: INFO:cavs-fw:Powering down DSP cores
DEBUG   - DEVICE: INFO:cavs-fw:Configuring HDA stream 7 to transfer firmware image
DEBUG   - DEVICE: INFO:cavs-fw:Mapped 2M huge page at 0x150600000 to contain 110592 bytes of firmware
DEBUG   - DEVICE: INFO:cavs-fw:Filled the buffer descriptor list (BDL) for DMA.
DEBUG   - DEVICE: INFO:cavs-fw:Starting DSP, ADSPCS = 0x303
DEBUG   - DEVICE: INFO:cavs-fw:Wait for ROM startup
DEBUG   - DEVICE: INFO:cavs-fw:Sending IPC command, HIPCR = 0x81004e00
DEBUG   - DEVICE: INFO:cavs-fw:Starting DMA, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:Waiting for firmware handoff, FW_STATUS = 0x5000001
DEBUG   - DEVICE: INFO:cavs-fw:cAVS firmware load complete
DEBUG   - DEVICE: --
DEBUG   - DEVICE: [00000062] <err> syst: Error message example.
DEBUG   - DEVICE: [00000062] <wrn> syst: Warning message example.
DEBUG   - DEVICE: [00000062] <inf> syst: Info message example.
DEBUG   - DEVICE: [00000062] <dbg> syst: Debug message example.
DEBUG   - DEVICE: [00000062] <dbg> syst: Debug message example, 1
DEBUG   - DEVICE: [00000063] <dbg> syst: Debug message example, 1, 2
DEBUG   - DEVICE: [00000063] <dbg> syst: Debug message example, 1, 2, 3
DEBUG   - DEVICE: [00000063] <dbg> syst: Debug message example, 1, 2, 3, 0x4
DEBUG   - DEVICE: [00000063] <dbg> syst: char !
DEBUG   - DEVICE: [00000063] <dbg> syst: s str static str
DEBUG   - DEVICE: [00000063] <dbg> syst: d str dynamic str
DEBUG   - DEVICE: [00000063] <dbg> syst: mixed str dynamic str --- dynamic str --- another dynamic str --- another dynamic str
DEBUG   - DEVICE: [00000063] <dbg> syst: mixed c/s ! static str dynamic str static str !
DEBUG   - DEVICE: [00000063] <dbg> syst: Debug message example, %f
DEBUG   - DEVICE: [00000063] <err> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000064] <wrn> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000064] <inf> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000064] <dbg> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: hello sys-t on board intel_adsp_cavs15
DEBUG   - DEVICE: [00000064] <err> syst: Error message example.
DEBUG   - DEVICE: [00000064] <wrn> syst: Warning message example.
DEBUG   - DEVICE: [00000064] <inf> syst: Info message example.
DEBUG   - DEVICE: [00000065] <dbg> syst: Debug message example.
DEBUG   - DEVICE: [00000065] <dbg> syst: Debug message example, 1
DEBUG   - DEVICE: [00000065] <dbg> syst: Debug message example, 1, 2
DEBUG   - DEVICE: [00000065] <dbg> syst: Debug message example, 1, 2, 3
DEBUG   - DEVICE: [00000065] <dbg> syst: Debug message example, 1, 2, 3, 0x4
DEBUG   - DEVICE: [00000065] <dbg> syst: char !
DEBUG   - DEVICE: [00000065] <dbg> syst: s str static str
DEBUG   - DEVICE: [00000065] <dbg> syst: d str dynamic str
DEBUG   - DEVICE: [00000065] <dbg> syst: mixed str dynamic str --- dynamic str --- another dynamic str --- another dynamic str
DEBUG   - DEVICE: [00000065] <dbg> syst: mixed c/s ! static str dynamic str static str !
DEBUG   - DEVICE: [00000065] <dbg> syst: Debug message example, %f
DEBUG   - DEVICE: [00000065] <err> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000066] <wrn> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000066] <inf> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000066] <dbg> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: hello sys-t on board intel_adsp_cavs15
DEBUG   - DEVICE: [00000066] <err> syst: Error message example.
DEBUG   - DEVICE: [00000066] <wrn> syst: Warning message example.
DEBUG   - DEVICE: [00000067] <inf> syst: Info message example.
DEBUG   - DEVICE: [00000067] <dbg> syst: Debug message example.
DEBUG   - DEVICE: [00000067] <dbg> syst: Debug message example, 1
DEBUG   - DEVICE: [00000067] <dbg> syst: Debug message example, 1, 2
DEBUG   - DEVICE: [00000067] <dbg> syst: Debug message example, 1, 2, 3
DEBUG   - DEVICE: [00000067] <dbg> syst: Debug message example, 1, 2, 3, 0x4
DEBUG   - DEVICE: [00000067] <dbg> syst: char !
DEBUG   - DEVICE: [00000067] <dbg> syst: s str static str
DEBUG   - DEVICE: [00000067] <dbg> syst: d str dynamic str
DEBUG   - DEVICE: [00000067] <dbg> syst: mixed str dynamic str --- dynamic str --- another dynamic str --- another dynamic str
DEBUG   - DEVICE: [00000067] <dbg> syst: mixed c/s ! static str dynamic str static str !
DEBUG   - DEVICE: [00000068] <dbg> syst: Debug message example, %f
DEBUG   - DEVICE: [00000068] <err> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000068] <wrn> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000068] <inf> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: [00000068] <dbg> syst: frame
DEBUG   - DEVICE: 03 d2 04 00 00 08 01 02  03 04 05 06 07 08       |........ ......
DEBUG   - DEVICE: hello sys-t on board intel_adsp_cavs15
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs15
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain (e.g Zephyr SDK, ...) - SDK 13.2
 - Commit SHA: 04ea1ee1afab

nashif, 2022-03-08:
This sample does not build correctly on this platform, needs fixing:

```
[6/154] Generating include/generated/version.h
-- Zephyr version: 3.0.99 (/home/anashif/zephyrproject/zephyr), build: zephyr-v3.0.0-883-g47ddbc2ed6b0
[110/154] Building C object modules/mipi-sys-t/CMakeFiles/..__modules__debug__mipi-sys-t.dir/library/src/mipi_syst_api.c.obj
In file included from /home/anashif/zephyrproject/zephyr/build/modules/mipi-sys-t/library/include/mipi_syst.h:96,
                 from /home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:41:
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c: In function 'buildPrintfPayload':
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:939:34: warning: 'wchar_t' {aka 'short unsigned int'} is promoted to 'int' when passed through '...'
  939 |        COPY_ARG32(mipi_syst_u32, wchar_t);
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/include/mipi_syst/compiler.h:161:31: note: in definition of macro 'MIPI_SYST_HTOLE32'
  161 | #define MIPI_SYST_HTOLE32(v) (v)
      |                               ^
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:939:8: note: in expansion of macro 'COPY_ARG32'
  939 |        COPY_ARG32(mipi_syst_u32, wchar_t);
      |        ^~~~~~~~~~
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:939:34: note: (so you should pass 'int' not 'wchar_t' {aka 'short unsigned int'} to 'va_arg')
  939 |        COPY_ARG32(mipi_syst_u32, wchar_t);
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/include/mipi_syst/compiler.h:161:31: note: in definition of macro 'MIPI_SYST_HTOLE32'
  161 | #define MIPI_SYST_HTOLE32(v) (v)
      |                               ^
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:939:8: note: in expansion of macro 'COPY_ARG32'
  939 |        COPY_ARG32(mipi_syst_u32, wchar_t);
      |        ^~~~~~~~~~
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:939:34: note: if this code is reached, the program will abort
  939 |        COPY_ARG32(mipi_syst_u32, wchar_t);
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/include/mipi_syst/compiler.h:161:31: note: in definition of macro 'MIPI_SYST_HTOLE32'
  161 | #define MIPI_SYST_HTOLE32(v) (v)
      |                               ^
/home/anashif/zephyrproject/modules/debug/mipi-sys-t/library/src/mipi_syst_api.c:939:8: note: in expansion of macro 'COPY_ARG32'
  939 |        COPY_ARG32(mipi_syst_u32, wchar_t);
      |        ^~~~~~~~~~
[143/154] Linking C executable zephyr/zephyr_pre0.elf
```

nashif, 2022-03-08:
moving back to medium.

nashif, 2022-03-08:
need to remove `platform_allow: mps2_an385 qemu_x86 sam_e70_xplained` and use `integration_platforms` instead. This just limits testing to two architectures and emulation boards.

aasthagr, 2022-03-09:
Looking into this issue.

aasthagr, 2022-03-10:
I am able to replicate this bug and the above build warnings as well.

aasthagr, 2022-03-10:
The intel_adsp_cavs25 platform should use the log backend ADSP instead of the UART backend. When used CONFIG_LOG_BACKEND_ADSP and CONFIG_LOG_BACKEND_ADSP_OUTPUT_SYST in the prj.conf, this sample passes in twister. Although the logs indicates the 1 line output which is in SYST format. Still working to figure out why there is just one line output.

[image]

aasthagr, 2022-03-23:
Seems like the call from MIPI_SYST_VPRINTF never returns after printing the first line of SYST output. Looks like somethings up with how MIPI SYST processes for xtensa.

PR #44491 (2022-04-20): sample: syst: Add config for testing on intel_adsp_cavs platforms
Duplicated branch solution (see PR #44232). The original author is on vacation and I am unable to push updates to the original PR due to deviations caused by a rebase).

Original description below ...

Adding configuration for enabling this test on cavs platforms.
This patch also contains a fix to mipi_syst library which resolves
the memory alignment issue across different architectures.

Fixes #43344
commit 7141c876 (2022-04-20): sample: syst: Add intel_adsp_cavs support

Adding configurations for enabling this sample project on cavs
platforms. This patch also contains a fix to mipi_syst library
which resolves the memory alignment issue across different
architectures.

Fixes #43344
closed 2022-04-20 by nashif