zephyrproject-rtos/zephyr #43356: DisplayDriver st7735 - Rotation at runtime

tfir1981, 2022-03-02:
**Is your feature request related to a problem? Please describe.**
I am using a type `st7735r` display in my project and face the need of display-rotation at runtime. The display api provides the function called `st7735r_set_orientation()` which is not implemented/ not supported yet. Per default the Driver is implemented in landscape-mode. So that is okay for me. In addition, i only need the 180°-rotation. Softwarerotation is not an option for me.

**Describe the solution you'd like**
I temporary implemented the solution als follows:

```
static int st7735r_set_orientation(const struct device *dev,
				   const enum display_orientation orientation)
{
	struct st7735r_config *config = (struct st7735r_config *)dev->config;
	struct st7735r_data *data = (struct st7735r_data *)dev->data;
	int ret = 0;
	uint8_t madctl = config->madctl;

	madctl &= ~(ST7735R_MADCTL_MV |
				ST7735R_MADCTL_MX |
				ST7735R_MADCTL_MY);

	switch (orientation)
	{
	case DISPLAY_ORIENTATION_NORMAL:
		madctl |= ST7735R_MADCTL_MX | ST7735R_MADCTL_MV;
		break;
	case DISPLAY_ORIENTATION_ROTATED_90:
        return -ENOTSUP;
		break;
	case DISPLAY_ORIENTATION_ROTATED_180:
		madctl |= ST7735R_MADCTL_MY | ST7735R_MADCTL_MV;
		break;
	case DISPLAY_ORIENTATION_ROTATED_270:
        return -ENOTSUP;
		break;
	default:
		return -ENOTSUP;
		break;
	}

	ret = st7735r_transmit(data, ST7735R_CMD_MADCTL, &madctl, sizeof(madctl));
	if (ret < 0) {
		return ret;
	}

	return 0;
}
```

**Describe alternatives you've considered**
I know about the possibility to rotate the screen by DeviceTree, but i neet to rotate at runtime.
I thinke the final solution is not as trivial as mentioned above, because the DeviceTree-Settings are defined as `DISPLAY_ORIENTATION_NORMAL`.

tfir1981, 2022-03-02:
I would be very pleased if a PR will come. Thank you

gmarull, 2022-03-02:
> I would be very pleased if a PR will come. Thank you

Can you do it yourself?

tfir1981, 2022-03-02:
Yes, I will do it.