zephyrproject-rtos/zephyr #43369: Use Zephyr crc implementation for LittleFS

tejlmand, 2022-03-02:
**Is your enhancement proposal related to a problem? Please describe.**
LittleFS used in Zephyr is shipped with a naive crc implementation, not really suited for production environment.

Instead of using LittleFS crc, a proper Zephyr crc should be used.
This is also indirectly recommended by LittleFS, because as soon as a developer provides a custom configuration header, then the built-in CRC automatically gets disabled and the developer is forced to provide a custom implementation.

**Describe the solution you'd like**
Possibility of using one of Zephyr natives CRC implementations from here:
https://github.com/zephyrproject-rtos/zephyr/tree/main/lib/os

**Describe alternatives you've considered**
N/A

**Additional context**
See this PR: https://github.com/zephyrproject-rtos/zephyr/pull/41729

SdtElectronics, 2022-05-22:
The [native implementation in Zephyr](https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/os/crc32_sw.c) looks almost identical to the implementation in LFS, except two additional bitwise complement operations were applied to the first parameter and the return value. If the Zephyr implementation will be used, a wrapper will be required to keep the behavior the same as the LFS implementation. I'm not sure if this is the supposed "enhancement", though. Since the Zephyr implementation is no different, IMHO.

cfriedt, 2022-05-31:
Looks like this issue should have been closed with #41729

@SdtElectronics - feel free to open a new issue with your concerns.

tejlmand, 2022-08-12:
> Looks like this issue should have been closed with #41729

@cfriedt quite the opposite, if you look at [this issue](https://github.com/zephyrproject-rtos/zephyr/issues/43369) description, #41729 is stated as having additional context.
See for example this comment: https://github.com/zephyrproject-rtos/zephyr/pull/41729#discussion_r817665077

So we need a better (production quality) CRC for little FS, and that is not what #41729.
#41729 just brought the naive basic CRC from little FS into Zephyr tree, just to have things still working.

This issue is still valid.

de-nordic, 2024-02-08:
Seems that CRC function is in:
https://github.com/zephyrproject-rtos/littlefs/blob/408c16a909dd6cf128874a76f21c793798c9e423/lfs_util.c#L15
and is added to build with:
https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/littlefs/CMakeLists.txt#L14
which is within Zephyr tree.
Would it be possible to define our own function named as expected by LFS that would be actually calling Zephyr CRC subroutines?