zephyrproject-rtos/zephyr #43378: TLS availability misdetection when ZEPHYR_TOOLCHAIN_VARIANT is not set

stephanosio, 2022-03-02:
**Describe the bug**
The `THREAD_LOCAL_STORAGE` Kconfig symbol, which enables the thread local storage (TLS) feature, depends on the `TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` symbol, which defaults to `y` when the Zephyr SDK is used.

https://github.com/zephyrproject-rtos/zephyr/blob/96b4a86a5120898a114ffd96c5d1edfdcd146d0d/kernel/Kconfig#L907-L909

The `TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` Kconfig symbol is specified such that it defaults to `y` when the `ZEPHYR_TOOLCHAIN_VARIANT` environment variable is set to `zephyr` (i.e. Zephyr SDK toolchain is used).

This is not a problem when the `ZEPHYR_TOOLCHAIN_VARIANT` environment variable is set to `zephyr`, but the Zephyr SDK may also be implicitly used when the `ZEPHYR_TOOLCHAIN_VARIANT` environment variable is not set to any value and the Zephyr SDK CMake package is registered in the package registry, in which case the `TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` Kconfig symbol will not be set to `y` even if the Zephyr SDK is used, effectively making the thread local storage feature unavailable.

**To Reproduce**
1. Install Zephyr SDK and register CMake pacakge.
2. Run `tests/kernel/threads/tls` test without `ZEPHYR_TOOLCHAIN_VARIANT` environment variable not set.
3. Observe the following warning message and subsequent link failure:
```
warning: THREAD_LOCAL_STORAGE (defined at kernel/Kconfig:914) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_THREAD_LOCAL_STORAGE.html and/or look
up THREAD_LOCAL_STORAGE in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.
```

**Expected behavior**
`TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` is always set to `y` when the Zephyr SDK is used regardless of the `ZEPHYR_TOOLCHAIN_VARIANT` environment variable being set to `zephyr`. 

**Impact**
This can be very misleading for the users who are getting started with the Zephyr, especially because the Zephyr Getting Started Guide does not mention anything about setting the `ZEPHYR_TOOLCHAIN_VARIANT=zephyr` when the Zephyr SDK is used.

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.13.2
 - Commit SHA: 0f8d9b430b2c2a56915ad646b0190c1b8d8dcfd0

**Additional context**
The default value for `TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` Kconfig symbol should be derived from a more reliable source than the `ZEPHYR_TOOLCHAIN_VARIANT` environment variable, which may or may not be set to `zephyr` when the Zephyr SDK is used.

stephanosio, 2022-03-02:
FYI @tejlmand @dcpleung

dcpleung, 2022-03-02:
IIRC, this was done before the CMake package changes, and there still was Kconfig files in the SDK (where `TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` could be set).

Maybe we can have CMake test the TLS compile flag, and forces an error if it fails? Then there is no need for the `TOOLCHAIN_SUPPORTS_THREAD_LOCAL_STORAGE` Kconfig anymore.

PR #43402 (2022-03-04): cmake: export ZEPHYR_TOOLCHAIN_VARIANT to kconfiglib
Fixes: #43378

Settings used in Zephyr Kconfig tree must be exported by CMake to ensure
correct behaviour in all situations.

The ZEPHYR_TOOLCHAIN_VARIANT is used in the Kconfig tree but was not
exported by CMake, and thus Kconfig tree relied on this setting being
set by the user in the environment.

Fix this by adding ZEPHYR_TOOLCHAIN_VARIANT to list of exported
variables.
commit 491aea49 (2022-03-04): cmake: move ZEPHYR_TOOLCHAIN_VARIANT export to common settings

Fixes: #43378

Move the exporting of ZEPHYR_TOOLCHAIN_VARIANT from the custom target to
the COMMON_KCONFIG_ENV_SETTINGS variable.
This ensures that the setting is exported both when running the initial
kconfiglib parsing but also on later menuconfig / guiconfig invocations.
closed 2022-03-04 by MaureenHelm