zephyrproject-rtos/zephyr #43390: gPTP broken in Zephyr 3.0

mdkf, 2022-03-02:
**Describe the bug**
I have tried to run the gPTP sample on both a FRDM-K64F and IMXRT1050-EVKB and both have just print "<wrn> net_gptp: Reset Pdelay requests" endlessly. It looks like this cod was recently changed in https://github.com/zephyrproject-rtos/zephyr/issues/33747
#33747
#29599
I have attached a Wireshark trace

**To Reproduce**
Steps to reproduce the behavior:
1. build and flash the gPTP demo

**Expected behavior**
Time Sync should occur

**Impact**
gPTP does not work

**Logs and console output**
uart:~$ *** Booting Zephyr OS build zephyr-v3.0.0-337-g52bee41bba11  ***
[00:00:00.016,326] <inf> net_config: Initializing network
[00:00:00.016,357] <inf> net_config: Waiting interface 1 (0x80002550) to be up...
[00:00:03.011,230] <inf> net_config: Interface 1 (0x80002550) coming up
[00:00:03.011,230] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:03.011,260] <inf> net_config: IPv4 address: 192.0.2.2
[00:00:04.015,625] <wrn> net_gptp: Reset Pdelay requests

**Environment (please complete the following information):**
 - Using ptp4l 3.1 on Ubuntu Impish
 - Zephyr 3.0.99

**Additional context**
[gPTP Zephyr 3.0 iMXRT1050.pcapng.gz](https://github.com/zephyrproject-rtos/zephyr/files/8172812/gPTP.Zephyr.3.0.iMXRT1050.pcapng.gz)

mmahadevan108, 2022-03-04:
@hakehuang. Can you please take a look at this issue.

hakehuang, 2022-03-07:
@mdkf  I tries on my board which is OK on zephyr-v3.0.0-721-gc32d10cc9e latest version, and look at your steps, I want to configrm that do you run the gptp daemon on pc host? the steps are in readme. code https://github.com/AVnu. and below is my command line:
[gptp_cfg.zip](https://github.com/zephyrproject-rtos/zephyr/files/8194711/gptp_cfg.zip)

```
sudo ./gptp eth0 -F gptp_cfg.ini
```

mmahadevan108, 2022-07-15:
@mdkf can you please take a look at the suggestions from @hakehuang

mmahadevan108, 2022-09-14:
@mdkf I am closing this issue, please reopen if you have any additional information.

closed 2022-09-14 by mmahadevan108