zephyrproject-rtos/zephyr #43411: STM32 SPI DMA issue

darexa, 2022-03-03:
**Describe the bug**
The SPI LL driver for STM32 enables chip select before enabling SPI when using DMA.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
STM32L452 with a SPI display.
 - What have you tried to diagnose or workaround this issue?
 I noticed my SPI display worked fine without DMA, but enver could get it working with DMA.
I took a look at the signals and could see that the idle clock state was incorrect when CS was pulled low.
The idle clock state change slightly after.
After going through the code it is clear why.
In the transceive()-function, "LL_SPI_Enable(spi);" is called before "spi_stm32_cs_control(dev, true);".
In transceive_dma() it's the other way around.
Simply swapping places makes it work.

**To Reproduce**
Steps to reproduce the behavior:
Use SPI DMA on any STM32 device with a peripheral that uses idle high clock state.
It's probably peripheral dependent if they can handle this issue also.

**Expected behavior**
Expecting same behaviour on DMA as non-DMA.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
No impact for me since I worked around it.
Impact for anyone else with this epcific setup though.

ycsin, 2022-03-04:
Seems like you have a fix already, do you mind sending a PR to fix this?

darexa, 2022-03-04:
I'll see what I can do when I get some spare time. I've never pushed a PR here before so I need to read through the guidelines etc.

ycsin, 2022-03-04:
> I'll see what I can do when I get some spare time. I've never pushed a PR here before so I need to read through the guidelines etc.

No worries, it took me three times to get my first PR done right

FRASTM, 2022-03-10:
@darexa do you need support to submit a PR, 
take a look at https://github.com/FRASTM/zephyr/tree/issue43411   where I moved the `spi_stm32_cs_control` after `spi_stm32_cs_control` in the transceive_dma function of the ./drivers/spi/spi_ll_stm32.c

darexa, 2022-03-10:
@FRASTM Sure, I am a little bit swamped with other stuff at the moment so if you support that is great. The code looks correct in your branch.

FRASTM, 2022-03-10:
> @FRASTM Sure, I am a little bit swamped with other stuff at the moment so if you support that is great. The code looks correct in your branch.

Pushed in that way, please add your review

closed 2022-03-11 by carlescufi

bhavikbhansali-eaton, 2022-11-23:
Is there any solution for this I am still facing the issue.