zephyrproject-rtos/zephyr #43440: Bluetooth: L2CAP send le data lack calling net_buf_unref() function

yajunzhi, 2022-03-04:
**Describe the bug**
In l2cap_chan_le_send() function, if err return value is not success, why need check the segment is the original buffer or not.
The source code is this, 
        /* If the segment is not the original buffer release it since it
         * won't be needed anymore.
         */
        if (seg != buf) {
            net_buf_unref(seg);
        }

I think it should be:
       /* release the segment   
         */
        net_buf_unref(seg);
// and relese the seg in l2cap_chan_tx_process() function,

Is my understand right?

carlescufi, 2022-03-10:
@Vudentz would you mind checking this?

yajunzhi, 2022-03-24:
Also, I found that these places also need call unref(), is right?

if (conn->rx->len < sizeof(uint16_t)) {
		/* Still not enough data received to retrieve the L2CAP header
		 * length field.
		 */
+++        net_buf_unref(buf);
		return;
	}

	acl_total_len = sys_get_le16(conn->rx->data) + sizeof(struct bt_l2cap_hdr);

	if (conn->rx->len < acl_total_len) {
		/* L2CAP frame not complete. */
 +++       net_buf_unref(buf);
		return;
	}

cvinayak, 2022-04-28:
@yajunzhi Could you provide the symptoms you are observing with the reported issue. Also, it will be helpful if you can provide a sample test application to reproduce the issues.

Also, if there is issue with `bt_acl_recv`, this should be a new github issue.

Closing this issue, until more details around reproducing the issue is provided, and feel free to reopen.

closed 2022-04-28 by cvinayak