zephyrproject-rtos/zephyr #43456: winc1500 wifi driver fails to build

brentalanmiller, 2022-03-04:
**Describe the bug**
I was trying to integrate the winc1500 driver into a project that I was working on and noticed that the build was failing when I enabled the driver. I then tried to to build the `samples/net/wifi` project using the `reel_board` as the target (which uses the winc1500) and it fails with the exact same error.

**To Reproduce**
Steps to reproduce the behavior:
1. cd $ZEPHYR_BASE
2. git checkout v3.0.0
3. west update
4. west build -p -b reel_board samples/net/wifi
5. See error

**Expected behavior**
Sample code to compile without error.

**Impact**
Unable to use the winc1500 module with Zephyr.

**Logs and console output**
```
% west build -p -b reel_board samples/net/wifi
-- west build: making build dir /Users/bmiller/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /Users/bmiller/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /Users/bmiller/zephyrproject/zephyr/samples/net/wifi
-- Zephyr version: 3.0.0 (/Users/bmiller/zephyrproject/zephyr)
-- Found Python3: /Users/bmiller/zephyrproject/.venv/bin/python3.9 (found suitable exact version "3.9.10") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: reel_board
-- Cache files will be written to: /Users/bmiller/Library/Caches/zephyr
-- Found dtc: /opt/homebrew/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/opt/homebrew)
-- Found BOARD.dts: /Users/bmiller/zephyrproject/zephyr/boards/arm/reel_board/reel_board.dts
-- Found devicetree overlay: /Users/bmiller/zephyrproject/zephyr/samples/net/wifi/boards/reel_board.overlay
-- Generated zephyr.dts: /Users/bmiller/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /Users/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /Users/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /Users/bmiller/zephyrproject/zephyr/build/zephyr/dts.cmake

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

Parsing /Users/bmiller/zephyrproject/zephyr/Kconfig
Loaded configuration '/Users/bmiller/zephyrproject/zephyr/boards/arm/reel_board/reel_board_defconfig'
Merged configuration '/Users/bmiller/zephyrproject/zephyr/samples/net/wifi/prj.conf'
Merged configuration '/Users/bmiller/zephyrproject/zephyr/samples/net/wifi/boards/reel_board.conf'
Configuration saved to '/Users/bmiller/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/Users/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/homebrew/bin/arm-none-eabi-gcc
CMake Warning at /Users/bmiller/zephyrproject/zephyr/CMakeLists.txt:775 (message):
  No SOURCES given to Zephyr library: subsys__net__pkt_filter

  Excluding target from build.

-- Configuring done
-- Generating done
-- Build files have been written to: /Users/bmiller/zephyrproject/zephyr/build
-- west build: building application
[1/325] Preparing syscall dependency handling

[3/325] Generating include/generated/version.h
-- Zephyr version: 3.0.0 (/Users/bmiller/zephyrproject/zephyr), build: zephyr-v3.0.0
[10/325] Building C object zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/driver/source/nmspi.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/driver/source/nmspi.c.obj 
[... 89 lines cut ...]
/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/./bsp/include/nm_bsp_internal.h:11:10: fatal error: wifi_winc1500_nm_bsp_internal.h: No such file or directory
   11 | #include <wifi_winc1500_nm_bsp_internal.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[19/325] Building C object zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c.obj 
ccache /opt/homebrew/bin/arm-none-eabi-gcc -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/Users/bmiller/zephyrproject/zephyr/kernel/include -I/Users/bmiller/zephyrproject/zephyr/arch/arm/include -I/Users/bmiller/zephyrproject/zephyr/include -I/Users/bmiller/zephyrproject/zephyr/build/zephyr/include/generated -I/Users/bmiller/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/Users/bmiller/zephyrproject/zephyr/lib/util/fnmatch/. -I/Users/bmiller/zephyrproject/zephyr/soc/arm/nordic_nrf/common/. -I/Users/bmiller/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/. -I/Users/bmiller/zephyrproject/zephyr/include/drivers -I/Users/bmiller/zephyrproject/modules/hal/nordic/nrfx -I/Users/bmiller/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/Users/bmiller/zephyrproject/modules/hal/nordic/nrfx/mdk -I/Users/bmiller/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -isystem /Users/bmiller/zephyrproject/zephyr/lib/libc/minimal/include -isystem /Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/include -isystem /Applications/ARM/bin/../lib/gcc/arm-none-eabi/10.3.1/include-fixed -Os -imacros /Users/bmiller/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros /Users/bmiller/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/Users/bmiller/zephyrproject/zephyr/samples/net/wifi=CMAKE_SOURCE_DIR -fmacro-prefix-map=/Users/bmiller/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/Users/bmiller/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c.obj -MF zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c.obj.d -o zephyr/CMakeFiles/zephyr.dir/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c.obj -c /Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c
In file included from /Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/./common/include/nm_debug.h:46,
                 from /Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/./common/include/nm_common.h:46,
                 from /Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/common/source/nm_common.c:41:
/Users/bmiller/zephyrproject/modules/hal/atmel/asf/common/components/wifi/winc1500/./bsp/include/nm_bsp_internal.h:11:10: fatal error: wifi_winc1500_nm_bsp_internal.h: No such file or directory
   11 | #include <wifi_winc1500_nm_bsp_internal.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/cmake --build /Users/bmiller/zephyrproject/zephyr/build
```

**Environment (please complete the following information):**
 - OS: MacOS 12.2.1 and Ubuntu 20.04
 - Toolchain: GNU Arm Embedded Toolchain 10.3-2021.10 on MacOS, Zephyr SDK on Ubuntu
 - Commit: tag v3.0.0

**Additional context**
The error message above is from building on MacOS, but I also tried building on Linux and got the same error.

mbolivar-nordic, 2022-03-08:
@brentalanmiller did you run `west update`?

brentalanmiller, 2022-03-08:
Hi, @mbolivar-nordic . Yes, I made sure to do that.

I was also trying out previous releases and the last Zephyr release that was able to build this sample was v2.6.0 (v2.7.0 fails,) so it looks like the issue was introduced around that time.

mbolivar-nordic, 2022-03-12:
I can reproduce. @brentalanmiller thanks for the hints about it working before. This allowed me to run a bisection. It would help if you could provide bisection data in future bug reports as well.

579ca90e25271a693044667418acef69669f353e is the first bad commit

```
commit 579ca90e25271a693044667418acef69669f353e
Author: Daniel Leung <daniel.leung@intel.com>
Date:   Wed Aug 4 14:25:37 2021 -0700

    drivers: wifi: build as static library
    
    Instead of putting object files inside libzephyr.a,
    simply build a separate static library as most other
    driver types are doing this already.
    
    Signed-off-by: Daniel Leung <daniel.leung@intel.com>

 drivers/wifi/CMakeLists.txt            |  9 +++++----
 drivers/wifi/esp32/CMakeLists.txt      |  2 +-
 drivers/wifi/esp_at/CMakeLists.txt     |  2 +-
 drivers/wifi/eswifi/CMakeLists.txt     | 10 +++++-----
 drivers/wifi/simplelink/CMakeLists.txt |  6 +++---
 drivers/wifi/winc1500/CMakeLists.txt   |  4 ++--
 6 files changed, 17 insertions(+), 16 deletions(-)
```

@dcpleung to reproduce the failure that bisect shows here:

```
west build -p -b reel_board samples/net/wifi
```

PR #43779 (2022-03-15): wifi: fix build error on wifi_winc1500_nm_bsp_internal.h
Previous commit 579ca90e25271a693044667418acef69669f353e to
build wifi drivers as a library changed the include path for
the WINC1500 driver, which results in the include path being
local to the library. However, the Atmel HAL requires
wifi_winc1500_nm_bsp_internal.h to in the search path. So
change the include path to be global.

Fixes #43456
commit e186ebc3 (2022-03-15): wifi: fix build error on wifi_winc1500_nm_bsp_internal.h

Previous commit 579ca90e25271a693044667418acef69669f353e to
build wifi drivers as a library changed the include path for
the WINC1500 driver, which results in the include path being
local to the library. However, the Atmel HAL requires
wifi_winc1500_nm_bsp_internal.h to in the search path. So
change the include path to be global.

Fixes #43456
closed 2022-03-15 by nashif