zephyrproject-rtos/zephyr #43465: 'Malformed data' on bt_data_parse() for every ble adv packet on bbc_microbit

atoncetti, 2022-03-05:
**Describe the bug**
I am trying to inspect ble advertising packets on a bbc_microbit. I ran into the problem that for every adv packet `bt_data_parse()` logs 'Malformed data' and returns. ([bt_data_parse() relevant part](https://github.com/zephyrproject-rtos/zephyr/blob/c32d10cc9e13d9e050d27b5804c0213f6cd3f452/subsys/bluetooth/host/hci_core.c#L3806))

Relevant code:
```
static bool data_cb(struct bt_data *data, void *user_data)
{
	int err;

	switch (data->type) {

		case BT_DATA_MANUFACTURER_DATA:
			printk("Found!");

			err = bt_le_scan_stop();
			if (err) {
				printk("Stop LE scan failed (err %d)\n", err);
				return false;
			}

			printk("Stopping.");

			return false;
		default:
			return true;		
	}
}

static void device_found(const bt_addr_le_t *addr, int8_t rssi, uint8_t type,
			 struct net_buf_simple *ad)
{
	char dev[BT_ADDR_LE_STR_LEN];

	bt_addr_le_to_str(addr, dev, sizeof(dev));

	bt_data_parse(ad, data_cb, (void *)addr);
}

static void start_scan(void)
{
	int err;
	
	struct bt_le_scan_param scan_param = {
		.type       = BT_LE_SCAN_TYPE_PASSIVE,
		.options    = BT_LE_SCAN_OPT_NONE,
		.interval   = BT_GAP_SCAN_FAST_INTERVAL,
		.window     = BT_GAP_SCAN_FAST_WINDOW,
	};

	err = bt_le_scan_start(&scan_param, device_found);
	if (err) {
		return;
	}

	printk("Scanning successfully started\n");
}

void main(void)
{
	int err;
	err = bt_enable(NULL);

	if (err) {
		return;
	}

	start_scan();
}
```

prj.conf
```
CONFIG_BT=y
CONFIG_BT_CENTRAL=y
CONFIG_LOG=y
CONFIG_LOG_MODE_MINIMAL=y
CONFIG_LOG_MAX_LEVEL=4
```

**Environment**
- Zephyr v3.0.0
- Zephyr SDK v0.13.2
- Ubuntu 21.10

carlescufi, 2022-04-07:
@atoncetti can you please print the raw adveritising data tha triggers this warning and copy it here so we can analyze it manually?

atoncetti, 2022-04-07:
Hi @carlescufi, I was just doing that yesterday but couldn't point out where is the issue.

```
[DEVICE]: 35:4C:73:2A:95:07 (random), AD evt type 0, AD data len 19, RSSI -82
Raw adv data: 4C20404340E180E19F9
len: 76, ad->len: 18
Malformed data
[DEVICE]: --:--:--:--:--:03 (public), AD evt type 3, AD data len 28, RSSI -80
Raw adv data: 71E3BFE408820
len: 113, ad->len: 27
Malformed data
[DEVICE]: --:--:--:--:--:03 (public), AD evt type 3, AD data len 28, RSSI -82
Raw adv data: 71E3BFE408820
len: 113, ad->len: 27
Malformed data
[DEVICE]: 35:4C:73:2A:95:07 (random), AD evt type 0, AD data len 19, RSSI -83
Raw adv data: 4C20404340E180E19F9
len: 76, ad->len: 18
Malformed data
```

`len` and `ad->len` refer to the same variables in [`bt_data_parse()` ](https://github.com/zephyrproject-rtos/zephyr/blob/e97a2ee6d0d64a3c1711942db854e6ccf78ea8d6/subsys/bluetooth/host/hci_core.c#L3913)

cvinayak, 2022-04-28:
@atoncetti Could you let us know the peer device that is advertising?

The raw data does not follow the AD Data format of lenght-type-value.

atoncetti, 2022-05-05:
Hi @cvinayak

Here is an example of a scan from the bbc microbit (left) and from my ubuntu machine running a scan with `bluetoothctl` (right):

[image]

Strangely, the bbc microbit picks up a different device address even though there is a single device currently advertising.

Moreover, if I start advertising on my laptop, the bbc microbit picks up the same address as before but the adv data differs for the new device that is advertising:

[image]

I hope this helps, please let me know if you need something else.

cvinayak, 2022-05-06:
@atoncetti what do you get when using the samples/bluetooth/central or the samples/bluetooth/central_hr sample? Please use the peripheral_hr sample as the peer and let me know if the advertising data is parsed correctly by the central.

If you share your branch, steps to build your application and the peer device used for advertising, I can try to reproduce at my end.

atoncetti, 2022-05-10:
Hi @cvinayak,

unfortunately at the moment I do not have another device at hand on which I can flash the peripheral_hr (is there a way to simulate one?)

Anyways I've tried to run the samples/bluetooth/central on the bbc_microbit and I got similar results as above, namely, the bbc_microbit picks up a device address that is incorrect:

```
First test:
bbc_microbit "Device found: CE:BD:AF:BD:FD:03 (random) (RSSI -75)" |  correct device address 55:5C:75:13:56:BF

Second test:
bbc_microbit "Device found: 6E:B4:EA:F1:BF:02 (random) (RSSI -73)" | correct device address 6B:75:A7:57:8D:FA

Third test:
bbc_microbit "Device found: AB:BA:39:7A:22:00 (random) (RSSI -76)" | correct device address 7D:5D:D1:CB:D1:10

```

For all three tests I've used a device created with the _nRF Connect_ android app with the following configs:

Random address, connectable, Device type: LE only, Advertising type: Legacy

cvinayak, 2022-05-12:
@atoncetti I will will try out with nRF Connect android app and check if I get similar malformed data using samples/bluetooth/central on bbc_microbit board is observed.

atoncetti, 2022-05-12:
Hi @cvinayak,

Thank you for your support!

As a follow up, I have created a temporary repository with the exact same zephyr project I ran at the beginning of this issue.

Here is the code: https://github.com/atoncetti/zephyr_bbc_microbit_ble_fault_example

cvinayak, 2022-05-17:
I am able to reproduce the malformed data, it appears the device address and ad data are corrupt, the length seems correct. I will be investigating as part of the v3.1 release stabilization.

atoncetti, 2022-05-17:
Sounds great! Please let me know if there is anything I can help you with.

cvinayak, 2022-05-18:
Regression introduced in commit https://github.com/zephyrproject-rtos/zephyr/commit/1dcbe73cc8f31b6a207d23a3c825a7ed4a6b15f0 ("Bluetooth:
controller: radio: add setting of S1 byte in radio packet").

Fix posted in #45749.

Thank you for reporting the bug.

PR #45749 (2022-05-18): Bluetooth: Controller: Fix nRF51x advertising channel PDU regression
Fix nRF51x series advertising channel PDU S1 byte
configuration regression, that causes corrupted advertising
PDUs being reported to Host, by adding the missing bits_s1
local variable initialization.

Regression introduced in commit 1dcbe73cc8f3 ("Bluetooth:
controller: radio: add setting of S1 byte in radio packet").

Fixes #43465.
closed 2022-05-18 by carlescufi