zephyrproject-rtos/zephyr #43511: lvgl: upgrade to 8.2 build problem

ck-telecom, 2022-03-08:
**Describe the bug**
when enable
CONFIG_LVGL=y
CONFIG_LV_CONF_MINIMAL=y

all widget built-in and the binary and SRAM is mass, all files built-in even only CONFIG_LV_CONF_MINIMAL=y, this doesn't make any sense. `modules/lib/gui/lvgl/zephyr/lv_conf.h' need fixed

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   the component could be configured, but the actual build system is not correct

 - What have you tried to diagnose or workaround this issue?
    None

**To Reproduce**
Steps to reproduce the behavior:
1. CONFIG_LVGL=y CONFIG_LV_CONF_MINIMAL=y
2. west build -p auto -b <board> -s samples/subsys/display/lvgl
3 check the image size

**Expected behavior**
all component should only built-in when configured

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

ck-telecom, 2022-03-08:
#43192 only fix build error, no configure problem fixed

gmarull, 2022-03-08:
I see this:
```
west build -b pinetime_devkit0 samples/subsys/display/lvgl/ -p -- -DCONFIG_LV_CONF_MINIMAL=y
...
Memory region         Used Size  Region Size  %age Used
           FLASH:      106424 B       512 KB     20.30%
            SRAM:       47992 B        64 KB     73.23%
        IDT_LIST:          0 GB         2 KB      0.00%
```
In this case this is the rom_report for the LVGL part:
```
│       └── lib                                                                                 64691  60.79%
│           └── gui                                                                             64691  60.79%
│               └── lvgl                                                                        64691  60.79%
│                   ├── src                                                                     63235  59.43%
│                   │   ├── core                                                                21073  19.80%
│                   │   │   ├── lv_disp.c                                                         220   0.21%
│                   │   │   │   ├── lv_disp_get_layer_sys                                          52   0.05%
│                   │   │   │   ├── lv_disp_get_layer_top                                          52   0.05%
│                   │   │   │   ├── lv_disp_get_scr_act                                            52   0.05%
│                   │   │   │   ├── lv_disp_get_scr_prev                                           52   0.05%
│                   │   │   │   └── lv_disp_get_theme                                              12   0.01%
│                   │   │   ├── lv_event.c                                                        708   0.67%
│                   │   │   │   ├── _lv_event_mark_deleted                                         36   0.03%
│                   │   │   │   ├── event_send_core                                               356   0.33%
│                   │   │   │   ├── lv_event_get_code                                               8   0.01%
│                   │   │   │   ├── lv_event_get_current_target                                     4   0.00%
│                   │   │   │   ├── lv_event_get_draw_ctx                                          52   0.05%
│                   │   │   │   ├── lv_event_get_indev                                             52   0.05%
│                   │   │   │   ├── lv_event_get_param                                              4   0.00%
│                   │   │   │   ├── lv_event_get_target                                             4   0.00%
│                   │   │   │   ├── lv_event_send                                                  64   0.06%
│                   │   │   │   ├── lv_event_set_ext_draw_size                                     64   0.06%
│                   │   │   │   ├── lv_obj_event_base                                              42   0.04%
│                   │   │   │   └── lv_obj_get_event_dsc.isra.0                                    22   0.02%
│                   │   │   ├── lv_group.c                                                        656   0.62%
│                   │   │   │   ├── _lv_group_init                                                 12   0.01%
│                   │   │   │   ├── focus_next_core                                               226   0.21%
│                   │   │   │   ├── get_indev                                                      76   0.07%
│                   │   │   │   ├── lv_group_add_obj                                              104   0.10%
│                   │   │   │   ├── lv_group_focus_next                                            16   0.02%
│                   │   │   │   ├── lv_group_focus_prev                                            16   0.02%
│                   │   │   │   ├── lv_group_get_default                                           12   0.01%
│                   │   │   │   ├── lv_group_get_editing                                           10   0.01%
│                   │   │   │   ├── lv_group_refocus                                               40   0.04%
│                   │   │   │   └── lv_group_remove_obj                                           144   0.14%
│                   │   │   ├── lv_indev.c                                                         70   0.07%
│                   │   │   │   ├── lv_indev_get_act                                               12   0.01%
│                   │   │   │   ├── lv_indev_get_scroll_dir                                        28   0.03%
│                   │   │   │   ├── lv_indev_get_scroll_obj                                        22   0.02%
│                   │   │   │   └── lv_indev_get_type                                               8   0.01%
│                   │   │   ├── lv_obj.c                                                         3104   2.92%
│                   │   │   │   ├── lv_init                                                       116   0.11%
│                   │   │   │   ├── lv_obj_add_state                                               18   0.02%
│                   │   │   │   ├── lv_obj_allocate_spec_attr                                      56   0.05%
│                   │   │   │   ├── lv_obj_clear_flag                                             120   0.11%
│                   │   │   │   ├── lv_obj_clear_state                                             18   0.02%
[... cut ...]
│                           ├── lvgl_malloc                                                        40   0.04%
│                           └── lvgl_realloc                                                       44   0.04%
```

```
west build -b pinetime_devkit0 samples/subsys/display/lvgl/ -p
...
Memory region         Used Size  Region Size  %age Used
           FLASH:      196532 B       512 KB     37.49%
            SRAM:       48280 B        64 KB     73.67%
        IDT_LIST:          0 GB         2 KB      0.00%
```

ck-telecom, 2022-03-08:
`$ ls build/modules/lvgl/CMakeFiles/..__modules__lib__gui__lvgl__zephyr.dir/D_/personal/pinetime/modules/lib/gui/lvgl/src/widgets/ -lh
total 64K
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_arc.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_bar.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:29 lv_btn.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_btnmatrix.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.1K  3月  8 17:29 lv_canvas.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.1K  3月  8 17:30 lv_checkbox.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.1K  3月  8 17:30 lv_dropdown.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_img.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:29 lv_label.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_line.c.obj
-rw-r--r-- 1 GW00221994 1049089 1.7K  3月  8 17:30 lv_objx_templ.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:29 lv_roller.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_slider.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:29 lv_switch.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:29 lv_table.c.obj
-rw-r--r-- 1 GW00221994 1049089 2.0K  3月  8 17:30 lv_textarea.c.obj
`
It seems that all source file complied, no MACRO definition in 'modules/lib/gui/lvgl/zephyr/lv_conf.h' like
`1444 #ifndef LV_USE_SLIDER
1445     #ifdef _LV_KCONFIG_PRESENT
1446         #ifdef CONFIG_LV_USE_SLIDER
1447             #define LV_USE_SLIDER CONFIG_LV_USE_SLIDER
1448         #else
1449             #define LV_USE_SLIDER 0
1450         #endif
1451     #else
1452         #define LV_USE_SLIDER     1   /*Requires: lv_bar*/
1453     #endif
1454 #endif
`

gmarull, 2022-03-08:
@ck-telecom can you provide steps to reproduce this? I can't see the problems you describe with the commands I posted.

mbolivar-nordic, 2022-03-08:
cc   @brgl

ck-telecom, 2022-03-09:
It seems that all source file built-in, linked with only referenced symbols, this has no impact on firmware size, that should be OK

but I wounder why so many SRAM needed, I can't reduced even make some changes with LV_MEM, maybe releated to all source file is fully builtin.

gmarull, 2022-03-09:
I'm gonna close this one as it can't be reproduced using the upstream sample (only enabled widgets are built when `CONFIG_LV_CONF_MINIMAL=y` as detailed in https://github.com/zephyrproject-rtos/zephyr/issues/43511#issuecomment-1061573734). Please re-open with steps to reproduce in case you find a combination that triggers the problem.

closed 2022-03-09 by gmarull