zephyrproject-rtos/zephyr #43515: reel_board: failed to run tests/kernel/workq/work randomly

LixinGuoX, 2022-03-08:
**Describe the bug**
test_1cpu_running_cancel or test_1cpu_running_cancel_sync failed on reel_board, it is a random failure, the incidence can be 30% 

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p reel_board --device-testing -T tests/kernel/workq/work --device-serial=/dev/ttyACM0

Or use west:
1. west build -b reel_board tests/kernel/workq/work
2. west flash

**Expected behavior**
Test case should pass smoothly.

**Logs and console output**
```
DEBUG   - DEVICE: START - test_1cpu_running_cancel_sync
DEBUG   - DEVICE:
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:785: test_1cpu_running_cancel_sync: ctx->busy_rc not equal to K_WORK_RUNNING | K_WORK_CANCELING
DEBUG   - DEVICE:
DEBUG   - DEVICE: FAIL - test_1cpu_running_cancel_sync in 0.16 seconds
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2
 - Commit d74cb2a3df23

nashif, 2022-03-08:
can't reproduce, and it is reported as sporadic so lower priority.

peter-mitsis, 2022-03-10:
Although it is recognized that the raised issue references the reel_board, I have been trying to reproduce this issue on the frdm_k64f board for the following reasons:

1. I do not have a reel_board target, but I do have a frdm_k64f target.
2. Both the reel_board and the frdm_k64f use the cortex_m4, so it is hoped that they are sufficiently similar.

Thus far, I have been unable to reproduce the reported issue on the frdm_k64f. This has been attempted approximately 100 times each at both commit d74cb2a and the head (currently 8c069c36).

I am not sure if the following observation is relevant or not to this particular issue. There have been issues with the workq tests in the recent past on riscv's hifive1 target. Although that one is no longer causing issues, there seemed to be an update to the workq test suite that resolved that particular issue. This makes me at least suspect that there might be an issue with the test suite.

Next steps: Review the workq test suite to search for problems that might cause the reported issue.

nashif, 2022-03-15:
can't reproduce

PerMac, 2022-05-12:
I am reopening this issue since I also see this failures on nrf boards: nrf9160dk_nrf9160 nrf5340dk_nrf5340_cpuapp (zephyr-v3.0.0-3725-g806ee8601c). E.g. I can get it failed on nrf9160dk on my desk ~5/10 runs.
Log from nrf9160dk
```
DEBUG   - DEVICE: START - test_1cpu_drain_wait
DEBUG   - DEVICE: ASSERTION FAIL [handler != ((void *)0)] @ WEST_TOPDIR/zephyr/kernel/work.c:667
DEBUG   - DEVICE: E: r0/a1:  0x00000004  r1/a2:  0x0000029b  r2/a3:  0x00000001
DEBUG   - DEVICE: E: r3/a4:  0x00005ab5 r12/ip:  0x00000000 r14/lr:  0x00007bff
DEBUG   - DEVICE: E:  xpsr:  0x41000000
DEBUG   - DEVICE: E: Faulting instruction address (r15/pc): 0x0000ab98
DEBUG   - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG   - DEVICE: E: Current thread: 0x200004a8 (wq.coophi)
DEBUG   - DEVICE: E: Halting system
``` 
Also (roughly same often) I see such failure (not sure how related):
```
DEBUG   - DEVICE: START - test_1cpu_running_cancel_sync
DEBUG   - DEVICE: 
DEBUG   - DEVICE: Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/workq/work/src/main.c:785: test_1cpu_running_cancel_sync: ctx->busy_rc not equal to K_WORK_RUNNING | K_WORK_CANCELING
DEBUG   - DEVICE: 
DEBUG   - DEVICE: FAIL - test_1cpu_running_cancel_sync in 0.16 seconds
```

It can happen that this test fails 3 times in a row making our on-target test suite failing (we use --retry-failed 2). I guess it is more on the "annoyance" side, but still I think it is worth to reinvestigate such instability in the kernel.

peter-mitsis, 2022-05-16:
I don't have an NRF board available on my own desk to use to debug. However, I have made the following observations.

1. Both the reel_board and the NRF boards mentioned use the nrf_rtc_timer (CONFIG_NRF_RTC_TIMER=y).
2. The board I was previously unable to reproduce the problem on was the frdm_k64f, which uses the cortex_m_systick (CONFIG_CORTEX_M_SYSTICK=y).
3. This issue was originally opened March 8th-2022, and there were changes to nrf_rtc_timer.c dated March 3rd-2022.

It may be coincidence, but it seems that this is a good place to begin investigating.

peter-mitsis, 2022-05-16:
Minor update: It looks like the nrf_rtc_timer change was a bit of a red herring. That is, the disassembled output of the timer code both before and after the change were found to be identical.

closed 2022-08-01 by github-actions[bot]