zephyrproject-rtos/zephyr #43518: 'DT_N_S_soc_S_timers_40012c00_S_pwm' undeclared

zzzh, 2022-03-08:
**Describe the bug**
Got compile error when trying to move code from zephyr-2.3.0 to zephyr-3.0.99

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?   ARM-STM32F0
 - What have you tried to diagnose or workaround this issue?  changing dts file
 - changed as an example in sample, but don't work

**To Reproduce**
Steps to reproduce the behavior:
in my project folder, run west build, then error as below would appear:

my_project_path/build/zephyr/include/generated/devicetree_unfixed.h:4751:54: error: 'DT_N_S_soc_S_timers_40012c00_S_pwm' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_timers_40012c00_S_pwm_ORD'?
 4751 | #define DT_N_S_pwmleds_S_pwm_led_all_P_pwms_IDX_0_PH DT_N_S_soc_S_timers_40012c00_S_pwm

**Expected behavior**
A clear and concise description of what you expected to happen.
No error

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Can't continue

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

my_project_path/build/zephyr/include/generated/devicetree_unfixed.h:4751:54: error: 'DT_N_S_soc_S_timers_40012c00_S_pwm' undeclared here (not in a function); did you mean 'DT_N_S_soc_S_timers_40012c00_S_pwm_ORD'?
 4751 | #define DT_N_S_pwmleds_S_pwm_led_all_P_pwms_IDX_0_PH DT_N_S_soc_S_timers_40012c00_S_pwm

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows) Linux
 - Toolchain (e.g Zephyr SDK, ...) Zephyr SDK
 - Commit SHA or Version used   zephyr-3.0.99

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

code in dts file:

&timers1 {
    //st,prescaler = <10000>;
    status = "okay";

    pwm1: pwm {
        status = "okay";        
        pinctrl-0 = <&tim1_ch4_pa11 &tim1_ch3n_pb15 &tim1_ch2n_pb14 &tim1_ch1n_pb13>;
        pinctrl-names = "default";
    };
};

erwango, 2022-03-08:
Can you clarify which board you're using ? `ARM-STM32F0` is not an official board name

zzzh, 2022-03-09:
It's not an official board, but compile well in zephyr-2.3, some pinctrl thing had been added.

zzzh, 2022-03-09:
DT_PWMS_LABEL is deprecated, which command should be used instead?

erwango, 2022-03-09:
> DT_PWMS_LABEL is deprecated, which command should be used instead?

It depends what you need to do. Do you have an example of usage ?

erwango, 2022-03-09:
Should be something like `DT_PWMS_CTLR`

zzzh, 2022-03-09:
I used DT_PWMS_CTLR to replace DT_PWMS_LABEL, seems this is the source of the error.

for example:
it was: DT_PWMS_LABEL(DT_NODALABEL(xxx)), and work well in zephyr-2.3
now is: DT_PWMS_CTLR(DT_NODALABEL(xxx)), and don't work in zephyr-3.0.99

erwango, 2022-03-09:
I suggest to look at `samples/basic/blinky_pwm/src/main.c` to see how it is used in current code.

zzzh, 2022-03-10:
Got it, thank you.

zzzh, 2022-03-14:
@erwango 
I tried the way in example, it don't work.
Got undeclared issue in compile.

erwango, 2022-03-14:
Can you