zephyrproject-rtos/zephyr #43552: samples: bluetooth: direction_finding: Sample fails on nrf5340

PerMac, 2022-03-08:
**Describe the bug**
Samples defined at https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/direction_finding_connectionless_rx/sample.yaml and https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/direction_finding_connectionless_tx/sample.yaml fail on nrf5340dk_nrf5340_cpuapp.:
Failing ones: sample.bluetooth.direction_finding_connectionless_rx.aod and sample.bluetooth.direction_finding_connectionless.aoa
(FYI The other 2 scenarios in those yamls are skipped since we don't have "bluetooth" harness defined.)

Probably those tests fail on other allowed platforms but I cannot verify this. 
I think there are 2 issues here.
- the tests have no means of validating a results. Neither ztest nor console harnesses are used, hence such tests will always end up with a timeout
- there are errors present in the output

**To Reproduce**
1. `scripts/twister -T samples/bluetooth/direction_finding_connectionless_rx/ -p nrf5340dk_nrf5340_cpuapp --device-testing --device-serial /dev/ttyACM2 -v -v`
2. See error

**Expected behavior**
Tests pass or are skipped

**Impact**
Not clear 

**Logs and console output**
```
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v3.0.0-883-g6650a5304011  ***
DEBUG   - DEVICE: Starting Connectionless Locator Demo
DEBUG   - DEVICE: Bluetooth initialization...failed (err -11)
DEBUG   - DEVICE: success
DEBUG   - DEVICE: Scan callbacks register...success.
DEBUG   - DEVICE: Periodic Advertising callbacks register...success.
DEBUG   - DEVICE: Start scanning...failed (err -11)
```
and
```
*** Booting Zephyr OS build zephyr-v3.0.0-838-gc050d1f2ecb6  ***
Starting Direction Finding periodic advertising Beacon Demo
Bluetooth initialization...failed (err -11)
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

ppryga, 2022-03-09:
The PR to fix the problem: https://github.com/nrfconnect/sdk-nrf/pull/7085

PerMac, 2022-03-09:
@ppryga The issue relates to samples existing here, in the upstream Zephyr: https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/direction_finding_connectionless_rx/sample.yaml and https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/bluetooth/direction_finding_connectionless_tx/sample.yaml
Thanks for clearing sdk-nrf samples, but also please fix the ones from this issue.

PerMac, 2022-03-09:
I realized you did this as well: this is the fix https://github.com/zephyrproject-rtos/zephyr/pull/43592

closed 2022-03-15 by nashif