zephyrproject-rtos/zephyr #43553: Request to configure SPBTLE-1S of STEVAL-MKSBOX1V1

dbortoluzzi, 2022-03-08:
**Is your enhancement proposal related to a problem? Please describe.**
I would like to use the Ble Module (SPBTLE-1S) of the **STEVAL-MKSBOX1V1** (SensorTile.box) in a zephyr project. This mcu is supported by zephyr, but the .dts doesn't configure the spi2 interface used by bluetooth. 

**Describe the solution you'd like**
According with the hardware specs, I tried to start from ST BLE sensor project example and configure:

- *prj.conf*:
```
*CONFIG_BT_SPI*=y
*CONFIG_BT_HCI*=y1
*CONFIG_BT*=y
*CONFIG_BT_DEBUG_LOG*=y
*CONFIG_BT_PERIPHERAL*=y
*CONFIG_BT_DEVICE_NAME*="SPBTLE-1S"
*CONFIG_BT_GATT_CLIENT*=y
```

- *app.overlay*:
```
/ {
	chosen {
		zephyr,bt-hci-spi = &zephyr_bt_hci_spi;
		zephyr,bt-hci-spi-slave = &zephyr_bt_hci_spi_slave;
	};
};

&spi2 {
	pinctrl-0 = <&spi2_nss_pd0 &spi2_sck_pd1 &spi2_miso_pd3 &spi2_mosi_pc3>;
	pinctrl-names = "default";
	status = "okay";

	cs-gpios = <&gpiod 0 GPIO_ACTIVE_LOW>; 

	zephyr_bt_hci_spi: zephyr_bt_hci_spi@0 {
		compatible = "zephyr,bt-hci-spi";
		reg = <0>;
		irq-gpios = <&gpiod 4 GPIO_ACTIVE_LOW>;
		reset-gpios = <&gpioa 8 GPIO_ACTIVE_LOW>;
		spi-max-frequency = <1000000>;
		label = "SPBTLE-1S";
	};

	zephyr_bt_hci_spi_slave: zephyr_bt_hci_spi_slave@1 {
		compatible = "zephyr,bt-hci-spi-slave";
		reg = <1>;
		irq-gpios = <&gpiod 4 GPIO_ACTIVE_LOW>;
	};
};
```
	
Unfortunately, bluetooth isn't enabled correctly and the system hangs up:
in the main.c, the *bt_enable* waits endlessly, because the function *bt_spi_rx_thread* of the file https://github.com/zephyrproject-rtos/zephyr/blob/v2.7-branch/drivers/bluetooth/hci/spi.c (line 343) doesn't return, looping in the do-while.

**Describe alternatives you've considered**

**Additional context**

*Zephyr version*: 2.7.0

*Sources*: 
- https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/bluetooth/st_ble_sensor : st ble example project
- https://www.st.com/resource/en/schematic_pack/steval-mksbox1v1_schematic.pdf : electric schema of the sensortile.box
- https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/arm/stm32l562e_dk/stm32l562e_dk_common.dtsi : sjmilar dtsi (SPBTLE-RF module)

erwango, 2022-03-10:
@avisconti I think I recall you ported that board. Would you be able to check that at least the configuration is ok ?

erwango, 2022-03-10:
@dbortoluzzi I'd suggest to add the following flags if not done already:
```
CONFIG_SPI=y
CONFIG_BT_HCI_ACL_FLOW_CONTROL=n
CONFIG_BT_HCI_VS_EXT=n
```
Then I'd try both configurations on top:
```
CONFIG_BT_BLUENRG_ACI=y
CONFIG_BT_SPI_BLUENRG=n
```
and if not working:
```
CONFIG_BT_BLUENRG_ACI=y
CONFIG_BT_SPI_BLUENRG=y
```
If none of these working, and dts configuration is proven to be correct.
Then, I'd suggest to study SPBTLE-1S module documentation against `BT_BLUENRG_ACI` and `BT_SPI_BLUENRG`.

avisconti, 2022-03-11:
> @avisconti I think I recall you ported that board. Would you be able to check that at least the configuration is ok ?

Yes, I ported that board some time ago. If I remember well the BLE part had been skipped at that time and only USB was available.

avisconti, 2022-03-11:
@erwango @dbortoluzzi 
I think I'll require some time to re-construct the BLE situation. I'll try next week.

erwango, 2022-03-21:
Moving to "Discussion"