zephyrproject-rtos/zephyr #43555: Variables not properly initialized when using data relocation with SDRAM

hjuul, 2022-03-08:
When using data relocation to store variables in external SDRAM the kernel will initialize the data section from ROM and clear the bss section. But this is done *before* the SDRAM is initialized, so the data will not be retained. In my application I see that all variables that are stored in SDRAM have wrong initialization values.

I'm seeing this with an stm32f7, using driver `memc_stm32_sdram.c`. This driver has the initialization level POST_KERNEL. I have verified that `kernel/init.c` and `kernel/xip.c` correctly initialize the relocated data, and also that the data is lost during `memc_stm32_sdram_init()`.

As a workaround I'm calling `bss_zeroing_relocation()` and `data_copy_xip_relocation()` at the end of `memc_stm32_sdram_init()`. This fixes the problem for me, but it is not a general solution because it may overwrite any other data relocated to internal RAM.

One possible solution to this is to initialize the SDRAM a lot earlier, but that will probably require a rewrite of the driver (to not depend on kernel, clock initalization, etc)

**Environment**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Zephyr v2.7.0

hjuul, 2022-03-16:
@erwango , I've created a demonstration for this based on `samples/application_development/code_relocation`. I've tested it on stm32f769i_disco and it fails. You can find it in my zephyr fork:
https://github.com/hjuul/zephyr/tree/reproduce/43555

To reproduce on that branch:
```
west build zephyr/samples/application_development/code_relocation -b stm32f769i_disco 
west flash
```
When running this, somewhere in the output you should see something like:
```
***** var_sdram_data=558401858, expected 10
***** var_sdram_bss=-2147326859, expected 0
```

Now, if you revert the CMakeLists.txt file that I've modified, so that the relocation is not to SDRAM, you will see that variables are correctly initialized.

hjuul, 2022-07-16:
Please remove the stale label, this is a quite critical bug.

hjuul, 2022-10-24:
@erwango, I already have a workaround for this in my fork, so it doesn't block me. But it is one of those bugs that really makes you question the maturity of Zephyr, when something like RAM initialization doesn't work out of the box on such a popular platform as the stm32.
Please consider removing the stale label and then eventually fixing it.

FRASTM, 2022-11-25:
@hjuul could you please share a branch (remotes/hjuul/reproduce/43555 is quite old) 
Trying to build for another platform (stm32f746g_disco) also fails.

cfriedt, 2023-05-27:
> I'm seeing this with an stm32f7, using driver `memc_stm32_sdram.c`. This driver has the initialization level POST_KERNEL. I have verified that `kernel/init.c` and `kernel/xip.c` correctly initialize the relocated data, and also that the data is lost during `memc_stm32_sdram_init()`.

@hjuul - it seems like the best solution is to make a PR where you change the init level for the driver to `PRE_KERNEL_1`. Are you able to verify that fix works and submit a PR?

closed 2023-08-11 by github-actions[bot]