zephyrproject-rtos/zephyr #43562: Setting and/or documentation of Timer and counter use/requirements for Nordic Bluetooth driver

ballardr, 2022-03-08:
**Describe the bug**
I'm trying to enable a device with bluetooth low energy (BLE), however my firmware is crashing once enabled, and I think this may be due to the firmware using rtc0 and timer0. Documentation at https://docs.zephyrproject.org/latest/guides/bluetooth/bluetooth-arch.html indicates that an rtc instance and 2 timer instances (counters I assume) are required. My assumption here comes from the fact that as soon as I enable "CONFIG_BT=y", my TIMER0 device disappears (and hence when I found the above link)

Is there anyway to set which specific timers/counters instances and which rtc instances are used in the nordic bluetooth driver, alternatively, can the hardcoded instance requirements be documented.

I have been unable to find the code/logic which disables TIMER0 when CONFIG_BT is enabled.

**Expected behavior**
An ability to set the device instances being used by the driver or more complete documentation.

**Impact**
Inability to use bluetooth in current state.

**Environment (please complete the following information):**
 - Zephyr 2.7
 - Nordic nrf52840

**Additional context**
The same linked document also says two GPIOs are required, but gives no indication of what these are or where they are set.

I've looked in the dts files (that are created during the build process) and can't see any mention of timers (or GPIO) required by BLE.

My crash only occurs when BT is enabled, and I've already increased the heap and stack sizes so I don't think that is the issue, and so when I saw the TIMER0 device disappear from the DT, I thought I better follow this lead before following other potential troubleshooting paths.

ballardr, 2022-03-22:
For anyone looking at this, for the nrf52840, the BT subsys drivers use TIMER0 and RTC0 by default (timer may be TIMER4 if CONFIG_BT_CTRL_TIFS_HW) is enabled. Also note that RTC1 is not selectable as a kconfig so RTC2 would need to be used as an alternative/

carlescufi, 2022-05-19:
@cvinayak can you work on this for 3.1 please?

cvinayak, 2022-09-07:
Some discussions to consider while adding documentation:
https://discord.com/channels/720317445772017664/720317445772017667/1016411681221525667
https://github.com/zephyrproject-rtos/zephyr/issues/21917

PR #50842 (2023-04-25): doc: connectivity: Bluetooth: Add initial Controller sections
Add initial Controller architecture and design sections. These are images contributed by Zephyr member companies and collected over the various Face-to-Face meetings we have been hold in the past years of Bluetooth Low Energy Controller implementation.

Document the Controller SoC peripheral requirements for Nordic Semiconductor SoCs.

Fixes #43562.
closed 2023-04-25 by carlescufi