zephyrproject-rtos/zephyr #43600: tests: mec15xxevb_assy6853: most of the test cases failed

shaoanxli, 2022-03-09:
**Describe the bug**
Most of the test cases failed after commit id: 303469350978f29b4108eff57d9dd25df24a3f63<drivers: i2c: MEC15xx: Improved error handling>.The root cause is not clear yet.
Take tests/kernel/threads/thread_apis as an example :

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. west build -p always -b mec15xxevb_assy6853 tests/kernel/threads/thread_apis 
3. Open another terminal and execute: picocom -b 115200 /dev/ttyUSB0
4. sh pre_script.sh (which is used for pulling down the nRST)
5. west flash 
6. sh post_script.sh (which is used for pulling up the nRST)
7. See error
or
2. twister --hardware-map $HOME/mec1.file --device-testing -T tests/kernel/threads/thread_apis -vv
3. See error

**Expected behavior**
Test case should pass as expected.

**Impact**
Most of the test cases failed. 

**Logs and console output**
```
DEBUG -
Searching for connected FTDI devices with USB VID:PID=0x0403:0x6001...
SKIPPING: FTDI device 0 (FTDI TTL232R-3V3, Serial #: FT9FPJYP)
MATCH: FTDI device 1 (FTDI TTL232R-3V3, Serial #: FT9FTP8W)
Setting GPIO 0x01 to 1.
Done!

DEBUG - DEVICE: ASSERTION FAIL [data->sda_gpio != data->scl_gpio] @ WEST_TOPDIR/zephyr/drivers/i2c/i2c_mchp_xec.c:902
DEBUG - DEVICE: Both i2c pins on same GPIO
DEBUG - DEVICE: E: r0/a1: 0x00000004 r1/a2: 0x00000386 r2/a3: 0x00118b80
DEBUG - DEVICE: E: r3/a4: 0x000e4319 r12/ip: 0x00000000 r14/lr: 0x000e47f5
DEBUG - DEVICE: E: xpsr: 0x41000000
DEBUG - DEVICE: E: Faulting instruction address (r15/pc): 0x000e930a
DEBUG - DEVICE: E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
DEBUG - DEVICE: E: Current thread: 0x118b80 (main)
DEBUG - DEVICE: Caught system error -- reason 4 0
DEBUG - DEVICE: Fatal error was unexpected, aborting...
```
**Environment:**

OS: Linux,
Toolchain Zephyr SDK 0.13.2
Commit SHA or Version: https://github.com/zephyrproject-rtos/zephyr/commit/303469350978f29b4108eff57d9dd25df24a3f63

enjiamai, 2022-03-09:
Hi @jvasanth1 , this seems happens after https://github.com/zephyrproject-rtos/zephyr/pull/43080 got merged, could you please take a look at this, thanks!

jvasanth1, 2022-03-09:
The assertion is invalid since the clk and data gpio could be same gpio group; will remove it.

enjiamai, 2022-03-14:
Verified issue fixed. Close it.

closed 2022-03-14 by enjiamai